RIOT-OS/RIOT #13667: [Tracking] modules depending on 64-bit xtimer implementation

jue89, 2020-03-20:
#### Description

The new ztimer offers the pseudo-module `ztimer_xtimer_compat` that replaces all xtimer related API by ztimer functions. Since ztimer doesn't offer 64-bit timers, no replacement functions can be found for the `xtimer_*64()` functions.

In most cases, those 64-bit timers are used for long-running timers. Howover, the high precision is not required.

This issue tracks modules that require the 64-bit xtimer functions and are not compatible with `ztimer_xtimer_compat`.

##### sys
- sys/arduino
- sys/event
- sys/evtimer
- sys/net/dhcpv6/client
- sys/net/sntp
- sys/net/gnrc/gomach
- sys/net/gnrc/rpl
- sys/net/gnrc/ipv6/nib
- sys/net/fib
- sys/newlib_syscalls_default
- sys/posix
- sys/random/fortuna
- sys/sema
- sys/trickle

##### drivers
- drivers/ata8520e
- drivers/ltc4150
- drivers/pir
- drivers/rn2xx3
- drivers/xbee

##### pkg
- pkg/lwip
- pkg/semtech-loramac
- pkg/wakaama

I already opened a PR for removing the need for 64-bit xtimer of `gnrc_ipv6_nib`: #13666

kfessel, 2021-03-03:
i went throug the list updated and added some notes

##### sys

* **sys/arduino** : used for unsigned long millis() -> ZTIMER_MSEC or MODULE_ZTIMER_NOW64

* sys/event :  deprecate event_wait_timeout64 (no user in RIOT ouside of tests)

* ~**sys/evtimer**~ : #13661 pending merged 2020-09-217

* **sys/net/dhcpv6/client** : timstamp -> needs much less resolution of 1/100 Sec (cs for centisecond)-> ZTIMER_MSEC or MODULE_ZTIMER_NOW64 

* sys/net/sntp : timstamp -> MODULE_ZTIMER_NOW64 (may realy need that much resolution

* sys/net/gnrc/gomach

* **sys/net/gnrc/rpl**: calculates milliseconds -> ZTIMER_MSEC or MODULE_ZTIMER_NOW64

* ~**sys/net/gnrc/ipv6/nib**~: #13666 removed the dependency

* **sys/net/fib**: used to track a lifetime in milliseconds (timestamps ->  MODULE_ZTIMER_NOW64) the code is quiet spread -> not a fast rewrite to ZTIMER_MSEC

* sys/newlib_syscalls_default :  gettimeofday uses the size and the resolution of a 64bit timestamp-> MODULE_ZTIMER_NOW64 

* sys/posix: pthread_rwlock_timedlock and cond and semaphore use the size and the resolution to setup a timer by struct timespec definition

* sys/random/fortuna : timestamps to track the reseed interval dokumentation states 100ms is good ->  millisecond tracking is enaugh ZTIMER_MSEC 

* sys/sema : timestamps and xtimer_mutex_lock_timeout in `_sema_wait(sema_t *sema, int block, uint64_t us)`

* sys/trickle : trickels in ms -> rewrite to ZTIMER_MSEC

##### drivers

* **drivers/ata8520e**: wait for uint8 (256) seconds with us resolution -> 32 bit is enaugh (one function rewrite)

* drivers/ltc4150: has a 64Bit timestamp callback MODULE_ZTIMER_NOW64 but would considder redefinig the api to 32Bit

* drivers/pir: timestamp to calculate the occupation between readout i think this can live with 32 Bit (time between readout < the 2^32 usec) or MODULE_ZTIMER_NOW64

* drivers/rn2xx3: wait for uint8 (256) seconds with us resolution -> 32 bit is enaugh (one function rewrite)

* drivers/xbee: wait 1 second with us precision -> 32 bit is enaugh (one function rewrite)

##### pkg

* pkg/lwip: now need 32Bit of MSEC and wait on semaphore for 32Bit msec

* ~**pkg/semtech-loramac**~: is ported to ztimer

* **pkg/wakaama**   time_t lwm2m_gettime(void) timestamps of 32Bit seconds

fjmolinas, 2021-06-15:
@leandrolanzieri could `wakama` use `ZTIMER_SEC` instead to tick an item of the list?

fjmolinas, 2021-11-02:
I moved the list to the issue header, easier to keep track of these.