RIOT-OS/RIOT #13686: [REQ] - LWMAC/GOMACH on Openmote-b

pietrotedeschi, 2020-03-23:
It is possible to enable the LWMAC or GOMACH on Openmote-b? When I tried to compile the example gnrc_networking_mac (by commenting by the #BOARD_WHITELIST entry) I am not able to compile it. So, is there a solution to work with the MAC layer on Openmote-b?

benemorius, 2020-07-08:
It should be possible with minimal effort to get both LWMAC and GoMacH working on any platform that has `periph_rtt` available. I don't know whether that includes `openmote-b`.

If you need a port of LWMAC or GoMacH from `periph_rtt` to `xtimer` you can try [this](https://github.com/benemorius/RIOT/tree/openlabs) branch, but I haven't maintained either port since a year ago aside from merge conflicts and I don't know whether they still compile or work.

If you can post what compile errors you're running in to that would help someone guide you to getting it working.

zhuoshuguo, 2020-07-09:
> It should be possible with minimal effort to get both LWMAC and GoMacH working on any platform that has periph_rtt available. I don't know whether that includes openmote-b.

Yes, that is ture! Both GoMacH and LWMAC need RTT module for timing.

Also, make sure that openmote-b supports radio RX-start and TX-END interrupts (at least, GoMacH relies on these hardware functionalities).

nancy-0, 2020-07-13:
Hi @benemorius and @zhuoshuguo,

I have just tested the gnrc_lwmac.c file under the tests directory in order to test the MAC layer on openmote-b, and I have changed the BOARD_WHITELIST to openmote-b in the makefile, and everything compiles successfully with this warning (There are unsatisfied feature requirements: periph_rtt  EXPECT ERRORS!)
And I was successfully able to send and receive packets between 2 motes (please see the attached picture). 
So, my question how it is possible to get rid of this error/warning? and am I able to send and receive in this way with lwmac? 
 
[image]

Thank you, 
Nancy

nancy-0, 2020-07-17:
Hi @benemorius, @zhuoshuguo and @miri64,

I would like to update you that I have solved the error/warning related to the both LWMAC and GoMacH by adding the rtt feature in the makefile of the directory that I was testing. and hence the problem was gone. 

Thanks, 
Nancy

closed 2021-03-19 by stale[bot]