RIOT-OS/RIOT #13721: Read 16 bit data from SPI

CJayakumar, 2020-03-26:
Hello Team,

We could not find in riotos to connect wifi with board b-l475e-iot01a.So we are trying to write driver to connect wifi with b-l475e-iot01a, The board is using esWifi module so esp32 will not support here.
Issue is,
We can  read and write 8 bit data using SPI in riotos but we could not find option to read 16 bit data.If there is  way to read 16 bit data from SPI, it would helpful  us to resolve the issue.

Thank you

benpicco, 2020-03-26:
Can you not just use `spi_transfer_bytes()`?

```C
uint16_t out = 0x1234;
uint16_t in;
spi_transfer_bytes(bus, cs, false, &out, NULL, sizeof(out));
spi_transfer_bytes(bus, cs, true, NULL, &in, sizeof(in));
```

CJayakumar, 2020-03-27:
Hello Team,

Thanks for your update.

Yes we are using the below method for receive data,
spi_transfer_bytes(dev,SPI_CS_UNDEF, false,NULL, pData, len); 
and below method for send data,
spi_transfer_bytes(dev,SPI_CS_UNDEF, false,pdata, NULL, len);

But in spi.c, we found only configuration SPI for 8-bit data. there is no option for 16 bit data.
Here attached file link for your reference and kindly check and provide your suggestion.
https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32_common/periph/spi.c

Thank you.

benpicco, 2020-03-27:
Ah I see - stm32 has an option to load 16 bit or two data bytes into `DR` in one go to improve performance.
Yes this has not been implemented yet - patches welcome :smiley: 

From my brief skimming of the data sheet, for polling mode it might be enough to set the `DFF` bit as long as there is still an even number of bytes in the transfer buffer, then always read two bytes and write them to `DR`, if needed clear `DFF` for the last byte.

For DMA mode you might just transfer one byte manually if the transfer buffer has an odd size and write the rest in 16-bit mode, always reading/writing two bytes at at time.

CJayakumar, 2020-03-30:
Hello Team,
Thanks for the update.
We implemented and tested code in spi.c to read and write 16bit data.
It will not affect existing functional call.
Could you please tell us the procedure to check-in code in riotos, so that somebody can use this feature.

Thank you.

benpicco, 2020-03-30:
If you have your RIOT fork on Github, you can click that "New pull request" button and select your source branch. You'll get a template with a description to fill out.

[CONTRIBUTING.md](https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md) has a more in-depth explanation.

miri64, 2020-07-06:
@CJayakumar are your questions answered? Can this issue be closed?

closed 2021-02-13 by stale[bot]