RIOT-OS/RIOT #13744: sys: can: raw: conn_can_raw_recv returns an improper ETIMEDOUT

Ciusss89, 2020-03-28:
#### Description

As reported in [sys/include/can/conn/raw.h](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/can/conn/raw.h):
```
/**
 *  @brief  Generic can receive
 *
 * @param[in] conn          CAN connection
 * @param[out] frame        CAN frame to receive
 * @param[in] timeout       timeout in us, 0 for infinite
 *
 * @return the number of bytes received
 * @return any other negative number in case of an error
 */
int conn_can_raw_recv(conn_can_raw_t *conn, struct can_frame *frame, uint32_t timeout);
```
The function returns `ETIMEDOUT` (-116) when it's timeout is expired. This behaviour is correct when we pass a timeout equal to 0 ( that it means infinite) but It gives wrong feedback if a not 0 timeout has been set.

I suggest to edit line 233 from:
https://github.com/RIOT-OS/RIOT/blob/1d5485aef3ace3137f3ea4ae44b1549e2b7b8833/sys/can/conn/raw.c#L233
to

`ret = timeout ? 0 : -ETIMEDOUT;`

#### Steps to reproduce the issue

Use the [tests/conn_can](https://github.com/RIOT-OS/RIOT/tree/master/tests/conn_can)

Now:
```
2020-03-28 10:31:08,280 # 0: launching receive_thread
2020-03-28 10:31:08,283 # 1: launching receive_thread
> 2020-03-28 10:31:14,514 #  test_can recv 0 0 10000000 32
> 2020-03-28 10:31:19,357 #  0: can_stm32_0 32  [4]  01 02 03 04
2020-03-28 10:31:19,809 # 0: can_stm32_0 32  [4]  01 02 03 04
2020-03-28 10:31:20,363 # 0: can_stm32_0 32  [4]  01 02 03 04
2020-03-28 10:31:20,802 # 0: can_stm32_0 32  [5]  01 02 03 04 05
2020-03-28 10:31:21,243 # 0: can_stm32_0 32  [4]  01 02 03 04
2020-03-28 10:31:31,242 # 0: recv terminated: ret=-116
```

With the suggested fix:
Last line switch to 0.

I can open a PR to fix it.

#### Expected results

#### Actual results

#### Versions

Ciusss89, 2020-04-01:
ping @JannesVolkens @wosym @vincent-d

wosym, 2020-04-01:
line 233 is in `    case _TIMEOUT_RX_MSG_TYPE:`. 
I think if we get there, a timeout has certainly occurred, and I don't see why you'd want to return 0? 

Is it possible that it's because you're using the con_cann test-app, with the `test_can recv` command that it keeps receiving for as long as you specified in the timeout? (10 seconds in this case)
This means that it will return a positive number 5 times (because it received data 5 times while receiving), and then finally timeout. 

I don't think you'll experience this bug if you use the function outside of the test_app. 

(But it's entirely possible that I'm saying dumb things here. I haven't used conn_can that much yet)

Ciusss89, 2020-04-13:
> line 233 is in `case _TIMEOUT_RX_MSG_TYPE:`.
> I think if we get there, a timeout has certainly occurred, and I don't see why you'd want to return 0?
>
> Is it possible that it's because you're using the [...]

If a timeout has been set for the rcv when it will be expired; I would expect to know how much bytes have been received by the socket, but this information is lost because It returns ETIMEDOUT.

If a timeout is set how we can differentiate these two cases:
- 1. socket has received XXX bytes
- 2. socket doesn't receive anything

In both case now ETIMEDOUT is returned.

wosym, 2020-04-14:
Could you clarify if you're talking about the return value of `conn_can_raw_recv()` or about the `test_can recv` shell command in the conn_can test app?

Because I think this is where the confusion lies. 
`conn_can_raw_recv()` returns `rx->data.iov_len` when receiving data. It does not loop by itself, and thus after receiving a single can message, it will no longer do anything unless it is called again. If it does not receive data before the timeout it will return `-ETIMEDOUT`. 

The shell command `test_can recv` however, will (the way I see it by quickly glancing over it) start a loop and do multiple `conn_can_raw_recv()` calls in a row, up until the chosen time has passed. 
Every time this function is called,it will return the number of bytes received. But the test app does currently not print this, it only displays the received bytes. If you want the message length to be displayed, you can add an extra line in there printing the `ret` value. 
The ret value currently is only printed when the time is up and the timeout is returned. 

So as far as I can tell, this is all working as intended. But I might be entirely missing the point here. Sorry if I am :)

Ciusss89, 2020-04-18:
Thanks @wosym , you're right. 

I misunderstand the API description and your point of view is clear, I was wrong.  I close the issue. :)

closed 2020-04-18 by Ciusss89