RIOT-OS/RIOT #13745: ethos: Unable to choose global source address.

trickeydan, 2020-03-28:
#### Description

When using ethos on a border router, it is not possible to add a global subnet to the link network. Thus all packets that are routed from the border router have a link-local address as the source address.

#### Steps to reproduce the issue

Using samr21-xpro connected to the edbg port.

Route some global v6 to the computer that the border router will be connected to. Ensure that firewall allows ICMP for testing.

Compile and flash the `gnrc_border_router` example. Use `sudo make term` to start ethos on the computer that the border router is attached to.

Attempt to ping a global IPv6 address. It should not be in the subnet that is used for the wireless net. This will fail as the source address is a link-local.

We can attempt to add a global `/64` to the link using `ifconfig`

#### Expected results

Add a global address to the ethos interface on RIOT. Thus packets can be globally routed.

`error: unable to add IPv6 address`

Having dug into the RIOT source, this appears to be related to the flags on the netdev interface, but I cannot work out the exact cause.

This error does not happen if a different interface is used instead of `ethos`, i.e `usb_cdc_ecm`.

#### Actual results

[image]

#### Versions

```
Operating System Environment
-----------------------------
       Operating System: "Arch Linux" 
                 Kernel: Linux 5.4.26-1-lts x86_64 unknown

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Arch Linux 9.3.0-1) 9.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 9.2.0
                avr-gcc: avr-gcc (GCC) 9.2.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.1.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.16.5
               cppcheck: missing
                doxygen: 1.8.17
                    git: git version 2.25.2
                   make: GNU Make 4.3
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 3.8.2
                python2: Python 2.7.17
                python3: Python 3.8.2
                 flake8: error: /usr/bin/python3: No module named flake8
             coccinelle: missing
```

MBradbury, 2020-04-20:
I encountered the same issue of `error: unable to add IPv6 address` when trying to add a global IPv6 address to the ethos interface of my border router.

The issue was the maximum number of IPv6 addresses was set to 2 and the ethos iface was given fe80::a:6dff:febf:82e1 and fe80::2.

I can add a global IPv6 address when allowing three IPv6 addresses per interface in the makefile
`CFLAGS += -DCONFIG_GNRC_NETIF_IPV6_ADDRS_NUMOF=3`, but it seems to me that the ethos iface should not be given two link IPv6 addresses.

```
ifconfig
Iface  7  HWaddr: 2F:0A  Channel: 26  Page: 0  NID: 0x23
          Long HWaddr: 00:12:4B:00:14:D5:2F:0A
           TX-Power: 0dBm  State: IDLE
          AUTOACK  L2-PDU:102 MTU:1280  HL:64  RTR
          RTR_ADV  6LO  IPHC
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::212:4b00:14d5:2f0a  scope: link  VAL
          inet6 addr: ab12::212:4b00:14d5:2f0a  scope: global  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ffd5:2f0a

Iface  6  HWaddr: 02:0A:6D:BF:82:E1
          L2-PDU:1500 MTU:1500  HL:64  RTR
          Source address length: 6
          Link type: wired
          inet6 addr: fe80::a:6dff:febf:82e1  scope: link  VAL
          inet6 addr: fe80::2  scope: link  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ffbf:82e1
          inet6 group: ff02::1:ff00:2
```

miri64, 2020-04-22:
Hi, sorry for the late reply, but since a firewall is mentioned: have you allowed the UHCP port as well (12345)? Alternatively, you can use DHCPv6 with setting the environment variable `USE_DHCPV6=1` (and clear port 547 as well of course in the firewall).

miri64, 2020-04-22:
> I can add a global IPv6 address when allowing three IPv6 addresses per interface in the makefile
> `CFLAGS += -DCONFIG_GNRC_NETIF_IPV6_ADDRS_NUMOF=3`, but it seems to me that the ethos iface should not be given two link IPv6 addresses.

The new address is not added to the ethos interface, but to the WPAN interface, which in your case happened. The additional link-local address is there to simplify the configuration of the static route between the host machine and the border router. You can deactivate this (but might need to configure the routes yourself then, if your upstream host does not support RAs e.g.) by setting `STATIC_ROUTES=0` in your environment.

maribu, 2023-05-19:
Since the reporter is unresponsive, there is little we can do now. So I'm closing this now.

I'm happy to reopen the issue when activity resumes.

closed 2023-05-19 by maribu