RIOT-OS/RIOT #13822: Reaching tap interface from native node not working

leandrolanzieri, 2020-04-06:
#### Description

Pinging the tap interface on a Linux host from a `native` node, using a global IP does not seem to work. Linux is not responding to NS messages:
[image]

On the other hand, if I assign the global IP to `tapbr0`, things work correctly:
[image]

Any idea why this might be happening?

#### Steps to reproduce the issue
1. Execute `tapsetup` script
```shell
dist/tools/tapsetup/tapsetup -c
```
2. Assign global scope IP to `tap0`
```shell
ip addr add fd00:bbbb::1/64 dev tap0 scope global
```
3. On `examples/gnrc_networking`, assign also a global IP
```shell
ifconfig 6 add unicast fd00:bbbb::2/64
```
4. Ping the Linux host from `native` node
```shell
ping6 fd00:bbbb::1
```

#### Expected results
Ping succeeds.

#### Actual results
Ping fails
```
--- fd00:bbbb::1 PING statistics ---
3 packets transmitted, 0 packets received, 100% packet loss
```
#### Versions
```
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "19.04 (Disco Dingo)"
                   Kernel: Linux 5.0.0-38-generic x86_64 x86_64
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 8.3.0-6ubuntu1) 8.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
               msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 8.0.0-3 (tags/RELEASE_800/final)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "2.5.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: "2.2.0"
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.6
                    cmake: cmake version 3.13.5
                 cppcheck: Cppcheck 1.89
                  doxygen: 1.8.13
                      git: git version 2.20.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-00523-g2a3b709a (2018-08-23-10:48)
                   python: Python 2.7.16
                  python2: Python 2.7.16
                  python3: Python 3.7.3
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

miri64, 2020-04-06:
> ```
> dist/tools/tapsetup/tapsetup -c
> ```

As you noticed: This creates two TAP devices and assigns them to a bridge. If an interface is in a bridge, IPv6 can only be accessed through the bridge. Imagine the bridge as a switch, with the bridge interface being the port you put your host system and the other interfaces put to the `native` instances. The expected result is to my knowledge just not possible with Linux.

closed 2020-04-06 by miri64

leandrolanzieri, 2020-04-06:
@miri64 thanks for the explanation!