RIOT-OS/RIOT #13834: DHCPv6 client ends up in busy loop after a while

benpicco, 2020-04-07:
#### Description
I noticed that after a while, the DHCP client would get stuck [in a loop](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/application_layer/dhcpv6/client.c#L823).

#### Steps to reproduce the issue

Yet unknown :(

But this was with `examples/gnrc_border_router` with `CFLAGS=-DDHCPV6_CLIENT_PFX_LEASE_MAX=2 USE_DHCPV6=1`.

It took maybe an hour before this happened.

#### Expected results
The DHCPv6 client is reliable & stable.

#### Actual results

```
DHCPv6 client: send RENEW
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
```
will repeat until the node is restarted.

#### Versions
RIOT master

benpicco, 2020-04-07:
This seems to happen quite reliably:

```
DHCPv6 client: send RENEW
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
120 < 4 || 65e140 != 8576ba
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
120 < 4 || 38fdfe != 8576ba
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
120 < 4 || 472fd1 != 8576ba
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
120 < 4 || e0cd1b != 65e140
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
120 < 4 || b59eb1 != 65e140
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
120 < 4 || 3b6316 != 38fdfe
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
120 < 4 || 661800 != 38fdfe
DHCPv6 client: send REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
120 < 4 || 54e64b != 472fd1
```

Turns out the transaction ID is wrong

benpicco, 2020-04-07:
Looks like if there are multiple events in `event_queue`, the next event will call `_generate_tid()` again and overwrite the `transaction_id` that the previous event is waiting for.

benpicco, 2020-04-07:
I *think* the solution is to just use the same transaction ID used by `REQUEST` in `RENEW`.

miri64, 2020-04-08:
I think that breaks the protocol, because you now use the same TID for different transactions. I think we rather should use a distinct TID variable (from the counter) for REQUEST and RENEW.

benpicco, 2020-04-08:
> Does not make sense since `REQUEST` and `RENEW` have the same response type (`REPLY`).

So they should be the same transaction?
I just noticed the server would answer to `RENEW` with the same transaction ID that was used for `REQUEST`.

```
DHCPv6 client: resend SOLICIT
DHCPv6 client: received ADVERTISE
DHCPv6 client: scheduling REQUEST
DHCPv6 client: send REQUEST
transaction_id: 6b7d8
DHCPv6 client: resend REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: scheduling RENEW in 10 sec
DHCPv6 client: scheduling REBIND in 20000 sec
DHCPv6 client: send RENEW
transaction_id: 15eded
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
expected 15eded, got 6b7d8
DHCPv6 client: send REQUEST
transaction_id: bf2cd4
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
expected bf2cd4, got 6b7d8
DHCPv6 client: send REQUEST
transaction_id: 431846
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
expected 431846, got 6b7d8
DHCPv6 client: send REQUEST
transaction_id: c04cee
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
expected c04cee, got 15eded
DHCPv6 client: send REQUEST
transaction_id: 1f71ec
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
expected 1f71ec, got 15eded
DHCPv6 client: send REQUEST
transaction_id: 8cc4e
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
expected 8cc4e, got bf2cd4
DHCPv6 client: send REQUEST
transaction_id: ad98e8
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
expected ad98e8, got bf2cd4
DHCPv6 client: send REQUEST
transaction_id: 505a4
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
expected 505a4, got 431846
DHCPv6 client: send REQUEST
transaction_id: a636c7
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
expected a636c7, got 431846
DHCPv6 client: send REQUEST
transaction_id: 7fbe86
DHCPv6 client: received REPLY
DHCPv6 client: packet too small or transaction ID wrong
expected 7fbe86, got c04cee

```

miri64, 2020-04-08:
> > Does not make sense since `REQUEST` and `RENEW` have the same response type (`REPLY`).
> 
> So they should be the same transaction?

What does this even mean? A transaction is a pair of query/response messages. If they are different transactions they must have different transaction IDs. Since `REQUEST` and `RENEW` are distinct query messages, they can't be the same transaction.

benpicco, 2020-04-08:
Yea I checked with Wireshark and I get the right transaction ID in the REPLY, so the client must read an old buffer.

benpicco, 2020-04-08:
Yes this also works:

```patch
--- a/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c
+++ b/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c
@@ -673,6 +687,14 @@ static bool _parse_reply(uint8_t *rep, size_t len)
     return true;
 }
 
+static void _udp_flush(sock_udp_t *sock)
+{
+    int res;
+    while ((res = sock_udp_recv(sock, recv_buf, sizeof(recv_buf), 0, NULL)) > 0) {
+        printf("discarding %d bytes\n", res);
+    }
+}
+
 static void _solicit_servers(event_t *event)
 {
     dhcpv6_msg_t *msg = (dhcpv6_msg_t *)&send_buf[0];
@@ -695,6 +717,9 @@ static void _solicit_servers(event_t *event)
     msg_len += _compose_oro_opt((dhcpv6_opt_oro_t *)&send_buf[msg_len], oro_opts,
                                 ARRAY_SIZE(oro_opts));
     msg_len += _add_ia_pd_from_config(&send_buf[msg_len]);
+
+    _udp_flush(&sock);
+
     DEBUG("DHCPv6 client: send SOLICIT\n");
     res = sock_udp_send(&sock, send_buf, msg_len, &remote);
     assert(res > 0);    /* something went terribly wrong */
@@ -802,6 +827,9 @@ static void _request_renew_rebind(uint8_t type)
     msg_len += _compose_oro_opt((dhcpv6_opt_oro_t *)&send_buf[msg_len], oro_opts,
                                 ARRAY_SIZE(oro_opts));
     msg_len += _add_ia_pd_from_config(&send_buf[msg_len]);
+
+    _udp_flush(&sock);
+
     while (sock_udp_send(&sock, send_buf, msg_len, &remote) <= 0) {}
     while (((res = sock_udp_recv(&sock, recv_buf, sizeof(recv_buf),
                                  retrans_timeout, NULL)) <= 0) ||

```

Will print:

```
DHCPv6 client: send REQUEST
transaction_id: 9ad668
DHCPv6 client: resend REQUEST
DHCPv6 client: received REPLY
DHCPv6 client: scheduling RENEW in 10 sec
DHCPv6 client: scheduling REBIND in 20 sec
DHCPv6 client: send RENEW
transaction_id: 3d8a4f
discarding 120 bytes
discarding 120 bytes
discarding 120 bytes
DHCPv6 client: received REPLY
```

commit 2d116042 (2020-06-25): sys/net/dhcpv6: discard stale replies

The DHCPv6 server might send reponses multiple times.
The DHCPv6 client will only handle the first response, if additional
responses are comming in they are left in the RX queue.

That results in the client always reading the response of a previous
transaction on any subsequent transactions.
In this case the client will try again, creating a new transaction - that
will again only read the previous response.

To fix this, discard previous responses by flushing the RX queue before
sending a new message to the DHCPv6 server.

fixes #13834
closed 2020-06-25 by miri64