RIOT-OS/RIOT #13838: Global Address remains in state TNT[3] forever

benpicco, 2020-04-08:
#### Description
Now this is a bug I've seen sporadically where a node will get a prefix from the border router, but the address remains in the `TNT[3]` state forever.

I've recorded an exchange where the `openmote-b` would not get a valid prefix, but another node would: 
[openmote_tnt3.pcapng.zip](https://github.com/RIOT-OS/RIOT/files/4450972/openmote_tnt3.pcapng.zip)

#### Steps to reproduce the issue
Run a border router with `DHCPV6=1`

Run a node with the `gnrc_networking` example.

Still trying to narrow this down.

#### Expected results
The global address becomes valid.

#### Actual results

```
2020-04-08 16:13:18,265 # Iface  7  HWaddr: 3A:A6  Channel: 0  Page: 2  NID: 0x23
2020-04-08 16:13:18,267 #           Long HWaddr: 22:68:31:23:59:F5:D2:3A 
2020-04-08 16:13:18,269 #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2020-04-08 16:13:18,282 #           AUTOACK  ACK_REQ  CSMA  L2-PDU:102 MTU:1280  HL:64  RTR  
2020-04-08 16:13:18,283 #           RTR_ADV  6LO  IPHC  
2020-04-08 16:13:18,284 #           Source address length: 8
2020-04-08 16:13:18,284 #           Link type: wireless
2020-04-08 16:13:18,286 #           inet6 addr: fe80::2068:3123:59f5:d23a  scope: link  VAL
2020-04-08 16:13:18,299 #           inet6 addr: 2001:db8:0:5:2068:3123:59f5:d23a  scope: global  TNT[3]
2020-04-08 16:13:18,300 #           inet6 group: ff02::2
2020-04-08 16:13:18,301 #           inet6 group: ff02::1
2020-04-08 16:13:18,302 #           inet6 group: ff02::1:fff5:d23a
2020-04-08 16:13:18,313 #           
2020-04-08 16:13:18,314 #           Statistics for Layer 2
2020-04-08 16:13:18,314 #             RX packets 9  bytes 624
2020-04-08 16:13:18,315 #             TX packets 7 (Multicast: 2)  bytes 534
2020-04-08 16:13:18,316 #             TX succeeded 7 errors 0
2020-04-08 16:13:18,329 #           Statistics for IPv6
2020-04-08 16:13:18,330 #             RX packets 3  bytes 352
2020-04-08 16:13:18,332 #             TX packets 7 (Multicast: 2)  bytes 592
2020-04-08 16:13:18,333 #             TX succeeded 7 errors 0
```

#### Versions
Riot master

miri64, 2020-04-08:
Do you have a PCAP for that?

benpicco, 2020-04-08:
It's right in the description: [openmote_tnt3.pcapng.zip](https://github.com/RIOT-OS/RIOT/files/4450972/openmote_tnt3.pcapng.zip)

miri64, 2020-04-08:
> It's right in the description: [openmote_tnt3.pcapng.zip](https://github.com/RIOT-OS/RIOT/files/4450972/openmote_tnt3.pcapng.zip)

Oh sorry, somehow I missed it.

miri64, 2020-04-08:
From the PCAP it looks like the `openmode-b` can't handle the fragmented router advertisement and thus is never able to validate the global address using the NS/NA handshake with ARO option. It just keeps resending RS. Can you try pinging the `openmode-b`'s link local address from one of the other nodes with payloads >100? Maybe its radio has problems with handling 6LoWPAN fragmented packets. Or maybe the fragments are too large for the receive buffer of the radio. See also https://github.com/RIOT-OS/RIOT/issues/4822 for reference.

benpicco, 2020-04-09:
The `openmode-b` handles fragmented packets just fine - it also gets a global (valid) prefix - except when it doesn't.

What definitely seems to trigger the issue is having two 6LoWPAN interfaces (sub-GHz & 2.4 GHz) on the border router, even when the clients are sub-GHz only.

miri64, 2020-04-09:
In that case: could you try to reproduce this with `ENABLE_DEBUG` activated in

- `sys/net/gnrc/network_layer/ipv6/nib/nib.c`,
- `sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c`, and
- `sys/net/gnrc/network_layer/ipv6/nib/_nib-slaac.c`

please. `_handle_rereg_address()` does not seem to be executed correctly on that node.

benpicco, 2020-04-09:
Ok!
This generates quite some output.

<details>
<summary>openmote-b</summary>

```
2020-04-09 10:49:58,945 # nib: Handle timer event (ctx = 0x20001952, type = 0x4fcf, now = 1125ms)
2020-04-09 10:49:58,948 # nib: Re-registering 2001:db8:0:c:2068:3123:59f5:d23a with upstream router fe80::18f9:8f5f:305d:58e
2020-04-09 10:49:58,950 # nib: get next hop link-layer address of fe80::18f9:8f5f:305d:58e%7
2020-04-09 10:49:58,952 # nib: fe80::18f9:8f5f:305d:58e is on-link or in NC, start address resolution
2020-04-09 10:49:58,953 # nib: resolve address fe80::18f9:8f5f:305d:58e%7 from neighbor cache
2020-04-09 10:49:59,860 # nib: Handle timer event (ctx = 0x20001952, type = 0x4fcf, now = 2139ms)
2020-04-09 10:49:59,863 # nib: Re-registering 2001:db8:0:c:2068:3123:59f5:d23a with upstream router fe80::18f9:8f5f:305d:58e
2020-04-09 10:49:59,865 # nib: get next hop link-layer address of fe80::18f9:8f5f:305d:58e%7
2020-04-09 10:49:59,877 # nib: fe80::18f9:8f5f:305d:58e is on-link or in NC, start address resolution
2020-04-09 10:49:59,878 # nib: resolve address fe80::18f9:8f5f:305d:58e%7 from neighbor cache
2020-04-09 10:50:00,005 # nib: Handle packet (icmpv6->type = 134)
2020-04-09 10:50:00,006 # nib: Received valid router advertisement:
2020-04-09 10:50:00,008 #      - Source address: fe80::18f9:8f5f:305d:58e
2020-04-09 10:50:00,020 #      - Destination address: fe80::2068:3123:59f5:d23a
2020-04-09 10:50:00,020 #      - Cur Hop Limit: 0
2020-04-09 10:50:00,021 #      - Flags: --
2020-04-09 10:50:00,021 #      - Router Lifetime: 1800s
2020-04-09 10:50:00,021 #      - Reachable Time: 0ms
2020-04-09 10:50:00,022 #      - Retrans Timer: 0ms
2020-04-09 10:50:00,036 # nib: Re-registering 2001:db8:0:c:2068:3123:59f5:d23a with upstream router fe80::18f9:8f5f:305d:58e
2020-04-09 10:50:00,038 # nib: received valid Prefix Information option:
2020-04-09 10:50:00,039 #      - Prefix: 2001:db8:0:c::/64
2020-04-09 10:50:00,039 #      - Flags: -A
2020-04-09 10:50:00,040 #      - Valid lifetime: 39907
2020-04-09 10:50:00,052 #      - Preferred lifetime: 29907
2020-04-09 10:50:00,054 # nib: add address based on 2001:db8:0:c::/64 automatically to interface 7
2020-04-09 10:50:00,055 # nib: get next hop link-layer address of fe80::18f9:8f5f:305d:58e%7
2020-04-09 10:50:00,069 # nib: fe80::18f9:8f5f:305d:58e is on-link or in NC, start address resolution
2020-04-09 10:50:00,071 # nib: resolve address fe80::18f9:8f5f:305d:58e%7 from neighbor cache
2020-04-09 10:50:02,853 # nib: Handle timer event (ctx = 0x20001308, type = 0x4fc8, now = 5136ms)
2020-04-09 10:50:02,856 # nib: get next hop link-layer address of fe80::18f9:8f5f:305d:58e%7
2020-04-09 10:50:02,868 # nib: fe80::18f9:8f5f:305d:58e is on-link or in NC, start address resolution
2020-04-09 10:50:02,871 # nib: resolve address fe80::18f9:8f5f:305d:58e%7 from neighbor cache
2020-04-09 10:50:02,932 # nib: Handle packet (icmpv6->type = 136)
2020-04-09 10:50:02,934 # nib: Received valid neighbor advertisement:
2020-04-09 10:50:02,935 #      - Target address: fe80::18f9:8f5f:305d:58e
2020-04-09 10:50:02,937 #      - Source address: fe80::18f9:8f5f:305d:58e
2020-04-09 10:50:02,949 #      - Destination address: fe80::2068:3123:59f5:d23a
2020-04-09 10:50:02,949 #      - Flags: RS-
2020-04-09 10:50:10,692 # nib: Handle packet (icmpv6->type = 133)
2020-04-09 10:50:10,694 # nib: Received valid router solicitation:
2020-04-09 10:50:10,708 #      - Source address: fe80::3452:9467:e14:1b8e
2020-04-09 10:50:10,709 #      - Destination address: ff02::2
2020-04-09 10:50:10,709 # nib: Handle packet (icmpv6->type = 133)
2020-04-09 10:50:10,710 # nib: Received valid router solicitation:
2020-04-09 10:50:10,723 #      - Source address: fe80::3452:9467:e14:1b8e
2020-04-09 10:50:10,725 #      - Destination address: ff02::2
2020-04-09 10:50:10,727 # nib: get next hop link-layer address of fe80::3452:9467:e14:1b8e%7
2020-04-09 10:50:10,739 # nib: fe80::3452:9467:e14:1b8e is on-link or in NC, start address resolution
2020-04-09 10:50:10,742 # nib: resolve address fe80::3452:9467:e14:1b8e%7 by reverse translating to 36:52:94:67:0E:14:1B:8E
2020-04-09 10:50:10,757 # nib: get next hop link-layer address of fe80::3452:9467:e14:1b8e%7
2020-04-09 10:50:10,759 # nib: fe80::3452:9467:e14:1b8e is on-link or in NC, start address resolution
2020-04-09 10:50:10,772 # nib: resolve address fe80::3452:9467:e14:1b8e%7 by reverse translating to 36:52:94:67:0E:14:1B:8E
2020-04-09 10:50:11,796 # nib: Handle packet (icmpv6->type = 133)
2020-04-09 10:50:11,798 # nib: Received valid router solicitation:
2020-04-09 10:50:11,812 #      - Source address: fe80::3452:9467:e14:1b8e
2020-04-09 10:50:11,813 #      - Destination address: ff02::2
2020-04-09 10:50:11,814 # nib: get next hop link-layer address of fe80::3452:9467:e14:1b8e%7
2020-04-09 10:50:11,828 # nib: fe80::3452:9467:e14:1b8e is on-link or in NC, start address resolution
2020-04-09 10:50:11,830 # nib: resolve address fe80::3452:9467:e14:1b8e%7 by reverse translating to 36:52:94:67:0E:14:1B:8E
ifconfig
2020-04-09 10:50:15,555 # ifconfig
2020-04-09 10:50:15,557 # Iface  7  HWaddr: 3A:A6  Channel: 0  Page: 2  NID: 0x23
[... 440 lines cut ...]
nib: host unreachable
nib: get next hop link-layer address of fe80::3452:9467:e14:1b8e%6
nib: fe80::3452:9467:e14:1b8e is on-link or in NC, start address resolution
nib: resolve address fe80::3452:9467:e14:1b8e%6 by reverse translating to 36:52:94:67:0E:14:1B:8E
nib: Handle packet (icmpv6->type = 135)
[... cut ...]
 - EUI-64: 36:52:94:67:0e:14:1b:8e
nib: get next hop link-layer address of 2001:db8:0:c:3452:9467:e14:1b8e%6
nib: 2001:db8:0:c:3452:9467:e14:1b8e is on-link or in NC, start address resolution
nib: resolve address 2001:db8:0:c:3452:9467:e14:1b8e by probing neighbors
nib: host unreachable
nib: Handle timer event (ctx = 0x20002574, type = 0x4fc1, now = 206266ms)
nib: Handle timer event (ctx = 0x20002574, type = 0x4fc1, now = 209257ms)
nib: Handle timer event (ctx = 0x20002574, type = 0x4fc1, now = 217592ms)
```
</details>

miri64, 2020-04-09:
@benpicco any news since our offline afternoon debugging session?

benpicco, 2020-04-15:
Not really, but I also see this behavior with [just one interface](https://github.com/RIOT-OS/RIOT/files/4482362/tracker_tnt3.pcapng.zip) now.

miri64, 2020-04-15:
The PCAPs alone don't help me. What do the logs say? Is there a way to reproduce it myself?

benpicco, 2020-04-15:
Yea I know, but I managed to narrow it down a bit:
I have a node that restarts every 10s for an unrelated test. (10s wake, then 10s sleep etc.)

During the first few cycles everything is fine and the global address becomes valid.
However, after a few cycles the address of the node stays `TNT[3]`. This can be fixed by restarting the border router.

So… does that help as to which debug outputs I should enable?

miri64, 2020-04-15:
> So… does that help as to which debug outputs I should enable?

In general, the same, as in our afternoon session. Since this does not seem to be timing related, the more the better ;-).

> During the first few cycles everything is fine and the global address becomes valid.
However, after a few cycles the address of the node stays TNT[3]. This can be fixed by restarting the border router.

Mhhhh does the address of the node change?

benpicco, 2020-04-15:
> Mhhhh ~does the address~ do any of the addresses of the node change?

```
2020-04-15 18:57:27,092 #           Long HWaddr: 36:52:94:67:0E:14:1B:8E 
2020-04-15 18:57:27,119 #           inet6 addr: fe80::3452:9467:e14:1b8e  scope: link  VAL
2020-04-15 18:57:27,124 #           inet6 addr: 2001:db8::3452:9467:e14:1b8e  scope: global  TNT[3]
```

```
2020-04-15 18:57:50,661 #           Long HWaddr: 36:52:94:67:0E:14:1B:8E 
2020-04-15 18:57:50,687 #           inet6 addr: fe80::3452:9467:e14:1b8e  scope: link  VAL
2020-04-15 18:57:50,693 #           inet6 addr: 2001:db8::3452:9467:e14:1b8e  scope: global  VAL
```

```
2020-04-15 19:45:39,498 #           Long HWaddr: 36:52:94:67:0E:14:1B:8E 
2020-04-15 19:45:39,523 #           inet6 addr: fe80::3452:9467:e14:1b8e  scope: link  VAL
2020-04-15 19:45:39,530 #           inet6 addr: 2001:db8::3452:9467:e14:1b8e  scope: global  VAL
```

```
2020-04-15 19:50:11,623 #           Long HWaddr: 36:52:94:67:0E:14:1B:8E 
2020-04-15 19:50:11,649 #           inet6 addr: fe80::3452:9467:e14:1b8e  scope: link  VAL
2020-04-15 19:50:11,656 #           inet6 addr: 2001:db8::3452:9467:e14:1b8e  scope: global  TNT[3]
```

Hmm but now I left the node sit there and didn't change anything while trying to build a minimal exmaple for `samr21-xpro` (doesn't help that wake from standby seems to be broken there now 😩) and until I started typing this it would always get a valid address…

On the border router it's

```
2001:db8::3452:9467:e14:1b8e dev #6 lladdr 36:52:94:67:0E:14:1B:8E  UNREACHABLE REGISTERED
```
 when in the `TNT[3]` state and

```
2001:db8::3452:9467:e14:1b8e dev #6 lladdr 36:52:94:67:0E:14:1B:8E  STALE REGISTERED
```

when valid.

benpicco, 2020-04-15:
I now have a way to reliably reproduce this on `samr21-xpro`:

```patch
diff --git a/examples/gnrc_networking/Makefile b/examples/gnrc_networking/Makefile
index 0ae56343f9..09e03aa305 100644
--- a/examples/gnrc_networking/Makefile
+++ b/examples/gnrc_networking/Makefile
@@ -14,7 +16,7 @@ USEMODULE += auto_init_gnrc_netif
 # Activate ICMPv6 error messages
 USEMODULE += gnrc_icmpv6_error
 # Specify the mandatory networking modules for IPv6 and UDP
-USEMODULE += gnrc_ipv6_router_default
+USEMODULE += gnrc_ipv6_default # this seems to be the key
 USEMODULE += gnrc_udp
 # Add a routing protocol
 USEMODULE += gnrc_rpl
@@ -31,6 +33,8 @@ USEMODULE += netstats_l2
 USEMODULE += netstats_ipv6
 USEMODULE += netstats_rpl
 
+FEATURES_REQUIRED += periph_rtc
+
 # Comment this out to disable code in RIOT that does safety checking
 # which is not needed in a production environment but helps in the
 # development process:
diff --git a/examples/gnrc_networking/main.c b/examples/gnrc_networking/main.c
index 6301f4291d..a91f488c9f 100644
--- a/examples/gnrc_networking/main.c
+++ b/examples/gnrc_networking/main.c
@@ -23,6 +23,11 @@
 #include "shell.h"
 #include "msg.h"
 
+#include "net/gnrc/netif.h"
+
+#include "periph/pm.h"
+#include "periph/rtc.h"
+
 #define MAIN_QUEUE_SIZE     (8)
 static msg_t _main_msg_queue[MAIN_QUEUE_SIZE];
 
@@ -33,6 +38,25 @@ static const shell_command_t shell_commands[] = {
     { NULL, NULL, NULL }
 };
 
+static char hibernator_stack[THREAD_STACKSIZE_DEFAULT];
+static void* hibernator(void *ctx)
+{
+    puts("will sleep in 10s");
+    xtimer_sleep(10);
+
+    puts("will sleep");
+
+    struct tm alarm;
+    rtc_get_time(&alarm);
+    alarm.tm_sec += 10;
+    rtc_set_alarm(&alarm, (rtc_alarm_cb_t)puts, "woke up");
+    pm_set(0);
+
+    pm_reboot();
+
+    return ctx;
+}
+
 int main(void)
 {
     /* we need a message queue for the thread running the shell in order to
@@ -40,6 +64,9 @@ int main(void)
     msg_init_queue(_main_msg_queue, MAIN_QUEUE_SIZE);
     puts("RIOT network stack example application");
 
+    thread_create(hibernator_stack, sizeof(hibernator_stack), 10, 0,
+                  hibernator, NULL, "hibernator");
+
     /* start shell */
     puts("All up, running the shell now");
     char line_buf[SHELL_DEFAULT_BUFSIZE];
```

After a few cycles I will always get

```
2020-04-15 23:35:14,664 #           inet6 addr: fe80::5806:59c9:2c36:7d78  scope: link  VAL
2020-04-15 23:35:14,670 #           inet6 addr: 2001:db8::5806:59c9:2c36:7d78  scope: global  TNT[3]
```

The key here seems to be that I changed `gnrc_ipv6_router_default` to `gnrc_ipv6_default`.
I did that because at this point my nodes are just nodes, they don't route traffic - but I might have hit some less well exercised code paths there.

miri64, 2020-04-16:
> On the border router it's
> 
> ```
> 2001:db8::3452:9467:e14:1b8e dev #6 lladdr 36:52:94:67:0E:14:1B:8E  UNREACHABLE REGISTERED
> ```
> 
> when in the `TNT[3]` state and
> 
> ```
> 2001:db8::3452:9467:e14:1b8e dev #6 lladdr 36:52:94:67:0E:14:1B:8E  STALE REGISTERED
> ```
> 
> when valid.

Yepp that makes sense, since I saw the BR sending NS to the node.

miri64, 2020-04-16:
(this is actually the cause of the `TNT[3]` as the BR is not able to reply to the NS of the node due to that).

miri64, 2020-04-16:
And yes 6LN instead of 6LR also makes sense since that deactivates the classic (multicast) address resolution state-machine of NDP.

miri64, 2020-04-16:
FYI: I was not able to compile with [your patch](https://github.com/RIOT-OS/RIOT/issues/13838#issuecomment-614293722) due to invalid function pointer casting on my Arch machine

```
Installed compiler toolchains
-----------------------------
               native gcc: gcc (Arch Linux 9.3.0-1) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 9.3.0
```

However, with the following adapted patch it worked:

```diff
diff --git a/examples/gnrc_networking/Makefile b/examples/gnrc_networking/Makefile
index 0ae56343f..30f67cba5 100644
--- a/examples/gnrc_networking/Makefile
+++ b/examples/gnrc_networking/Makefile
@@ -14,7 +14,7 @@ USEMODULE += auto_init_gnrc_netif
 # Activate ICMPv6 error messages
 USEMODULE += gnrc_icmpv6_error
 # Specify the mandatory networking modules for IPv6 and UDP
-USEMODULE += gnrc_ipv6_router_default
+USEMODULE += gnrc_ipv6_default # this seems to be the key
 USEMODULE += gnrc_udp
 # Add a routing protocol
 USEMODULE += gnrc_rpl
@@ -31,6 +31,8 @@ USEMODULE += netstats_l2
 USEMODULE += netstats_ipv6
 USEMODULE += netstats_rpl
 
+FEATURES_REQUIRED += periph_rtc
+
 # Comment this out to disable code in RIOT that does safety checking
 # which is not needed in a production environment but helps in the
 # development process:
diff --git a/examples/gnrc_networking/main.c b/examples/gnrc_networking/main.c
index 6301f4291..aacb27862 100644
--- a/examples/gnrc_networking/main.c
+++ b/examples/gnrc_networking/main.c
@@ -23,6 +23,11 @@
 #include "shell.h"
 #include "msg.h"
 
+#include "net/gnrc/netif.h"
+
+#include "periph/pm.h"
+#include "periph/rtc.h"
+
 #define MAIN_QUEUE_SIZE     (8)
 static msg_t _main_msg_queue[MAIN_QUEUE_SIZE];
 
@@ -33,6 +38,30 @@ static const shell_command_t shell_commands[] = {
     { NULL, NULL, NULL }
 };
 
+void _cb(void *arg)
+{
+    puts(arg);
+}
+
+static char hibernator_stack[THREAD_STACKSIZE_DEFAULT];
+static void* hibernator(void *ctx)
+{
+    puts("will sleep in 10s");
+    xtimer_sleep(10);
+
+    puts("will sleep");
+
+    struct tm alarm;
+    rtc_get_time(&alarm);
+    alarm.tm_sec += 10;
+    rtc_set_alarm(&alarm, _cb, "woke up");
+    pm_set(0);
+
+    pm_reboot();
+
+    return ctx;
+}
+
 int main(void)
 {
     /* we need a message queue for the thread running the shell in order to
@@ -40,6 +69,9 @@ int main(void)
     msg_init_queue(_main_msg_queue, MAIN_QUEUE_SIZE);
     puts("RIOT network stack example application");
 
+    thread_create(hibernator_stack, sizeof(hibernator_stack), 10, 0,
+                  hibernator, NULL, "hibernator");
+
     /* start shell */
     puts("All up, running the shell now");
     char line_buf[SHELL_DEFAULT_BUFSIZE];
```

miri64, 2020-04-16:
How long do I have to run this? I ran this with the `examples/gnrc_border_router` example untouched now for about 10 minutes and there was no occurence of that bug yet :-/

### Terminal 1:
```
$ SERIAL=ATML2127031800004989 make -C examples/gnrc_border_router/ flash -j16
[…]
$ SERIAL=ATML2127031800004989 make -C examples/gnrc_border_router/ term
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_border_router'
make -C ethos
cc -O3 -Wall ethos.c -o ethos
make -C uhcpd
cc -g -O3 -Wall -DUHCP_SERVER -I/home/mlenders/Repositories/RIOT-OS/RIOT/sys/include /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/application_layer/uhcp/uhcp.c uhcpd.c -o bin/uhcpd
sudo sh /home/mlenders/Repositories/RIOT-OS/RIOT/dist/tools/ethos/start_network.sh  /dev/ttyACM0 tap0 2001:db8::/64
[sudo] password for mlenders: 
net.ipv6.conf.tap0.forwarding = 1
net.ipv6.conf.tap0.accept_ra = 0
----> ethos: sending hello.
----> ethos: activating serial pass through.
----> ethos: hello reply received
uhcp_client(): no reply received
uhcp_client(): sending REQ...
got packet from fe80::e0bc:7dff:fecb:f54f port 59671
uhcp: push from fe80::e0bc:7dff:fecb:f54f:59671 prefix=2001:db8::/64
gnrc_uhcpc: uhcp_handle_prefix(): add compression context 0 for prefix 2001:db8::589d:9386:2208:6578/64
gnrc_uhcpc: uhcp_handle_prefix(): configured new prefix 2001:db8::589d:9386:2208:6578/64
uhcp_client(): sending REQ...
got packet from fe80::e0bc:7dff:fecb:f54f port 52737
uhcp: push from fe80::e0bc:7dff:fecb:f54f:52737 prefix=2001:db8::/64
gnrc_uhcpc: uhcp_handle_prefix(): got same prefix again
gnrc_uhcpc: uhcp_handle_prefix(): add compression context 0 for prefix 2001:db8::589d:9386:2208:6578/64
[…]
uhcp_client(): sending REQ...
got packet from fe80::e0bc:7dff:fecb:f54f port 33144
uhcp: push from fe80::e0bc:7dff:fecb:f54f:33144 prefix=2001:db8::/64
gnrc_uhcpc: uhcp_handle_prefix(): got same prefix again
gnrc_uhcpc: uhcp_handle_prefix(): add compression context 0 for prefix 2001:db8::589d:9386:2208:6578/64
ifconfig
ifconfig
Iface  7  HWaddr: 02:79:27:EA:94:E7 
          L2-PDU:1500 MTU:1500  HL:64  RTR  
          Source address length: 6
          Link type: wired
          inet6 addr: fe80::79:27ff:feea:94e7  scope: link  VAL
          inet6 addr: fe80::2  scope: link  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ffea:94e7
          inet6 group: ff02::1:ff00:2
          
Iface  6  HWaddr: 2D:4B  Channel: 26  Page: 0  NID: 0x23
          Long HWaddr: 5A:9D:93:86:22:08:65:78 
           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
          AUTOACK  ACK_REQ  CSMA  L2-PDU:102 MTU:1280  HL:64  RTR  
          RTR_ADV  6LO  IPHC  
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::589d:9386:2208:6578  scope: link  VAL
          inet6 addr: 2001:db8::589d:9386:2208:6578  scope: global  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff08:6578
          
> nib neigh
nib neigh
fe80::1 dev #7 lladdr E2:BC:7D:CB:F5:4F  REACHABLE GC
fe80::e0bc:7dff:fecb:f54f dev #7 lladdr E2:BC:7D:CB:F5:4F  STALE GC
2001:db8::840c:77ec:2333:6278 dev #6 lladdr 86:0C:77:EC:23:33:62:78  STALE REGISTERED
```

### Terminal 2
[... 44 lines cut ...]
ifconfig
2020-04-16 12:18:21,679 # ifconfig
2020-04-16 12:18:21,684 # Iface  7  HWaddr: 10:8B  Channel: 26  Page: 0  NID: 0x23
2020-04-16 12:18:21,694 #           Long HWaddr: 86:0C:77:EC:23:33:62:78 
2020-04-16 12:18:21,695 #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2020-04-16 12:18:21,702 #           AUTOACK  ACK_REQ  CSMA  L2-PDU:102 MTU:1280  HL:64  6LO  
2020-04-16 12:18:21,704 #           IPHC  
2020-04-16 12:18:21,707 #           Source address length: 8
2020-04-16 12:18:21,710 #           Link type: wireless
2020-04-16 12:18:21,716 #           inet6 addr: fe80::840c:77ec:2333:6278  scope: link  VAL
2020-04-16 12:18:21,722 #           inet6 addr: 2001:db8::840c:77ec:2333:6278  scope: global  VAL
2020-04-16 12:18:21,725 #           inet6 group: ff02::1
2020-04-16 12:18:21,726 #           
2020-04-16 12:18:21,729 #           Statistics for Layer 2
2020-04-16 12:18:21,732 #             RX packets 3  bytes 222
2020-04-16 12:18:21,737 #             TX packets 2 (Multicast: 1)  bytes 123
2020-04-16 12:18:21,740 #             TX succeeded 2 errors 0
2020-04-16 12:18:21,742 #           Statistics for IPv6
2020-04-16 12:18:21,746 #             RX packets 2  bytes 224
2020-04-16 12:18:21,750 #             TX packets 2 (Multicast: 1)  bytes 160
2020-04-16 12:18:21,754 #             TX succeeded 2 errors 0
2020-04-16 12:18:21,754 # 
```

Here is the [wireshark capture](https://box.fu-berlin.de/s/Q7fFZP4ygxecGKA)

miri64, 2020-04-16:
@benpicco what I can imagine, is that NUD fails within the period the node is sleeping and the NCE becomes UNREACHABLE and not becoming reachable for some reason again. However, I do remember we had a patch for that. Have you tried it with that?

```diff
diff --git a/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c b/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c
index 6e3686ed9..dfe9d544a 100644
--- a/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c
+++ b/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c
@@ -111,7 +111,8 @@ void _handle_sl2ao(gnrc_netif_t *netif, const ipv6_hdr_t *ipv6,
         _set_nud_state(netif, nce, GNRC_IPV6_NIB_NC_INFO_NUD_STATE_STALE);
     }
 #endif  /* CONFIG_GNRC_IPV6_NIB_ARSM */
-    if ((nce == NULL) || !(nce->mode & _NC)) {
+    if ((nce == NULL) || !(nce->mode & _NC) ||
+        (_get_nud_state(nce) == GNRC_IPV6_NIB_NC_INFO_NUD_STATE_UNREACHABLE)) {
         DEBUG("nib: Creating NCE for (ipv6 = %s, iface = %u, nud_state = STALE)\n",
               ipv6_addr_to_str(addr_str, &ipv6->src, sizeof(addr_str)),
               netif->pid);
```

miri64, 2020-04-16:
I ran the same setup with an DHCPv6 border router now with for about an hour and still was not able to reproduce. Please try it with my patch!

benpicco, 2020-04-16:
Now on the `openmote-b` and with a BR that uses two interfaces as well, I always see `scope: global  TNT[3]`. (no reboots required and with `gnrc_ipv6_router_default`)

Now this might be an entirely different bug because to the openmote-b it will appear as if there are two border routers - the subGhz and 2.4 GHz interface have different addresses after all.

Maybe it's still useful:

<details>
<summary>border router</summary>

```
RIOT border router example application
All up, running the shell now
> ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to 70 (network byteorder 4600)
ipv6: set next header to 17
ipv6: set packet source to fe80::2
ipv6: write protect up to payload to calculate checksum
ipv6: calculate checksum for upper header.
ipv6: send multicast over interface 8
ipv6: Sending (src = fe80::2, dst = ff02::1:2, next header = 17, length = 70)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::16, next header = 0, length = 96)
ipv6: packet destination not this host
ipv6: decrement hop limit to 0
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: NIB timer event received
nib: Handle timer event (ctx = 0x2000172c, type = 0x4fc3, now = 813ms)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to 16 (network byteorder 1000)
ipv6: set next header to 58
ipv6: write protect up to payload to calculate checksum
ipv6: calculate checksum for upper header.
ipv6: send multicast over interface 8
ipv6: Sending (src = fe80::2, dst = ff02::2, next header = 58, length = 16)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::1:ff7b:9784, next header = 58, length = 32)
ipv6: packet destination not this host
ipv6: decrement hop limit to 254
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: NIB timer event received
nib: Handle timer event (ctx = 0x20001743, type = 0x4fd2, now = 1044ms)
nib: validating address fe80::7d:2dff:fe57:2812 (idx: 0, netif: 8)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::16, next header = 0, length = 96)
ipv6: packet destination not this host
ipv6: decrement hop limit to 0
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::1:ff00:1, next header = 58, length = 32)
ipv6: packet destination not this host
ipv6: decrement hop limit to 254
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to 70 (network byteorder 4600)
ipv6: set next header to 17
ipv6: set packet source to fe80::7d:2dff:fe57:2812
ipv6: write protect up to payload to calculate checksum
ipv6: calculate checksum for upper header.
ipv6: send multicast over interface 8
ipv6: Sending (src = fe80::7d:2dff:fe57:2812, dst = ff02::1:2, next header = 17, length = 70)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
[... 1081 lines cut ...]
2020-04-16 18:58:39,321 # at86rf215: RX on interface 7
2020-04-16 18:58:39,336 # ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
2020-04-16 18:58:39,339 # ipv6: Received (src = fe80::18f9:8f5f:305d:58c, dst = fe80::2068:3123:59f5:d238, next header = 58, length = 104)
2020-04-16 18:58:39,340 # ipv6: forward nh = 58 to other threads
2020-04-16 18:58:39,352 # ipv6: handle ICMPv6 packet (nh = 58)
2020-04-16 18:58:39,353 # nib: Handle packet (icmpv6->type = 134)
2020-04-16 18:58:39,354 # nib: Received valid router advertisement:
2020-04-16 18:58:39,355 #      - Source address: fe80::18f9:8f5f:305d:58c
2020-04-16 18:58:39,368 #      - Destination address: fe80::2068:3123:59f5:d238
2020-04-16 18:58:39,369 #      - Cur Hop Limit: 0
2020-04-16 18:58:39,369 #      - Flags: --
2020-04-16 18:58:39,370 #      - Router Lifetime: 1800s
2020-04-16 18:58:39,371 #      - Reachable Time: 0ms
2020-04-16 18:58:39,372 #      - Retrans Timer: 0ms
2020-04-16 18:58:39,384 # nib: default router list is full. Ignoring RA from fe80::18f9:8f5f:305d:58c
2020-04-16 18:58:39,385 # ipv6: waiting for incoming message.
2020-04-16 18:58:41,992 # ipv6: NIB timer event received
2020-04-16 18:58:41,994 # nib: Handle timer event (ctx = 0x20001580, type = 0x4fc7, now = 36976ms)
2020-04-16 18:58:41,995 # nib: Timeout reachability
2020-04-16 18:58:41,996 # ipv6: waiting for incoming message.
nib neigh
2020-04-16 18:58:46,279 # nib neigh
2020-04-16 18:58:46,295 # fe80::18f9:8f5f:305d:58e dev #6 lladdr 1A:F9:8F:5F:30:5D:05:8E router STALE GC
```
</details>

miri64, 2020-04-16:
Is that with or without my patch?

miri64, 2020-04-16:
Apparently with (answered by @benpicco offline)...

miri64, 2020-04-16:
Can you send me the same log of the BR with the following patch, please?

```diff
diff --git a/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c b/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c
index 6e3686ed97..f509e7b193 100644
--- a/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c
+++ b/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c
@@ -99,6 +99,10 @@ void _handle_sl2ao(gnrc_netif_t *netif, const ipv6_hdr_t *ipv6,
         DEBUG("nib: Unexpected SL2AO length. Ignoring SL2AO\n");
         return;
     }
+    DEBUG("nib: Handling SL2AO with initial NCE state (nce = %p, NC? %s, "
+          "NUD state = %u)\n", (void *)nce,
+          (nce && (nce->mode & _NC)) ? "yes" : "no",
+          (nce) ? _get_nud_state(nce) : 0U);
 #if IS_ACTIVE(CONFIG_GNRC_IPV6_NIB_ARSM)
     if ((nce != NULL) && (nce->mode & _NC) &&
         ((nce->l2addr_len != l2addr_len) ||
```

benpicco, 2020-04-16:
<details>
<summary>border router log</summary>

```
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::16, next header = 0, length = 96)
ipv6: packet destination not this host
ipv6: decrement hop limit to 0
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::1:ff00:1, next header = 58, length = 32)
ipv6: packet destination not this host
ipv6: decrement hop limit to 254
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::16, next header = 0, length = 96)
ipv6: packet destination not this host
ipv6: decrement hop limit to 0
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::1:ffc4:dcda, next header = 58, length = 32)
ipv6: packet destination not this host
ipv6: decrement hop limit to 254
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: NIB timer event received
nib: Handle timer event (ctx = 0x2000172c, type = 0x4fc3, now = 5074ms)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to 16 (network byteorder 1000)
ipv6: set next header to 58
ipv6: write protect up to payload to calculate checksum
ipv6: calculate checksum for upper header.
ipv6: send multicast over interface 8
ipv6: Sending (src = fe80::2d:fff:fe6c:b3e2, dst = ff02::2, next header = 58, length = 16)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = fe80::1, dst = ff02::16, next header = 0, length = 116)
ipv6: packet destination not this host
ipv6: decrement hop limit to 0
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = fe80::1, dst = ff02::16, next header = 0, length = 36)
ipv6: packet destination not this host
ipv6: decrement hop limit to 0
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = fe80::1, dst = ff02::fb, next header = 17, length = 170)
ipv6: packet destination not this host
ipv6: decrement hop limit to 254
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = fe80::1, dst = ff02::fb, next header = 17, length = 72)
ipv6: packet destination not this host
ipv6: decrement hop limit to 254
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = fe80::1, dst = ff02::fb, next header = 17, length = 284)
ipv6: packet destination not this host
ipv6: decrement hop limit to 254
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
[... 773 lines cut ...]
          inet6 group: ff02::1
          inet6 group: ff02::1:ff5d:58c
          
Iface  6  HWaddr: 01:57  Channel: 0  Page: 2  NID: 0x23
          Long HWaddr: 1A:F9:8F:5F:30:5D:05:8E 
           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
          AUTOACK  ACK_REQ  CSMA  L2-PDU:102 MTU:1280  HL:64  RTR  
          RTR_ADV  6LO  IPHC  
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::18f9:8f5f:305d:58e  scope: link  VAL
          inet6 addr: 2001:db8:0:1a:18f9:8f5f:305d:58e  scope: global  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff5d:58e
          
> ipv6: NIB timer event received
nib: Handle timer event (ctx = 0x200026a8, type = 0x4fc7, now = 58306ms)
nib: Timeout reachability
ipv6: waiting for incoming message.
nib abr
nib abr
2001:db8:0:19:18f9:8f5f:305d:58c v2 expires 10000min
```
</details>

miri64, 2020-04-16:
Let's try another patch:

```diff
diff --git a/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c b/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c
index 6e3686ed97..e27af4ad43 100644
--- a/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c
+++ b/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c
@@ -99,6 +99,10 @@ void _handle_sl2ao(gnrc_netif_t *netif, const ipv6_hdr_t *ipv6,
         DEBUG("nib: Unexpected SL2AO length. Ignoring SL2AO\n");
         return;
     }
+    DEBUG("nib: Handling SL2AO with initial NCE state (nce = %p, NC? %s, "
+          "NUD state = %u)\n", (void *)nce,
+          (nce && (nce->mode & _NC)) ? "yes" : "no",
+          (nce) ? _get_nud_state(nce) : 0U);
 #if IS_ACTIVE(CONFIG_GNRC_IPV6_NIB_ARSM)
     if ((nce != NULL) && (nce->mode & _NC) &&
         ((nce->l2addr_len != l2addr_len) ||
@@ -111,7 +115,10 @@ void _handle_sl2ao(gnrc_netif_t *netif, const ipv6_hdr_t *ipv6,
         _set_nud_state(netif, nce, GNRC_IPV6_NIB_NC_INFO_NUD_STATE_STALE);
     }
 #endif  /* CONFIG_GNRC_IPV6_NIB_ARSM */
-    if ((nce == NULL) || !(nce->mode & _NC)) {
+    if ((nce == NULL) || !(nce->mode & _NC) ||
+        (_get_nud_state(nce) != GNRC_IPV6_NIB_NC_INFO_NUD_STATE_REACHABLE) ||
+        (_get_nud_state(nce) != GNRC_IPV6_NIB_NC_INFO_NUD_STATE_INCOMPLETE) ||
+        (_get_nud_state(nce) != GNRC_IPV6_NIB_NC_INFO_NUD_STATE_UNMANAGED)) {
         DEBUG("nib: Creating NCE for (ipv6 = %s, iface = %u, nud_state = STALE)\n",
               ipv6_addr_to_str(addr_str, &ipv6->src, sizeof(addr_str)),
               netif->pid);
```

benpicco, 2020-04-16:
<details>
<summary>border router log</summary>

```
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::16, next header = 0, length = 96)
ipv6: packet destination not this host
ipv6: decrement hop limit to 0
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: NIB timer event received
nib: Handle timer event (ctx = 0x2000172c, type = 0x4fc3, now = 360ms)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to 16 (network byteorder 1000)
ipv6: set next header to 58
ipv6: write protect up to payload to calculate checksum
ipv6: calculate checksum for upper header.
ipv6: send multicast over interface 8
ipv6: Sending (src = fe80::2, dst = ff02::2, next header = 58, length = 16)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to 70 (network byteorder 4600)
ipv6: set next header to 17
ipv6: set packet source to fe80::2
ipv6: write protect up to payload to calculate checksum
ipv6: calculate checksum for upper header.
ipv6: send multicast over interface 8
ipv6: Sending (src = fe80::2, dst = ff02::1:2, next header = 17, length = 70)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::1:ff00:1, next header = 58, length = 32)
ipv6: packet destination not this host
ipv6: decrement hop limit to 254
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::16, next header = 0, length = 96)
ipv6: packet destination not this host
ipv6: decrement hop limit to 0
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = ::, dst = ff02::1:ffe7:60a2, next header = 58, length = 32)
ipv6: packet destination not this host
ipv6: decrement hop limit to 254
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: NIB timer event received
nib: Handle timer event (ctx = 0x20001743, type = 0x4fd2, now = 1044ms)
nib: validating address fe80::b0:1ff:fe36:f411 (idx: 0, netif: 8)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: set payload length to 70 (network byteorder 4600)
ipv6: set next header to 17
ipv6: set packet source to fe80::b0:1ff:fe36:f411
ipv6: write protect up to payload to calculate checksum
ipv6: calculate checksum for upper header.
ipv6: send multicast over interface 8
ipv6: Sending (src = fe80::b0:1ff:fe36:f411, dst = ff02::1:2, next header = 17, length = 70)
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = fe80::1, dst = ff02::16, next header = 0, length = 116)
ipv6: packet destination not this host
ipv6: decrement hop limit to 0
ipv6: do not forward packets with link-local source or destination address
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
ipv6: Received (src = fe80::1, dst = ff02::16, next header = 0, length = 36)
ipv6: packet destination not this host
[... 652 lines cut ...]
ipv6: forward nh = 58 to other threads
ipv6: handle ICMPv6 packet (nh = 58)
nib: Handle packet (icmpv6->type = 133)
nib: Received valid router solicitation:
     - Source address: fe80::2068:3123:59f5:d238
     - Destination address: ff02::2
ipv6: waiting for incoming message.
ipv6: GNRC_NETAPI_MSG_TYPE_SND received
ipv6: send unicast
nib: get next hop link-layer address of fe80::2068:3123:59f5:d238%7
nib: fe80::2068:3123:59f5:d238 is on-link or in NC, start address resolution
nib: resolve address fe80::2068:3123:59f5:d238%7 by reverse translating to 22:68:31:23:59:F5:D2:38
ipv6: set payload length to 104 (network byteorder 6800)
ipv6: set next header to 58
ipv6: write protect up to payload to calculate checksum
ipv6: calculate checksum for upper header.
ipv6: add interface header to packet
ipv6: send unicast over interface 7
ipv6: Sending (src = fe80::18f9:8f5f:305d:58c, dst = fe80::2068:3123:59f5:d238, next header = 58, length = 104)
ipv6: send to 6LoWPAN instead
at86rf215: TX started on 7
at86rf215: TX started on 7
ipv6: waiting for incoming message.
```
</details>

miri64, 2020-04-16:
> ```
> 2020-04-16 20:33:24,154 # nib: default router list is full. Ignoring RA from fe80::18f9:8f5f:305d:58c
> ```

Since you have basically two networks running in parallel you need have two default router lists per node. So [`CONFIG_GNRC_IPV6_NIB_DEFAULT_ROUTER_NUMOF`](https://doc.riot-os.org/group__net__gnrc__ipv6__nib__conf.html#ga67f93b381ce2aad915fa59b99b1b6da5) needs to be set to 2.

benpicco, 2020-04-16:
> So [`CONFIG_GNRC_IPV6_NIB_DEFAULT_ROUTER_NUMOF`](https://doc.riot-os.org/group__net__gnrc__ipv6__nib__conf.html#ga67f93b381ce2aad915fa59b99b1b6da5) needs to be set to 2.

`2020-04-16 21:05:47,501 # `[`0x2066b3`](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c#L182)
`2020-04-16 21:05:47,501 # *** RIOT kernel panic:`
`2020-04-16 21:05:47,501 # FAILED ASSERTION.`

miri64, 2020-04-16:
YAP (yet another patch):

```
diff --git a/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c b/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c
index 7114c22a19..7d7120e50b 100644
--- a/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c
+++ b/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c
@@ -174,7 +174,7 @@ static inline bool _is_valid(const gnrc_netif_t *netif, int idx)
 void _handle_rereg_address(const ipv6_addr_t *addr)
 {
     gnrc_netif_t *netif = gnrc_netif_get_by_ipv6_addr(addr);
-    _nib_dr_entry_t *router = _nib_drl_get_dr();
+    _nib_dr_entry_t *router = _nib_drl_get(NULL, netif->pid);
     const bool router_reachable = (router != NULL) &&
                                   _is_reachable(router->next_hop);
 
diff --git a/sys/net/gnrc/network_layer/ipv6/nib/_nib-internal.c b/sys/net/gnrc/network_layer/ipv6/nib/_nib-internal.c
index 3a3f374dae..642a2971a5 100644
--- a/sys/net/gnrc/network_layer/ipv6/nib/_nib-internal.c
+++ b/sys/net/gnrc/network_layer/ipv6/nib/_nib-internal.c
@@ -408,8 +408,9 @@ _nib_dr_entry_t *_nib_drl_get(const ipv6_addr_t *router_addr, unsigned iface)
         _nib_onl_entry_t *node = def_router->next_hop;
 
         if ((node != NULL) &&
-            (_nib_onl_get_if(node) == iface) &&
-            (ipv6_addr_equal(router_addr, &node->ipv6))) {
+            ((iface == 0) || (_nib_onl_get_if(node) == iface)) &&
+            ((router_addr == NULL) ||
+             ipv6_addr_equal(router_addr, &node->ipv6))) {
             /* It is linked to the default router list so it *should* be set */
             assert(node->mode & _DRL);
             return def_router;
```

But this in general seems to be a bug distinct from what you reported with `samr21-xpro`

benpicco, 2020-04-16:
That one does the trick that now at least the behavior is consistent on both interfaces :smiley: 

```
2020-04-16 22:30:55,354 # Iface  7  HWaddr: 3A:A4  Channel: 26  Page: 0  NID: 0x23
2020-04-16 22:30:55,355 #           Long HWaddr: 22:68:31:23:59:F5:D2:38 
2020-04-16 22:30:55,369 #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2020-04-16 22:30:55,371 #           AUTOACK  ACK_REQ  CSMA  L2-PDU:102 MTU:1280  HL:64  RTR  
2020-04-16 22:30:55,372 #           RTR_ADV  6LO  IPHC  
2020-04-16 22:30:55,384 #           Source address length: 8
2020-04-16 22:30:55,384 #           Link type: wireless
2020-04-16 22:30:55,386 #           inet6 addr: fe80::2068:3123:59f5:d238  scope: link  VAL
2020-04-16 22:30:55,387 #           inet6 addr: 2001:db8:0:1f:2068:3123:59f5:d238  scope: global  TNT[3]
2020-04-16 22:30:55,400 #           inet6 group: ff02::2
2020-04-16 22:30:55,401 #           inet6 group: ff02::1
2020-04-16 22:30:55,402 #           inet6 group: ff02::1:fff5:d238
2020-04-16 22:30:55,402 #           
2020-04-16 22:30:55,402 #           Statistics for Layer 2
2020-04-16 22:30:55,403 #             RX packets 9  bytes 634
2020-04-16 22:30:55,424 #             TX packets 7 (Multicast: 3)  bytes 385
2020-04-16 22:30:55,424 #             TX succeeded 7 errors 0
2020-04-16 22:30:55,425 #           Statistics for IPv6
2020-04-16 22:30:55,425 #             RX packets 6  bytes 640
2020-04-16 22:30:55,432 #             TX packets 7 (Multicast: 3)  bytes 512
2020-04-16 22:30:55,432 #             TX succeeded 7 errors 0
2020-04-16 22:30:55,433 # 
2020-04-16 22:30:55,433 # Iface  6  HWaddr: 3A:A6  Channel: 0  Page: 2  NID: 0x23
2020-04-16 22:30:55,434 #           Long HWaddr: 22:68:31:23:59:F5:D2:3A 
2020-04-16 22:30:55,448 #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2020-04-16 22:30:55,449 #           AUTOACK  ACK_REQ  CSMA  L2-PDU:102 MTU:1280  HL:64  RTR  
2020-04-16 22:30:55,450 #           RTR_ADV  6LO  IPHC  
2020-04-16 22:30:55,450 #           Source address length: 8
2020-04-16 22:30:55,464 #           Link type: wireless
2020-04-16 22:30:55,465 #           inet6 addr: fe80::2068:3123:59f5:d23a  scope: link  VAL
2020-04-16 22:30:55,466 #           inet6 addr: 2001:db8:0:20:2068:3123:59f5:d23a  scope: global  TNT[3]
2020-04-16 22:30:55,480 #           inet6 group: ff02::2
2020-04-16 22:30:55,481 #           inet6 group: ff02::1
2020-04-16 22:30:55,481 #           inet6 group: ff02::1:fff5:d23a
2020-04-16 22:30:55,482 #           
2020-04-16 22:30:55,482 #           Statistics for Layer 2
2020-04-16 22:30:55,483 #             RX packets 15  bytes 1014
2020-04-16 22:30:55,496 #             TX packets 8 (Multicast: 2)  bytes 606
2020-04-16 22:30:55,497 #             TX succeeded 8 errors 0
2020-04-16 22:30:55,497 #           Statistics for IPv6
2020-04-16 22:30:55,498 #             RX packets 3  bytes 352
2020-04-16 22:30:55,512 #             TX packets 8 (Multicast: 2)  bytes 672
2020-04-16 22:30:55,513 #             TX succeeded 8 errors 0
```

miri64, 2020-04-16:
Some questions are raised: Do they use the correct router when registering the address, i.e. goes the RS to `fe80::18f9:8f5f:305d:58e` over interface 6 or `fe80::18f9:8f5f:305d:58c` over interface 7, respectively? Go the NS with the global addresses out the same interface. What does the neighbor cache of the BR say?

miri64, 2020-04-16:
Huh? Please confirm: The 2.4GHz does the full handshake (at least thats what I read from the PCAP), but the sub-GHz gets the valid address???!?

benpicco, 2020-04-16:
Yes, `2020-04-16 23:36:11,059 # Iface  6  HWaddr: 3A:A6  Channel: 0  Page: 2  NID: 0x23` is the sub-GHz one

miri64, 2020-04-16:
I am really confused ^^"

benpicco, 2020-04-16:
Maybe I also need to increment `CONFIG_GNRC_IPV6_NIB_ABR_NUMOF`?

miri64, 2020-04-16:
Maybe. You have two, right?

benpicco, 2020-04-16:
With `CONFIG_GNRC_IPV6_NIB_ABR_NUMOF=2` I both get stuck in a loop:

[border_router.log](https://pastebin.com/mJqyNdp0)
[openmote-b.log](https://pastebin.com/Xq2GUKSf)
[livelock.pcapng.zip](https://github.com/RIOT-OS/RIOT/files/4490085/livelock.pcapng.zip)

If I then restart the border router to break the cycle, the openmote-b ends up with a valid address. (It doesn't get packets routed outside the WPAN yet though)
```
2020-04-17 00:58:20,232 # Iface  7  HWaddr: 3A:A4  Channel: 26  Page: 0  NID: 0x23
2020-04-17 00:58:20,233 #           Long HWaddr: 22:68:31:23:59:F5:D2:38 
2020-04-17 00:58:20,247 #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2020-04-17 00:58:20,248 #           AUTOACK  ACK_REQ  CSMA  L2-PDU:102 MTU:1280  HL:64  RTR  
2020-04-17 00:58:20,249 #           RTR_ADV  6LO  IPHC  
2020-04-17 00:58:20,262 #           Source address length: 8
2020-04-17 00:58:20,263 #           Link type: wireless
2020-04-17 00:58:20,264 #           inet6 addr: fe80::2068:3123:59f5:d238  scope: link  VAL
2020-04-17 00:58:20,267 #           inet6 addr: 2001:db8:0:31:2068:3123:59f5:d238  scope: global  TNT[3]
2020-04-17 00:58:20,278 #           inet6 group: ff02::2
2020-04-17 00:58:20,279 #           inet6 group: ff02::1
2020-04-17 00:58:20,279 #           inet6 group: ff02::1:fff5:d238
2020-04-17 00:58:20,280 #           
2020-04-17 00:58:20,280 #           Statistics for Layer 2
2020-04-17 00:58:20,294 #             RX packets 2528  bytes 183640
2020-04-17 00:58:20,295 #             TX packets 1511 (Multicast: 702)  bytes 81962
2020-04-17 00:58:20,296 #             TX succeeded 1444 errors 15
2020-04-17 00:58:20,297 #           Statistics for IPv6
2020-04-17 00:58:20,298 #             RX packets 1507  bytes 169680
2020-04-17 00:58:20,311 #             TX packets 1511 (Multicast: 702)  bytes 109648
2020-04-17 00:58:20,312 #             TX succeeded 1511 errors 0
2020-04-17 00:58:20,312 # 
2020-04-17 00:58:20,314 # Iface  6  HWaddr: 3A:A6  Channel: 0  Page: 2  NID: 0x23
2020-04-17 00:58:20,326 #           Long HWaddr: 22:68:31:23:59:F5:D2:3A 
2020-04-17 00:58:20,328 #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2020-04-17 00:58:20,329 #           AUTOACK  ACK_REQ  CSMA  L2-PDU:102 MTU:1280  HL:64  RTR  
2020-04-17 00:58:20,330 #           RTR_ADV  6LO  IPHC  
2020-04-17 00:58:20,342 #           Source address length: 8
2020-04-17 00:58:20,343 #           Link type: wireless
2020-04-17 00:58:20,345 #           inet6 addr: fe80::2068:3123:59f5:d23a  scope: link  VAL
2020-04-17 00:58:20,358 #           inet6 addr: 2001:db8:0:32:2068:3123:59f5:d23a  scope: global  VAL
2020-04-17 00:58:20,359 #           inet6 group: ff02::2
2020-04-17 00:58:20,359 #           inet6 group: ff02::1
2020-04-17 00:58:20,360 #           inet6 group: ff02::1:fff5:d23a
2020-04-17 00:58:20,361 #           
2020-04-17 00:58:20,362 #           Statistics for Layer 2
2020-04-17 00:58:20,374 #             RX packets 10  bytes 671
2020-04-17 00:58:20,375 #             TX packets 18 (Multicast: 6)  bytes 1042
2020-04-17 00:58:20,376 #             TX succeeded 18 errors 0
2020-04-17 00:58:20,377 #           Statistics for IPv6
2020-04-17 00:58:20,378 #             RX packets 8  bytes 736
2020-04-17 00:58:20,390 #             TX packets 18 (Multicast: 6)  bytes 1312
2020-04-17 00:58:20,391 #             TX succeeded 18 errors 0
2020-04-17 00:58:20,391 # 
> nib neigh
2020-04-17 00:58:30,486 #  nib neigh
2020-04-17 00:58:30,502 # fe80::18f9:8f5f:305d:58e dev #6 lladdr 1A:F9:8F:5F:30:5D:05:8E router STALE GC
2020-04-17 00:58:30,504 # fe80::18f9:8f5f:305d:58c dev #7 lladdr 1A:F9:8F:5F:30:5D:05:8C router REACHABLE GC
> nib abr
2020-04-17 00:58:32,853 #  nib abr
2020-04-17 00:58:32,855 # 2001:db8:0:31:18f9:8f5f:305d:58c v0 expires 10000min
2020-04-17 00:58:32,857 # 2001:db8:0:32:18f9:8f5f:305d:58e v0 expires 10000min
> nib route
2020-04-17 00:58:35,349 #  nib route
2020-04-17 00:58:35,365 # 2001:db8:0:32::/64 dev #6
2020-04-17 00:58:35,366 # 2001:db8:0:31::/64 dev #7
2020-04-17 00:58:35,367 # default* via fe80::18f9:8f5f:305d:58e dev #6
2020-04-17 00:58:35,368 # default via fe80::18f9:8f5f:305d:58c dev #7
```

miri64, 2020-04-17:
> With `CONFIG_GNRC_IPV6_NIB_ABR_NUMOF=2` I both get stuck in a loop:
> 
> [border_router.log](https://pastebin.com/mJqyNdp0)
> [openmote-b.log](https://pastebin.com/Xq2GUKSf)
> [livelock.pcapng.zip](https://github.com/RIOT-OS/RIOT/files/4490085/livelock.pcapng.zip)

Can you summarize please, what you mean by "stuck in a loop"?

benpicco, 2020-04-17:
Both the border router and the node keep sending packets back and forth between but no progress is made.
A fast, repeating console log is produced on both ends that only stops when the border router is restarted.

miri64, 2020-04-17:
> A fast, repeating console log is produced on both ends that only stops when the border router is restarted.

Sounds similar to what we observed in https://github.com/RIOT-OS/RIOT/pull/13850. Is that one included in your history?

benpicco, 2020-04-17:
> Sounds similar to what we observed in #13850. Is that one included in your history?

Yup, head is at 91200aa6ead7dfe4a0f4750fb632eb5253aafb9e

closed 2020-04-27 by leandrolanzieri

benpicco, 2020-08-07:
I'm still seeing this issue with `uhcp` and only one 802.15.4 interface on both border router and node.
The node is always restarted, but after a certain time it will always get stuck in the `TNT[3]` state.

Restarting the border router fixes this temporarily, so the problem must lie there.

```
Iface  6
          MTU:65535  HL:64  RTR

          Link type: wired
          inet6 addr: fe80::2  scope: link  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff00:2

Iface  5  HWaddr: 50:5E  Channel: 26  Page: 0  NID: 0x23  PHY: O-QPSK

          Long HWaddr: 8F:1B:51:02:E4:02:D0:5E
           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4
          ACK_REQ  CSMA  L2-PDU:102  MTU:1280  HL:64  RTR
          RTR_ADV  6LO  IPHC
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::8d1b:5102:e402:d05e  scope: link  VAL
          inet6 addr: 2001:db8::8d1b:5102:e402:d05e  scope: global  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff02:d05e

> nib neigh
nib neigh
2001:db8::d0ae:c1b:2054:58e dev #5 lladdr D2:AE:0C:1B:20:54:05:8E  UNREACHABLE REGISTERED
```

Will provide more info for easy reproduction. Should I open a new issue for that?

miri64, 2020-08-07:
> Should I open a new issue for that?

Yes, as this was "fixed" a while ago, I'd say a new issue, describing the reproducible case would be better.