RIOT-OS/RIOT #13849: Problems with boards

valeriova96, 2020-04-09:
Hello everyone I am stuck into this problem and can't go on with the exercises. I'm using a mac with the Vagrant VM and at the compiling moment, if I try to change the board I get this error, anyone went through the same issue?
[image]

benpicco, 2020-04-10:
You need to install toolchain first. Check out [the documentation](https://doc.riot-os.org/group__cpu__esp8266.html#esp8266_manual_toolchain_installation).

aabadie, 2020-04-11:
The RIOT tutorials are using an old version of the Virtual Machine which doesn't contain the ESP8266 toolchain.
You can install the toolchain following the same way it's done for the latest version of the virtual machine:
https://github.com/RIOT-OS/RIOT/blob/c68475eb0d61d8729ea3ae8d835d5247638f01a8/dist/tools/packer/scripts/toolchains.sh#L80-L104

valeriova96, 2020-04-24:
thanks a lot, it worked!

closed 2020-05-08 by benpicco