RIOT-OS/RIOT #13920: stm32: ztimer RTT backend overflow

mzew, 2020-04-21:
# Description
Overflow In RTT backend for ztimer on platforms with RTT bit-width smaller than 32-bit. Tested on STM32L4. Issue causes assertion to fail in stm32's RTT implementation.

# Steps to reproduce the issue
In Makefile:
```
USEMODULE += ztimer_msec
USEMODULE += ztimer_periph_rtt
```

Use the following snippet:
```
while (1) {
  ztimer_sleep(ZTIMER_MSEC, 10000);
  foo();
}
```

# Expected results
Function foo() to be called every 10 seconds.

# Actual results
Function foo() is called 3-4 times, then assertion fails in stm32's RTT implementation:
in file cpu/stm32_common/periph/rtt.c at line 157 in function  rtt_set_alarm().

# Versions
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "19.04 (Disco Dingo)"
                   Kernel: Linux 5.0.0-38-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 8.3.0-6ubuntu1) 8.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 8.0.0-3 (tags/RELEASE_800/final)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: missing
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.20.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 2.7.16
                  python2: Python 2.7.16
                  python3: Python 3.7.3
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing

# Patch
[0002-fix-overflow-at-RTT-back-end-of-ztimer-relevant-for-.patch.txt](https://github.com/RIOT-OS/RIOT/files/4513063/0002-fix-overflow-at-RTT-back-end-of-ztimer-relevant-for-.patch.txt)

kaspar030, 2020-04-22:
Very nice report, even includes a fix. Thanks a bunch @mzew!

I've opened a PR with a similar fix in #13922. It is basically the same, but I've replaced `% clock->max_value` with ` & RTT_MAX_VALUE`, which saves a read and a modulo.

Could you confirm that this fixes your issue?

mzew, 2020-04-22:
I confirm, this fixes the issue

PR #13922 (2020-04-23): ztimer: periph_rtt: ensure RTT_MAX_VALUE is honored

### Contribution description

Previously, ztimer would happily set an absolute RTT alarm value that exceeds
RTT's maximum value (though not a longer interval), as the `val` was
simply added to `rtt_get_counter()`.
This commit ensures that the target value wraps around RTT_MAX_VALUE.

Fixes #13920.

### Testing procedure

See #13920.

### Issues/PRs references

Reported wonderfully in #13920.
commit d8c129fc (2020-04-23): ztimer: periph_rtt: ensure RTT_MAX_VALUE is honored

Previously, ztimer would happily set an absolute RTT alarm value that exceeds
RTT's maximum value (though not a longer interval), as the `val` was
simply added to `rtt_get_counter()`.
This commit ensures that the target value wraps around RTT_MAX_VALUE.

Fixes #13920.
PR #13938 (2020-04-24): ztimer: periph_rtt: ensure RTT_MAX_VALUE is honored [backport 2020.04]
# Backport of #13922

### Contribution description

Previously, ztimer would happily set an absolute RTT alarm value that exceeds
RTT's maximum value (though not a longer interval), as the `val` was
simply added to `rtt_get_counter()`.
This commit ensures that the target value wraps around RTT_MAX_VALUE.

Fixes #13920.

### Testing procedure

See #13920.

### Issues/PRs references

Reported wonderfully in #13920.
commit cb7df873 (2020-04-24): ztimer: periph_rtt: ensure RTT_MAX_VALUE is honored

Previously, ztimer would happily set an absolute RTT alarm value that exceeds
RTT's maximum value (though not a longer interval), as the `val` was
simply added to `rtt_get_counter()`.
This commit ensures that the target value wraps around RTT_MAX_VALUE.

Fixes #13920.

(cherry picked from commit d8c129fc476f63822ce6ae430298482631adcc9e)
closed 2020-04-22 by mzew