RIOT-OS/RIOT #13984: tests/pkg_ubasic: failing on esp32-wroom-32

miri64, 2020-04-30:
#### Description
The test `tests/pkg_ubasic` is failing in the nightlies for a few weeks now. See e.g. https://ci.riot-os.org/RIOT-OS/RIOT/master/549d7ff24f91f73120096cb7ac2c8496889f80aa/output/run_test/tests/pkg_ubasic/esp32-wroom-32:gnu.txt:

```
Leaving...
Hard resetting via RTS pin...
make: Nothing to be done for 'termdeps'.
make: Leaving directory '/tmp/dwq.0.8408429956368078/042a2782050bdf92adfc0f62de96b240/tests/pkg_ubasic'
make: Entering directory '/tmp/dwq.0.8408429956368078/042a2782050bdf92adfc0f62de96b240/tests/pkg_ubasic'
ESP32_SDK_DIR should be defined as /path/to/esp-idf directory
ESP32_SDK_DIR is set by default to /opt/esp/esp-idf
r
socat - open:/dev/ttyUSB0,b115200,echo=0,raw
READY
s
START
main(): This is RIOT! (Version: buildtest)
Running test #1... done. Run time: Timeout in expect script at "timeout=TIMEOUT)" (tests/pkg_ubasic/tests/01-run.py:21)
```

Not exactly sure, what is causing this, the timeout for this test is 180s, so the run time really does not seem to show. Maybe someone with access to the hardware can check.

#### Versions
Murdock's test environment

gschorcht, 2020-05-01:
Sorry that I'm reacting with such a big delay, but I'm fully busy with the preparation of my online courses for the online summer term at the university :worried:

The reason for the problem seems to be the new toolchain version with the `newlib-nano-formatted-io` we introduced a few weeks ago. The ESP32 locks completly when
```c
  printf("done. Run time: %.3f s\n", delta_t);
```
is called in `pkg/ubasic/tests.c`. Then it is no longer accessible, not even with the debugger.

So it seems that `printf` with `float` does not work with the `newlib-nano-formatted-io` anymore. I can reproduce this problem with a small test program and `printf` with `float`. Unfortunatly, I have no time at the moment to investigate in more detail why the `printf` function crashes with `float`.

@kaspar030 Should we revert the toolchain change to solve this problem for the moment? This would also solve the problem with the `%llu` format we already discussed in https://github.com/RIOT-OS/RIOT/pull/13812#issuecomment-613339440. Or should we blacklist the test, should we blacklist the test? But blacklisting the test wouldn't solve the reason for the problem and any user could try to use `printf` with float.

gschorcht, 2020-05-02:
Good news, I figured out the problem. The introduction of `newlib-nano-formatted-io'` additionally required setting one of the `CONFIG_NEWLIB_NANO_FORMAT` macros to work correctly with `float` in `printf`.

#14004 fixes the problem.

PR #14004 (2020-05-02): cpu/esp32: fix printf for float with newlib-nano
### Contribution description

This PR fixes the problem when using `printf` with `float` and the `newlib-nano-formatted-io` library that was introduced with the new toolchain version.

Using the `printf` function of the `newlib-nano-formated-io` library with `float` requires the definition of a `_printf_float` function in the newlib function stub table `s_stub_table`: https://github.com/RIOT-OS/RIOT/blob/99e8b04921aff99d197e5bc48ca2565b17b0490e/cpu/esp32/syscalls.c#L142-L145 This is the case, if the `CONFIG_NEWLIB_NANO_FORMAT` macro is defined.  [... cut]
closed 2020-05-02 by miri64