RIOT-OS/RIOT #13997: ping6 is failing when testing with cc2538dk

gustavosinbandera1, 2020-05-01:
#### Description
I Cannot get reply from ping6 command.
Building application "gnrc_networking" for "cc2538dk" with MCU "cc2538".

#### Steps to reproduce the issue
- cd RIOT/examples/gnrc_networking
- make
- make emulate

The emulation is from renode and I have deployed three nodes with the next configuration.
### machine.resc
```
mach create $name
using sysbus
machine LoadPlatformDescription @platforms/cpus/cc2538.repl

connector Connect sysbus.radio wireless

machine PyDevFromFile @scripts/pydev/rolling-bit.py 0x400D2004 0x4 True "sysctrl"

# get an id value starting with 1
$id = `next_value 1`

macro reset
"""
    # set node address based on the $id variable. 0x00 0x12 0x4B is TI OUI
    sysbus WriteByte 0x00280028 $id
    sysbus WriteByte 0x0028002C 0x00
    sysbus WriteByte 0x00280030 0xAB
    sysbus WriteByte 0x00280034 0x89
    sysbus WriteByte 0x00280038 0x00
    sysbus WriteByte 0x0028003C 0x4B
    sysbus WriteByte 0x00280040 0x12
    sysbus WriteByte 0x00280044 0x00
    sysbus LoadBinary @http://antmicro.com/projects/renode/cc2538_rom_dump.bin-s_524288-0c196cdc21b5397f82e0ff42b206d1cc4b6d7522 0x0
    sysbus LoadELF $image_file
    cpu VectorTableOffset 0x200000
"""

runMacro $reset
```
### board.resc
```
path add $CWD
emulation CreateWirelessMedium "wireless"

# The radio is using a range-based medium (with the `Range` set to `11`)
# If not set, the default SimpleMediumFunction will be used (where range is not relevant)
wireless SetRangeWirelessFunction 100

######################### Machine A - begin #########################
$name="Node-A"
i $ORIGIN/machine.resc
wireless SetPosition radio 0 0 0
showAnalyzer sysbus.uart0
mach clear
start
########################## Machine A - end ##########################

######################### Machine B - begin #########################
$name="Node-B"
i $ORIGIN/machine.resc
wireless SetPosition radio 70 70 0
showAnalyzer sysbus.uart0
mach clear
start
########################## Machine B - end ##########################

######################### Machine C - begin #########################
$name="Node-C"
i $ORIGIN/machine.resc
wireless SetPosition radio 170 70 0
showAnalyzer sysbus.uart0
mach clear
start
########################## Machine C - end ##########################
```
#### Behavior
When simulation start , three nodes with analyzer are available, the IP for each machine are:
fe80::200:1:0:0
fe80::200:2:0:0
fe80::200:3:0:0

#### Actual results
when ping6 is carry out over any node , the output is the following:
```
 ping6 fe80::200:3:0:0
12 bytes from fe80::200:3:0:0%7: icmp_seq=0 ttl=64 rssi=94 dBm time=0.000 ms
```
And the shell is blocked 
[image]
[image]

#### Versions
RIOT
commit 0e972f644500073ac8a0881faeedb38b1f7e6122 (HEAD -> 2020_02_04-cc13xx-driver)

AlfHou, 2022-09-26:
I'm experiencing the same problem on both firefly and cc2538dk.

jia200x, 2022-09-27:
The CC2538 radio driver has been replaced by the Radio HAL variant https://github.com/RIOT-OS/RIOT/pull/16628. The original driver had several issues that got fixed with the new driver. We (stress) tested the driver and all ping combinations seemed to work. If it doesn't work now, there might be a regression.

> I'm experiencing the same problem when emulating both firefly and cc2538dk.

Which RIOT version (or which commit) are you using? Do the nodes freeze when pinging? Which example are you trying to use?

AlfHou, 2022-09-28:
> Which RIOT version (or which commit) are you using? Do the nodes freeze when pinging? Which example are you trying to use?

This is with commit `b024458` which is from 31. August this year. I'm trying to run the `gnrc_networking` example. But I've also tried the `gcoap` example and the `gnrc_minimal` examples. The nodes do seem to freeze when I ping.

AlfHou, 2022-09-28:
Tried pulling the latest commit and still the same error. It also hangs when trying to ping the nodes own ip or with `ping6 ::1`

[image]

jia200x, 2022-09-29:
@AlfHou I cannot reproduce the issue with two `remote-revb` (cc2538):
Dest device:
```
ifconfig
2022-09-29 10:35:58,475 # ifconfig
2022-09-29 10:35:58,483 # Iface  6  HWaddr: 3C:86  Channel: 26  NID: 0x23  PHY: O-QPSK 
2022-09-29 10:35:58,487 #           Long HWaddr: AE:8D:FE:E1:60:4F:BC:86 
2022-09-29 10:35:58,488 #            State: IDLE 
2022-09-29 10:35:58,495 #           ACK_REQ  L2-PDU:102  MTU:1280  HL:64  RTR  
2022-09-29 10:35:58,497 #           6LO  IPHC  
2022-09-29 10:35:58,500 #           Source address length: 8
2022-09-29 10:35:58,502 #           Link type: wireless
2022-09-29 10:35:58,522 #           inet6 addr: fe80::ac8d:fee1:604f:bc86  scope: link  VAL
2022-09-29 10:35:58,524 #           inet6 group: ff02::2
2022-09-29 10:35:58,526 #           inet6 group: ff02::1
2022-09-29 10:35:58,528 #           inet6 group: ff02::1:ff4f:bc86
2022-09-29 10:35:58,530 #           inet6 group: ff02::1a
2022-09-29 10:35:58,531 #           
2022-09-29 10:35:58,535 #           Statistics for Layer 2
2022-09-29 10:35:58,537 #             RX packets 7  bytes 287
2022-09-29 10:35:58,539 #             TX packets 8 (Multicast: 8)  bytes 0
2022-09-29 10:35:58,541 #             TX succeeded 8 errors 0
2022-09-29 10:35:58,548 #           Statistics for IPv6
2022-09-29 10:35:58,549 #             RX packets 7  bytes 434
2022-09-29 10:35:58,551 #             TX packets 8 (Multicast: 8)  bytes 498
2022-09-29 10:35:58,552 #             TX succeeded 8 errors 0
2022-09-29 10:35:58,553 # 
```

Source device:
```
2022-09-29 10:36:44,939 # 12 bytes from fe80::ac8d:fee1:604f:bc86%6: icmp_seq=298 ttl=64 rssi=-89 dBm time=8.240 ms
2022-09-29 10:36:44,993 # 12 bytes from fe80::ac8d:fee1:604f:bc86%6: icmp_seq=299 ttl=64 rssi=-88 dBm time=9.863 ms
2022-09-29 10:36:45,975 # 
2022-09-29 10:36:45,976 # --- fe80::ac8d:fee1:604f:bc86 PING statistics ---
2022-09-29 10:36:45,982 # 300 packets transmitted, 299 packets received, 0% packet loss
2022-09-29 10:36:45,985 # round-trip min/avg/max = 6.967/9.147/14.019 ms
> help
2022-09-29 10:37:27,651 # help
2022-09-29 10:37:27,653 # Command              Description
2022-09-29 10:37:27,655 # ---------------------------------------
2022-09-29 10:37:27,659 # 6ctx                 6LoWPAN context configuration tool
2022-09-29 10:37:27,662 # ifconfig             Configure network interfaces
2022-09-29 10:37:27,669 # nib                  Configure neighbor information base
2022-09-29 10:37:27,672 # ping                 Alias for ping6
2022-09-29 10:37:27,674 # ping6                Ping via ICMPv6
2022-09-29 10:37:27,679 # pm                   interact with layered PM subsystem
2022-09-29 10:37:27,686 # ps                   Prints information about running threads.
2022-09-29 10:37:27,689 # reboot               Reboot the node
2022-09-29 10:37:27,695 # rpl                  rpl configuration tool ('rpl help' for more information)
2022-09-29 10:37:27,702 # udp                  send data over UDP and listen on UDP ports
2022-09-29 10:37:27,704 # version              Prints current RIOT_VERSION
2022-09-29 10:39:06,334 # ping fe80::ac8d:fee1:609a:709b
2022-09-29 10:39:06,343 # 12 bytes from fe80::ac8d:fee1:609a:709b: icmp_seq=0 ttl=64 time=0.654 ms
2022-09-29 10:39:07,345 # 12 bytes from fe80::ac8d:fee1:609a:709b: icmp_seq=1 ttl=64 time=0.655 ms
2022-09-29 10:39:08,350 # 12 bytes from fe80::ac8d:fee1:609a:709b: icmp_seq=2 ttl=64 time=0.655 ms
2022-09-29 10:39:08,350 # 
2022-09-29 10:39:08,352 # --- fe80::ac8d:fee1:609a:709b PING statistics ---
2022-09-29 10:39:08,353 # 3 packets transmitted, 3 packets received, 0% packet loss
2022-09-29 10:39:08,356 # round-trip min/avg/max = 0.654/0.654/0.655 ms
> ping6 ::1
2022-09-29 10:39:15,738 # ping6 ::1
2022-09-29 10:39:15,745 # 12 bytes from ::1: icmp_seq=0 ttl=64 time=0.738 ms
2022-09-29 10:39:16,746 # 12 bytes from ::1: icmp_seq=1 ttl=64 time=0.738 ms
2022-09-29 10:39:17,749 # 12 bytes from ::1: icmp_seq=2 ttl=64 time=0.738 ms
2022-09-29 10:39:17,749 # 
2022-09-29 10:39:17,757 # --- ::1 PING statistics ---
2022-09-29 10:39:17,759 # 3 packets transmitted, 3 packets received, 0% packet loss
2022-09-29 10:39:17,760 # round-trip min/avg/max = 0.738/0.738/0.738 ms
```

I'm using the latest master (3b7c1254d3ef23abb4dbc15c5810e06add7f3ea8)

AlfHou, 2022-09-29:
Is this in renode? Also do you build using docker or native toolchain?

I'll try to narrow down my issue a bit more. I'm able to send pings over tap when building for native.

Also, do you mind sharing the `boards.resc` you're using?

gustavowd, 2023-05-07:
> 

Hi. I have the same problem. Ping seems to freeze nodes. I'm using the latest stable release of RIOT (2023.04) and Renode ([1.13.3](https://github.com/renode/renode/releases/tag/v1.13.3)).

this is the board script I'm using:
`:name: CC2538DK

$name?="cc2538dk"

using sysbus

mach create $name

machine LoadPlatformDescription @platforms/cpus/cc2538.repl

machine PyDevFromFile @scripts/pydev/rolling-bit.py 0x400D2004 0x4 True "sysctrl"

# show the UART output
showAnalyzer uart0

# get an id value starting with 1
$id = `next_value 1`
$bin?=@/home/gustavo/RIOT/examples/emcute_mqttsn/bin/cc2538dk/emcute_mqttsn.elf

macro reset
"""
    # set node address based on the $id variable. 0x00 0x12 0x4B is TI OUI
    sysbus WriteDoubleWord 0x00280028 $id
    sysbus WriteDoubleWord 0x0028002C 0x00
    sysbus WriteDoubleWord 0x00280030 0xAB
    sysbus WriteDoubleWord 0x00280034 0x89
    sysbus WriteDoubleWord 0x00280038 0x00
    sysbus WriteDoubleWord 0x0028003C 0x4B
    sysbus WriteDoubleWord 0x00280040 0x12
    sysbus WriteDoubleWord 0x00280044 0x00

    sysbus LoadBinary @http://antmicro.com/projects/renode/cc2538_rom_dump.bin-s_524288-0c196cdc21b5397f82e0ff42b206d1cc4b6d7522 0x0
    sysbus LoadELF $bin
    
    cpu VectorTableOffset 0x200000
"""

runMacro $reset
`

Any help?
Thanks in advance.
Best regards,
Gustavo

gustavowd, 2023-05-07:
A lot of warning in the terminal:

```
09:14:41.3798 [INFO] Loaded monitor commands from: /opt/renode/scripts/monitor.py
09:14:47.5480 [INFO] Including script: /opt/renode/scripts/multi-node/cc2538/cc2538.resc
09:14:47.5563 [INFO] System bus created.
09:14:48.1552 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x00020000, 0x00031FFF>.
09:14:48.1552 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x00032000, 0x0004AFFF>.
09:14:48.1552 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x0004B000, 0x00063FFF>.
09:14:48.1552 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x00064000, 0x0007CFFF>.
09:14:48.1554 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x0007D000, 0x0007FFFF>.
09:14:48.1673 [INFO] sysbus: Loading segment of 80704 bytes length at 0x200000.
09:14:48.1675 [INFO] sysbus: Loading segment of 17960 bytes length at 0x213B40.
09:14:48.1675 [INFO] sysbus: Loading segment of 44 bytes length at 0x27FFD4.
09:14:48.1675 [INFO] sysbus: Loading segment of 512 bytes length at 0x20000000.
09:14:48.3165 [INFO] cpu: Setting initial values: PC = 0x201DA9, SP = 0x20000200.
09:14:48.3176 [INFO] cc2538dk: Machine started.
09:14:48.3449 [WARNING] nvic: Unhandled read from offset 0xD94.
09:14:48.3458 [WARNING] nvic: Unhandled write to offset 0xD94, value 0x5.
09:14:48.3489 [WARNING] sysbus: [cpu: 0x202006] (tag: 'SYS_CTRL_CLOCK_CTRL') WriteDoubleWord to non existing peripheral at 0x400D2098, value 0x1.
09:14:48.3532 [WARNING] sysbus: [cpu: 0x202006] WriteDoubleWord to non existing peripheral at 0x400D40F8, value 0x1.
09:14:48.3532 [WARNING] sysbus: [cpu: 0x202006] WriteDoubleWord to non existing peripheral at 0x400D40FC, value 0x1.
09:14:48.3534 [WARNING] sysbus: [cpu: 0x202006] (tag: 'SYS_CTRL_CLOCK_CTRL') ReadDoubleWord from non existing peripheral at 0x400D2000, returning 0x00000001.
09:14:48.3535 [WARNING] sysbus: [cpu: 0x202006] (tag: 'SYS_CTRL_CLOCK_CTRL') WriteDoubleWord to non existing peripheral at 0x400D2000, value 0x10001.
09:14:48.3735 [WARNING] sysbus: [cpu: 0x20205C] (tag: 'SYS_CTRL_CLOCK_CTRL') ReadDoubleWord from non existing peripheral at 0x400D2000, returning 0x00000001.
09:14:48.3735 [WARNING] sysbus: [cpu: 0x20205C] (tag: 'SYS_CTRL_CLOCK_CTRL') WriteDoubleWord to non existing peripheral at 0x400D2000, value 0x1.
09:14:48.3790 [WARNING] sysbus: [cpu: 0x202070] (tag: 'SYS_CTRL_CLOCK_CTRL') ReadDoubleWord from non existing peripheral at 0x400D2000, returning 0x00000001.
09:14:48.3791 [WARNING] sysbus: [cpu: 0x202070] (tag: 'SYS_CTRL_CLOCK_CTRL') WriteDoubleWord to non existing peripheral at 0x400D2000, value 0x1.
09:14:48.3791 [WARNING] sysbus: [cpu: 0x202070] (tag: 'SYS_CTRL_CLOCK_CTRL') ReadDoubleWord from non existing peripheral at 0x400D2000, returning 0x00000001.
09:14:48.3791 [WARNING] sysbus: [cpu: 0x202070] (tag: 'SYS_CTRL_CLOCK_CTRL') WriteDoubleWord to non existing peripheral at 0x400D2000, value 0x20001.
09:14:48.3791 [WARNING] sysbus: [cpu: 0x202070] (tag: 'SYS_CTRL_CLOCK_CTRL') ReadDoubleWord from non existing peripheral at 0x400D2000, returning 0x00000001.
09:14:48.3791 [WARNING] sysbus: [cpu: 0x202070] (tag: 'SYS_CTRL_CLOCK_CTRL') WriteDoubleWord to non existing peripheral at 0x400D2000, value 0x0.
09:14:48.3792 [WARNING] sysbus: [cpu: 0x202070] (tag: 'SYS_CTRL_CLOCK_CTRL') ReadDoubleWord from non existing peripheral at 0x400D2000, returning 0x00000001.
09:14:48.3792 [WARNING] sysbus: [cpu: 0x202070] (tag: 'SYS_CTRL_CLOCK_CTRL') WriteDoubleWord to non existing peripheral at 0x400D2000, value 0x101.
09:14:48.3864 [WARNING] sysbus: [cpu: 0x20AF54] WriteDoubleWord to non existing peripheral at 0x400D40C0, value 0x8.
09:14:48.3872 [WARNING] sysbus: [cpu: 0x20AF54] WriteDoubleWord to non existing peripheral at 0x400D40C4, value 0x8.
09:14:48.3873 [WARNING] sysbus: [cpu: 0x20AF54] WriteDoubleWord to non existing peripheral at 0x400D40C8, value 0x8.
09:14:48.3873 [WARNING] sysbus: [cpu: 0x20AF54] WriteDoubleWord to non existing peripheral at 0x400D40CC, value 0x8.
09:14:48.3874 [WARNING] sysbus: [cpu: 0x20AF8C] WriteDoubleWord to non existing peripheral at 0x400D4080, value 0x0.
09:14:48.3874 [WARNING] sysbus: [cpu: 0x20AF8C] WriteDoubleWord to non existing peripheral at 0x400D4100, value 0x0.
09:14:48.3895 [WARNING] gpioA: Unhandled write to offset 0x420. Unhandled bits: [0] when writing value 0x1. Tags: AFSEL (0x1).
09:14:48.3896 [WARNING] sysbus: [cpu: 0x20AF8C] WriteDoubleWord to non existing peripheral at 0x400D4084, value 0x8.
09:14:48.3896 [WARNING] sysbus: [cpu: 0x20AF8C] WriteDoubleWord to non existing peripheral at 0x400D4004, value 0x0.
09:14:48.3897 [WARNING] gpioA: Unhandled write to offset 0x420. Unhandled bits: [1] when writing value 0x2. Tags: AFSEL (0x2).
09:14:48.3897 [WARNING] sysbus: [cpu: 0x20B724] (tag: 'SYS_CTRL_CLOCK_CTRL') ReadDoubleWord from non existing peripheral at 0x400D2028, returning 0x00000001.
09:14:48.3897 [WARNING] sysbus: [cpu: 0x20B724] (tag: 'SYS_CTRL_CLOCK_CTRL') WriteDoubleWord to non existing peripheral at 0x400D2028, value 0x1.
09:14:48.3898 [WARNING] sysbus: [cpu: 0x20B724] (tag: 'SYS_CTRL_CLOCK_CTRL') ReadDoubleWord from non existing peripheral at 0x400D202C, returning 0x00000001.
09:14:48.3898 [WARNING] sysbus: [cpu: 0x20B724] (tag: 'SYS_CTRL_CLOCK_CTRL') WriteDoubleWord to non existing peripheral at 0x400D202C, value 0x1.
[... cut ...]
09:14:48.4086 [WARNING] sysbus: [cpu: 0x20B406] (tag: 'GPTIMER0') ReadDoubleWord from non existing peripheral at 0x40030018, returning 0x00000000.
09:14:48.4087 [WARNING] sysbus: [cpu: 0x20B406] (tag: 'GPTIMER0') WriteDoubleWord to non existing peripheral at 0x40030018, value 0x10.
09:14:48.4087 [WARNING] nvic: Unhandled write to offset 0xD98, value 0x2.
09:14:48.4087 [WARNING] nvic: Unhandled write to offset 0xD9C, value 0x20000C60.
09:14:48.4088 [WARNING] nvic: Unhandled write to offset 0xDA0, value 0x16060009.
```

jia200x, 2023-05-08:
I will test it again with some remote boards

gustavowd, 2023-05-17:
> 

Are you running from the renode docker image? If so, which version?

Stiwenz, 2024-06-11:
Any news? Also encountered this problem. In addition to the fact that the issue manifests itself in the gnrc_networking example, the simplest blinky example also gets stuck on the ztimer_sleep() function. Maybe this will help a little to solve the issue…