RIOT-OS/RIOT #14035: I can't use net module like w5100 to work in Release-2020.04 by example.

ouyangkan, 2020-05-06:
#### Description

I test w5100 on example RIOT/examples/gnrc_networking and RIOT/tests/driver_w5100. 

They can run to the shell, but cannot run shell commands. They didn't have this problem in Release-2020.01. Then I checked where w5100 was modified.  Found auto_init_w5100.c moved to RIOT/sys/net/gnrc/netif/init_devs/auto_init_w5100.c. I don't know what effect this has on the network module.

#### Steps to reproduce the issue

1.using the example
2.change Makefile
1)BOARD ?= stm32f4discovery
2)# Add w5100 to use
USEMODULE += w5100
3.compile and burning

ouyangkan, 2020-05-08:
I test it on BOARD ?= native           

#### Expected results(RIOT-2020.04)
```
con xxxx::xxxx:xxxx:xxxx:xxxx  10000
msg_send() core/msg.c:105: Sending from 2 to 4. block=0 src->state=11 target->state=4
msg_send: 2: Direct msg copy from 2 to 4.
msg_send() core/msg.c:105: Sending from 4 to 3. block=0 src->state=11 target->state=4
msg_send: 4: Direct msg copy from 4 to 3.
queue_msg(): queuing message
_msg_receive: 3: _msg_receive.
index = 4
_msg_receive: 3: _msg_receive(): We've got a queued message.
_msg_receive: 3: _msg_receive(): No thread in waiting list.
msg_send() core/msg.c:105: Sending from 3 to 5. block=0 src->state=11 target->state=4
msg_send: 3: Direct msg copy from 3 to 5.
_msg_receive: 5: _msg_receive.
index = -1
_msg_receive: 5: _msg_receive(): No thread in waiting list.
_msg_receive(): 5: No msg in queue. Going blocked.
_msg_receive: 3: _msg_receive.
index = -1
_msg_receive: 3: _msg_receive(): No thread in waiting list.
_msg_receive(): 3: No msg in queue. Going blocked.
```

#### Actual results(RIOT-2020.01)
```
con xxxx::xxxx:xxxx:xxxx:xxxx  10000
msg_send() core/msg.c:113: Sending from 2 to 4. block=0 src->state=11 target->state=4
msg_send: 2: Direct msg copy from 2 to 4.
msg_send() core/msg.c:113: Sending from 4 to 3. block=0 src->state=11 target->state=4
msg_send: 4: Direct msg copy from 4 to 3.
_msg_receive: 3: _msg_receive.
index = -1
_msg_receive: 3: _msg_receive(): No thread in waiting list.
_msg_receive(): 3: No msg in queue. Going blocked.
_msg_receive: 4: _msg_receive.
index = -1
_msg_receive: 4: _msg_receive(): No thread in waiting list.
_msg_receive(): 4: No msg in queue. Going blocked.
msg_send() core/msg.c:113: Sending from 2 to 4. block=0 src->state=11 target->state=4
msg_send: 2: Direct msg copy from 2 to 4.
msg_send() core/msg.c:113: Sending from 4 to 3. block=0 src->state=11 target->state=4
msg_send: 4: Direct msg copy from 4 to 3.
_msg_receive: 3: _msg_receive.
index = -1
_msg_receive: 3: _msg_receive(): No thread in waiting list.
_msg_receive(): 3: No msg in queue. Going blocked.
_msg_receive: 4: _msg_receive.
index = -1
_msg_receive: 4: _msg_receive(): No thread in waiting list.
_msg_receive(): 4: No msg in queue. Going blocked.

```

### I also tested [gnrc_sock_udp](https://github.com/RIOT-OS/RIOT/tree/master/tests/gnrc_sock_udp), he also has no index, I do n’t know how to have [index](https://github.com/RIOT-OS/RIOT/blob/bf9cfb1992857952b182388f39915a49f6c9b7fd/core/msg.c#L354) in _msg_receive.

ouyangkan, 2020-05-09:
#### I modified [gnrc_ep_set](https://github.com/RIOT-OS/RIOT/blob/bf9cfb1992857952b182388f39915a49f6c9b7fd/sys/net/gnrc/sock/include/gnrc_sock_internal.h#L94) in [RIOT-2020.04/sys/net/gnrc/sock/include/gnrc_sock_internal.h](sys/net/gnrc/sock/include/gnrc_sock_internal.h) . Make it the same as the previous versions;
```
static inline void gnrc_ep_set(sock_ip_ep_t *out, const sock_ip_ep_t *in,
                               size_t in_size)
{
    memcpy(out, in, in_size);
    // if (gnrc_netif_highlander()) {
        /* set interface implicitly */
        gnrc_netif_t *netif = gnrc_netif_iter(NULL);
        printf("sock_internal: netif = %d\n", out->netif);

        if (netif != NULL) {
            out->netif = netif->pid;
            printf("sock_internal: netif = %d\n", out->netif);

        }
    // }
}
```

#### @MrKevinWeiss    Now it is working working normally. Excuse me, if I want to directly use whether to add some flag to make it pass judgment.

MrKevinWeiss, 2020-05-11:
I still think this still may be an issue. Maybe @jia200x or @miri64 can comment on this?

miri64, 2020-05-12:
Is `xxxx::xxxx:xxxx:xxxx:xxxx` a link-local address (starts with `fe80::`)?

ouyangkan, 2020-05-12:
> Is `xxxx::xxxx:xxxx:xxxx:xxxx` a link-local address (starts with `fe80::`)?

Yes, I built MQTT broker on `fe80::`. And emcute_con to it.

miri64, 2020-05-12:
Then your real workaround should be appending the _outgoing_ interface's ID using `%`: `xxxx::xxxx:xxxx:xxxx:xxxx%<if_id>`. I try to find out, what is wrong with the optimization in `gnrc_ep_set` though.

miri64, 2020-05-12:
Ah, easy: The app does not include `-DGNRC_NETIF_SINGLE` so `highlander` always returns NULL.

miri64, 2020-05-12:
So the workaround is the legit solution for your problem ;-).

miri64, 2020-05-12:
Grml... `sock_udp_str2ep()` does not parse that... let me fix that.

miri64, 2020-05-12:
(the workaround for now is to set `GNRC_NETIF_SINGLE` in the `emcute` example).

ouyangkan, 2020-05-12:
Thank you for your reply. It will be very helpful.

miri64, 2020-05-12:
With https://github.com/RIOT-OS/RIOT/pull/14064 this is now also the case in master.

miri64, 2020-05-12:
> Grml... `sock_udp_str2ep()` does not parse that... let me fix that.

I was looking at the asymcute code, rather than the emcute code, so this does not help you. But see #14068 for the fix.

ouyangkan, 2020-05-13:
I added -DGNRC_NETIF_SINGLE and deleted the previous one. Now my stm32 is working, Thanks a lot. :blush:

miri64, 2020-05-13:
Then I guess your problem is fixed :-)

closed 2020-05-13 by miri64