RIOT-OS/RIOT #14074: Memory read beyond input buffer boundaries in nanocoap packet parser

mjurczak, 2020-05-13:
#### Description

The current packet parsing pointer is incremented by the token length provided in the packet header without checking if the resulting pointer is still in the valid input buffer boundaries:
https://github.com/RIOT-OS/RIOT/blob/595e8c631fec7f714f846f359475fd98d631bd0f/sys/net/application_layer/nanocoap/nanocoap.c#L82-L88

Additionaly, the coap_get_token_len() function allows for reserved values (>8) up to 15 to be returned without indicating an invalid packet. 
https://github.com/RIOT-OS/RIOT/blob/10a479c1c58a605935bbf61b03e33f1bde8a31ea/sys/include/net/nanocoap.h#L348-L351

The incremented pkt_pos pointer is further processed by the options parsing loop. If the pointer has skipped over the input buffer end, the loop will be entered and an invalid address read access may be made due to incorrect loop condition:
https://github.com/RIOT-OS/RIOT/blob/595e8c631fec7f714f846f359475fd98d631bd0f/sys/net/application_layer/nanocoap/nanocoap.c#L95

#### Steps to reproduce the issue
Provide an input packet with token length longer than the following packet length.
Examples:
[crash1.log](https://github.com/RIOT-OS/RIOT/files/4624863/crash1.log)
[crash2.log](https://github.com/RIOT-OS/RIOT/files/4624864/crash2.log)
[crash3.log](https://github.com/RIOT-OS/RIOT/files/4624866/crash3.log)

#### Expected results
An error code returned by the parsing function if a malformed packet is encountered. No access to memory areas out of the input buffer boundaries should be made.

```c
return -EBADMSG;
```

#### Actual results
Read of memory outside the input boundaries. The parser continues processing beyond the provided input buffer.
Depending on memory allocation method and memory protection means it may result in crash or data processing beyond the input packet buffer.

#### Versions
master: 5ef4b1843af6749bf8f40b5a37dba9f793474a73
tag: 2020.04

mjurczak, 2020-05-13:
Bugfix proposal:
https://github.com/RIOT-OS/RIOT/pull/14075

benpicco, 2020-06-07:
fixed by #14075

benpicco, 2020-06-07:
@nmeum this might be interesting for `fuzzing/gcoap`.

closed 2020-06-07 by benpicco