RIOT-OS/RIOT #14095: Tracker: WoT Thing Description

Citrullin, 2020-05-18:
I started working on a [WoT Thing Description implementation](https://www.w3.org/TR/wot-thing-description/) in RIOT. 

- [ ] Interaction Affordance
- [ ] Security Schema metadata definitions

pokgak, 2020-05-27:
Why is the definitions in the tracker list and why is it checked already?

Citrullin, 2020-05-27:
> Why are the definitions in the tracker list and why is it checked already?

A type definitions in the header files. They are checked, because I already implemented them. It's a tracker for myself or anybody who wants to join the effort. Just a simple way to inform you.

pokgak, 2020-05-27:
Oh okay, I thought it is already merged in master. I'm don't have much time to review it now but you can ping me if you need help with testing :)

Citrullin, 2020-07-06:
In case somebody is interested: I have a test to [print a thing description json string](https://github.com/Citrullin/RIOT/tree/wot_impl/tests/wot_full_json). There is probably a lot of room for improvements. Implementation can be found in [sys/net/application_layer/wot/](https://github.com/Citrullin/RIOT/tree/wot_impl/sys/net/application_layer/wot).

Citrullin, 2020-07-27:
Open questions: 
- Persistence
Only persist JSON result to SPI nor flash etc. or write everything to the file system?

sebastiankb, 2020-09-11:
I found this interesting issue by chance. May I ask about the status? 

End of September we are planning a WoT (online) PlugFest and there would be a good opportunity to test the implementation. What do you think?

Citrullin, 2020-09-11:
> I found this interesting issue by chance. May I ask about the status?

There is a kind of PoC which writes the result into a string buffer. This is, of course, nothing you can use in production.
So the next step is to send the result over the network. That was the last thing I started, but stopped development for now. 

> End of September we are planning a WoT (online) PlugFest and there would be a good opportunity to test the implementation. What do you think?

Probably not done by then. I am currently unemployed and this is just an interesting side project for me. So, there are other, more important topic for me at the moment. Would like to focus on this topic though. I am going to give a talk on Monday on the [RIOT OS summit](https://summit.riot-os.org/2020/#program). We can have a little chat about it on the summit.

Are you on the W3C mailing lists? I am going to write an email for my presentation. Would be great, if somebody checks the facts, so that I am not saying something wrong :)

sebastiankb, 2020-09-14:
Thanks for the update.

>Are you on the W3C mailing lists? I am going to write an email for my presentation. Would be great, if somebody checks the facts, so that I am not saying something wrong :)

Yes sure, we can have a chat and I can review your slides and yes, I'm on the W3C WoT mailing list.

Citrullin, 2020-09-14:
> Thanks for the update.
> 
> > Are you on the W3C mailing lists? I am going to write an email for my presentation. Would be great, if somebody checks the facts, so that I am not saying something wrong :)
> 
> Yes sure, we can have a chat and I can review your slides and yes, I'm on the W3C WoT mailing list.

Sorry, it was the plan, but I created the slides a bit too late, so it didn't make sense to send it to the mailing list. But I will share the video of the talk. Looking forward to get feedback for it. :)

sebastiankb, 2020-09-17:
No problem. Is the video already available?

Citrullin, 2020-09-17:
> No problem. Is the video already available?

Will be available at some point on the [Youtube Channel](https://www.youtube.com/c/RIOT-IoT/videos), I guess.

Citrullin, 2020-09-21:
@sebastiankb I sent an email to the ig and wg email list. I got a reply to approve my email. I did it, but at it looks like my mail didn't got distributed at all. Tbh: I am not that familiar with email lists and I also don't like them very much. ^^
But I would like to send some information to the other users as well. As it looks like I might get a first implementation done by the 28th of September. So, I could present it and the PlugFest.

@kaspar030 I think you asked me on the summit why I don't use gcoap. You were right... I switched to gcoap, it makes also way more sense in order to create a linked list of the coap resources. It fits better in my design ^^

jue89, 2021-04-19:
@Citrullin I finally touched the eSAUL project (or `saul_observer`) again. Would you mind to have a look on the [proposed interface and architecture](https://github.com/RIOT-OS/RIOT/issues/14121#issuecomment-821996391) and check if they fit your needs?

Citrullin, 2021-05-07:
> @Citrullin I finally touched the eSAUL project (or `saul_observer`) again. Would you mind to have a look on the [proposed interface and architecture](https://github.com/RIOT-OS/RIOT/issues/14121#issuecomment-821996391) and check if they fit your needs?

Thanks for pinging. Will check it.