RIOT-OS/RIOT #14115: Communication between two cc2538 motes using GNRC(Generic network stack)

hcnhcn012, 2020-05-22:
Hello guys, now I have two cc2538 motes and want to set up a  UDP communication tunnel between these two motes, I use `gnrc_networking` from RIOT examples, but I found that I could not ping mote from another successfully. Here are two motes' interface configs:
[image]
[image]
[image]
I noticed that each mote's ipv6 address scope is `link`, not `local`, I guess the problem is RIOT don't get the write RF information from board or I missed some flags for switching on RF
Thanks for helping:)

Citrullin, 2020-05-22:
You shouldn't use images. You can simply copy the text from your terminal and use Markdown in order to mark your code etc. "Syntax highlighting" is the part interesting for you in this case.
https://guides.github.com/features/mastering-markdown/

jeandudey, 2020-05-22:
@hcnhcn012 On the second image seems you're pinging yourself, could you try with the other mote address?

benpicco, 2020-05-24:
Actually pinging yourself should work too.
I'm confused because this should just work™

Since you are pinging link-local addresses, have you tried passing the interface ID with the address:

    ping6 fe80::212:4b00:1933:5e56%7

Or did you try to ping the broadcast address

    ping6 ff02::1%7

Do you have antennas connected?

hcnhcn012, 2020-05-25:
> You shouldn't use images. With text only your issue is more accessible for everybody. Especially for people with disabilities. They rely on screen readers. These tools are not necessary able to read the text in your image. You can simply copy the text from your terminal and use Markdown in order to mark your code etc. "Syntax highlighting" is the part interesting for you in this case.
> https://guides.github.com/features/mastering-markdown/

Thanks, I have adjusted my issue post so others can read it fluently and thanks to @benpicco this problem has been solved :)

hcnhcn012, 2020-05-25:
> Actually pinging yourself should work too.
> I'm confused because this should just work™
>
> Since you are pinging link-local addresses, have you tried passing the interface ID with the address:
>
> ``` [...]

Really appreciate that! I actually solved this by tacking your advice. For my board is self-made with a cc2538 MCU I believed that there was an issue in hardware abstract then I spent days reading codes of RIOT-os :)
Thanks again!

closed 2020-05-25 by hcnhcn012