RIOT-OS/RIOT #14133: arduino-due problem with time functions?

xuteka03, 2020-05-25:
Hi, with the same code I get proper output when running RIOT as native but when I flash RIOT to arduino-due, the output is only printing "printf" functions in which there is no time variable, not the rest of the code. There are no errors when flashing. My code is shown below but the problem is the same even with <time.h> and function clock().

`#include <stdio.h>
#include <xtimer.h>

int main(void)
{	
    uint32_t time=xtimer_now_usec();
		
    printf("\nYou are running RIOT on a(n) %s board.\n", RIOT_BOARD);
    printf("This board features a(n) %s MCU.\n", RIOT_MCU);
    
    time=xtimer_now_usec()-time;
    
    printf("Total time: %f ms", (float)time/1000);

    return 0;
}`

This is the output after flashing:
"make" -C /home/josef/RIOT/boards/arduino-due
"make" -C /home/josef/RIOT/boards/common/arduino-due
"make" -C /home/josef/RIOT/core
"make" -C /home/josef/RIOT/cpu/sam3
"make" -C /home/josef/RIOT/cpu/cortexm_common
"make" -C /home/josef/RIOT/cpu/cortexm_common/periph
"make" -C /home/josef/RIOT/cpu/sam3/periph
"make" -C /home/josef/RIOT/cpu/sam_common
"make" -C /home/josef/RIOT/drivers
"make" -C /home/josef/RIOT/drivers/periph_common
"make" -C /home/josef/RIOT/sys
"make" -C /home/josef/RIOT/sys/auto_init
"make" -C /home/josef/RIOT/sys/div
"make" -C /home/josef/RIOT/sys/newlib_syscalls_default
"make" -C /home/josef/RIOT/sys/stdio_uart
"make" -C /home/josef/RIOT/sys/xtimer
   text    data     bss     dec     hex filename
  13048     108    2604   15760    3d90 /home/josef/RIOT/examples/bc-time-due-xtimer/bin/arduino-due/bc-time-due.elf
stty -F /dev/ttyACM0 raw ispeed 1200 ospeed 1200 cs8 -cstopb ignpar eol 255 eof 255
/home/josef/RIOT/dist/tools/bossa-1.8/bossac -p /dev/ttyACM0  -e -i -w -v -b -R /home/josef/RIOT/examples/bc-time-due-xtimer/bin/arduino-due/bc-time-due.bin
Device       : ATSAM3X8
Chip ID      : 285e0a60
Version      : v1.1 Dec 15 2010 19:25:04
Address      : 524288
Pages        : 2048
Page Size    : 256 bytes
Total Size   : 512KB
Planes       : 2
Lock Regions : 32
Locked       : none
Security     : false
Boot Flash   : false
Erase flash

Done in 0.049 seconds
Write 13156 bytes to flash (52 pages)
[==============================] 100% (52/52 pages)
Done in 2.540 seconds
Verify 13156 bytes of flash
[==============================] 100% (52/52 pages)
Verify successful
Done in 2.482 seconds
Set boot flash true
CPU reset.
/home/josef/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200" 
2020-05-25 11:14:55,368 # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
2020-05-25 11:14:56,376 # on: 2020.07-devel-684-g70543b)
2020-05-25 11:14:56,376 # 
2020-05-25 11:14:56,381 # You are running RIOT on a(n) arduino-due board.
2020-05-25 11:14:56,383 # This board features a(n) sam3 MCU.
2020-05-25 11:14:56,384 # main(): This is RIOT! (Version: 2020.07-devel-684-g70543b)
2020-05-25 11:14:56,385 # 
2020-05-25 11:14:56,386 # You are running RIOT on a(n) arduino-due board.
2020-05-25 11:14:56,387 # This board features a(n) sam3 MCU.

And this is the output after calling **make print-versions**:
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "20.04 LTS (Focal Fossa)"
                   Kernel: Linux 5.4.0-31-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.16.3
                 cppcheck: missing
                  doxygen: 1.8.17
                      git: git version 2.25.1
                     make: GNU Make 4.2.1
                  openocd: missing
                   python: missing
                  python2: missing
                  python3: Python 3.8.2
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing

Thank you in advance for any help.

maribu, 2020-05-25:
Hi,

have you added `USEMODULE += printf_float` to your Makefile?

Btw.: It is recommended to try hard to avoid using `float` or `double`, as they are either quite expensive (with hardware FPU available) or extremely expensive (without hardware FPU).

xuteka03, 2020-05-25:
Hi @maribu, thank you for your reply. 

I tried to use the module above but it didn't help, neither did changing float to int. The output is the same.

PeterKietzmann, 2020-05-25:
Can you try adding a `\n` at the end, like: `printf("Total time: %f ms", (float)time/1000);`

xuteka03, 2020-05-25:
Thank you very much @PeterKietzmann, this actually works with integer variable!

Do you or @maribu have any suggestions how to make the measurement more precise "float style" when I am not supposed to use float?

maribu, 2020-05-25:
When you have to use `float` or `double`, then go for it. But here it is not needed. In fact, every API in xtimer works on integer types anyway. So just keep the time values in `uint32_t` or (if you need a higher range) in `uint64_t` and keep the resolution at µs. If you want to print the time in ms with a decimal place, you could do something like this:

```C
#include <stdint.h>
#include <inttypes.h>
#include <stdio.h>

static inline void print_us_as_ms(uint32_t time_us)
{
    uint32_t time_ms = time_us / 1000;
    uint32_t remainder = time_us % 1000;
    printf("%" PRIu32 ".%03" PRIu32 " ms", time_ms, remainder);
}
```

closed 2020-05-25 by maribu

xuteka03, 2020-05-25:
I figured it out - I am just going to use the integer variable time and I'll divide it by 1000 and then use modulo function for the decimal part.

Thank you for your help, I will close this thread now.