RIOT-OS/RIOT #14139: I can't see /dev/ttyUSBX or ttyACMX on Linux using cc2538dk

wspark91, 2020-05-26:
Hi~~~~

I am a beginner in RIOT development.

The test uses cc2538 board using gnrc_border_router, gnrc_networking, and gcap examples.

After uploading gnrc_border_router.elf on cc2538 board and connecting to linux, /dev/ttyACMX or /dev/ttyUSBX is not visible.

When uploading Contiki example on cc2538 board, ttyACM0 appears normally.

gnrc_border_router source was compiled with only the BOARD = cc2538dk option in the docker build environment.

What should I do?

Thank you.

benpicco, 2020-05-26:
Have you tried the instructions from https://doc.riot-os.org/group__boards__cc2538dk.html ?

Under Contiki, do you know if the cc2538 provides stdio over CDC ACM itself, that is without using a separate USB-Serial adapter?

wspark91, 2020-05-26:
@benpicco 
I have already tried but it doesn't work. 
Connected directly to the Linux PC and cc2538, not using separate USB-Serial adapter.

benpicco, 2020-05-27:
Ah that's it then.
There is no USB support for CC2538 in RIOT yet, but it's implemented [in Contiki](https://github.com/contiki-ng/contiki-ng/tree/develop/arch/cpu/cc2538/usb).

You'd have to implement `periph/usbdev.c` for cc2538 like it was done for nrf52, stm32 and sam0.

wspark91, 2020-05-27:
@benpicco 
Thank you. I will try.