RIOT-OS/RIOT #14142: Explain the setting for STLINK_VERSION more clearly.

jhaand, 2020-05-26:
#### Description
I just started using RIOT OS with the Blue Pill board and a generic cheap ST-LINK debugger. It has cost me the better part of a day set the correct STLINK_VERSION variable in the Makefile. Not one place in the documentation. For either OpenOCD or the Blue Pill mentions to set this variable for OpenOCD. 
The documentation for OpenOCD in the wiki does mention "Transports and board support" but not how to include this in the build mechanism. 
The file 'dist/tools/openocd/openocd.cfg' mentions nothing of this setting. 
The error that 'make flash' gives doesn't provide much help either. It just mentions: "Error: open failed.
A pointer to 'RIOT/makefiles/tools/openocd-adapters/stlink.inc.mk' would have helped. Either from the Blue Pill documentation or the OpenOCD documentation. 

### Work-around
I found a build command that manually uploaded the ELF file. 
```#!/bin/bash
# Sure fire way to upload an ELF binary to the Blue Pill board using a
# cheap ST-LINK v2. 
openocd -f /usr/share/openocd/scripts/interface/stlink-v2.cfg \
-c "transport select hla_swd" \
-f /usr/share/openocd/scripts/target/stm32f1x.cfg \
-c "program {{./bin/bluepill/foobar.elf}} verify reset; shutdown;" 
```

### Useful links
https://github.com/RIOT-OS/RIOT/wiki/OpenOCD
https://riot-os.readthedocs.io/generated/group/group__boards__bluepill.html
https://community.platformio.org/t/cannot-upload-to-stm32-bluepill-board-over-stlink-v2/3492

benpicco, 2020-05-26:
Hm with that workaround setting `STLINK_VERSION=2` should work.

Which version of OpenOCD are you using? IIRC when using the latest master branch, specifying the version is no longer needed.

jhaand, 2020-05-26:
It works flawlessly using `STLINK_VERSION=2`. That's not my problem. 
My main problem is that the documentation on that is hard to find. I had to hack myself to get the workaround going and then look backwards. Using `find . -type f -exec grep stlink {} \; ` That got me to  'dist/tools/openocd/openocd.cfg'. Which led to the setting for STLINK_VERSION in the Makefile. 

I'd rather would have seen in the docs on OpenOCD and ST-link for RIOT OS.

benpicco, 2020-05-26:
I actually had the very same issue #12228

Where would you suggest should this documentation live?

jhaand, 2020-05-26:
That's what I'm wondering at this moment. Because this is a generic STM32 ST-link issue. Not just an OpenOCD issue. At this moment I can't find a link for ST-LINK issues with OpenOCD. 

At the bare minimum. 
I would add a pointer in https://github.com/RIOT-OS/RIOT/wiki/OpenOCD towards the './RIOT/makefiles/tools/openocd-adapters/' directory. The paragraph on 'Transports and board support' should also tell how OpenOCD is integrated in RIOT OS. 
Finding the option for SRST and STLINK_VALUE would become easier. 

Also the directory for './RIOT/boards/common/blxxxpill/dist/openocd.cfg' could add the information of these variables. But then you would also need to add this for other ST boards. 
What do you think?

benpicco, 2020-05-26:
> I would add a pointer in https://github.com/RIOT-OS/RIOT/wiki/OpenOCD towards the './RIOT/makefiles/tools/openocd-adapters/' directory

I added [a section](https://github.com/RIOT-OS/RIOT/wiki/OpenOCD#flashing-with-st-link) there

jhaand, 2020-05-27:
That looks great. I expanded that paragraph with info on the SRST setting. That also costed me about 30 minutes to figure out and is a quite recent addition. 

I also included the pointer to the OpenOCD Makefiles in the paragraph https://github.com/RIOT-OS/RIOT/wiki/OpenOCD#transports-and-board-support

benpicco, 2020-05-27:
Thank you!
I guess this can be closed now?

jhaand, 2020-05-27:
Yes. This can be closed now.

closed 2020-05-27 by jhaand