RIOT-OS/RIOT #14148: Tracker: Kconfig: Model features provided by CPU and Board

jia200x, 2020-05-26:
## Description

It was decided during the Virtual Maintainers Assembly 2020.05 to continue with the next phases of the Kconfig migration. This means that Kconfig will be used to model all module’s configurations and dependencies in the build system.

The first milestone is to model all features provided by CPUs and boards. Also explicitly declare and document all the possible features that can be provided and used.

This milestone doesn’t interfere with the current Kconfig integration nor the build system, so it can be performed directly on the master branch. In order to keep the features in sync in the Makefile.features and the Kconfig files, a static check will be added, which will be run for every PR.

## What needs to be done
- Declare Feature symbols: Analog to the current build system, features like HAS_PERIPH_SPI, HAS_PERIPH_GPIO or HAS_RIOTBOOT have to be declared. Check the Annex for the full list of features
- Declare CPUs and boards common symbols: Each CPU and module should declare common symbols to track the hardware in compile time. See "Modelling CPU" and "Modelling boards" for more information.
- Select the provided features by CPUs and boards: Each CPU and Board should select the supported feature symbols.

https://github.com/RIOT-OS/RIOT/pull/13404 can be used as a reference.

## Tracker

- [ ] Declare feature symbols
- [ ] Add static check
- [ ] CPUs
    - [ ] cc1312r1f3
    - [ ] cc1352p1
    - [ ] cc1352r1
    - [ ] cc2538sf53
    - [ ] cc26x0f128
    - [ ] cc430f6137
    - [ ] efm32
      - [ ] efm32gg990f1024
      - [ ] efm32lg990f256
      - [ ] efm32pg12b500f1024gl125
      - [ ] efm32pg1b200f256gm48
      - [ ] efr32mg1p132f256gm32
      - [ ] efr32mg1p132f256gm48
    - [ ] esp32
    - [ ] esp8266
    - [ ] ezr32wg330f256r60
    - [ ] fe310
      - [ ] fe310_g000
      - [ ] fe310_g002
    - [ ] LM4F120H5QR
    - [ ] lpc1768
    - [ ] lpc2388
    - [ ] mk20dx256vlh7
    - [ ] mk22fn512vlh12
    - [ ] mk64fn1m0vll12
    - [ ] mk41z
      - [ ] mkw41z256vht4
      - [ ] mkw41z512vht4
    - [ ] msp430
      - [ ] msp430f1611
      - [ ] msp430f1612
      - [ ] msp430f2617
    - [ ] nrf51
      - [ ] nrf51x22xxaa
      - [ ] nrf51x22xxab
      - [ ] nrf51x22xxac
    - [ ] nrf52
      - [ ] nrf52832xxaa
      - [ ] nrf52840xxaa
    - [ ] p32mx470f512h
    - [ ] p32mz2048efg100
    - [ ] sam3x8ede16a
    - [ ] samd21
      - [ ] saml21j18a
      - [ ] samr21e18a
      - [ ] samr21g18a
      - [ ] samr30g18a
      - [ ] samr34j18b
    - [ ] stm32
      - [ ] stm32f030f4
      - [ ] stm32f030r8
      - [ ] stm32f031k6
      - [ ] stm32f042k6
      - [ ] stm32f051r8
      - [ ] stm32f070rb
      - [ ] stm32f072rb
      - [ ] stm32f091rc
      - [ ] stm32f103c8
      - [ ] stm32f103cb
      - [ ] stm32f103rb
      - [ ] stm32f103re
      - [ ] stm32f207zg
      - [ ] stm32f302r8
      - [ ] stm32f303k8
      - [ ] stm32f303re
      - [ ] stm32f303vc
      - [ ] stm32f303ze
      - [ ] stm32f334r8
      - [ ] stm32f401re
      - [ ] stm32f405rg
      - [ ] stm32f407vg
      - [ ] stm32f410rb
      - [ ] stm32f411re
      - [ ] stm32f412zg
      - [ ] stm32f413zh
      - [ ] stm32f415rg
      - [ ] stm32f429zi
      - [ ] stm32f437vg
      - [ ] stm32f446re
      - [ ] stm32f446ze
      - [ ] stm32f722ze
      - [ ] stm32f723ie
      - [ ] stm32f746zg
      - [ ] stm32f767zi
      - [ ] stm32f769ni
      - [ ] stm32l031k6
      - [ ] stm32l052t8
      - [ ] stm32l053c8
      - [ ] stm32l053r8
      - [ ] stm32l072cz
      - [ ] stm32l073rz
      - [ ] stm32l151cb
      - [ ] stm32l151cb_a
      - [ ] stm32l151rc
      - [ ] stm32l152re
      - [ ] stm32l412kb
      - [ ] stm32l432kc
      - [ ] stm32l433rc
      - [ ] stm32l452re
      - [ ] stm32l475vg
      - [ ] stm32l476rg
      - [ ] stm32l476vg
      - [ ] stm32l496ag
      - [ ] stm32l496zg
      - [ ] stm32l4r5zi
    - [ ] stm32wb55rg

- [ ] Boards (common symbols + feature selection)
   - [ ] 6lowpan-clicker
   - [ ] acd52832
   - [ ] adafruit-clue
   - [ ] airfy-beacon
   - [ ] arduino-due
   - [ ] arduino-duemilanove
   - [ ] arduino-leonardo
   - [ ] arduino-mega2560
   - [ ] arduino-mkr1000
   - [ ] arduino-mkrfox1200
   - [ ] arduino-mkrwan1300
   - [ ] arduino-mkrzero
   - [ ] arduino-nano
   - [ ] arduino-nano-33-ble
   - [ ] arduino-uno
   - [ ] arduino-zero
   - [ ] atmega1284p
   - [ ] atmega256rfr2-xpro
   - [ ] atmega328p
   - [ ] avr-rss2
   - [ ] avsextrem
   - [ ] b-l072z-lrwan1
   - [ ] b-l475e-iot01a
   - [ ] blackpill
   - [ ] blackpill-128kib
   - [ ] bluepill
   - [ ] bluepill-128kib
   - [ ] calliope-mini
   - [ ] cc1312-launchpad
   - [ ] cc1352-launchpad
   - [ ] cc1352p-launchpad
   - [ ] cc2538dk
   - [ ] cc2650-launchpad
   - [ ] cc2650stk
   - [ ] chronos
   - [ ] derfmega128
   - [ ] derfmega256
   - [ ] ek-lm4f120xl
   - [ ] esp32-heltec-lora32-v2
   - [ ] esp32-mh-et-live-minikit
   - [ ] esp32-olimex-evb
   - [ ] esp32-ttgo-t-beam
   - [ ] esp32-wemos-lolin-d32-pro
   - [ ] esp32-wroom-32
   - [ ] esp32-wrover-kit
   - [ ] esp8266-esp-12x
   - [ ] esp8266-olimex-mod
   - [ ] esp8266-sparkfun-thing
   - [ ] f4vi1
   - [ ] feather-m0
   - [ ] feather-nrf52840
   - [ ] firefly
   - [ ] fox
   - [ ] frdm-k22f
   - [ ] frdm-k64f
   - [ ] frdm-kw41z
   - [ ] hamilton
   - [ ] hifive1
   - [ ] hifive1b
   - [ ] i-nucleo-lrwan1
   - [ ] ikea-tradfri
   - [ ] im880b
   - [ ] iotlab-a8-m3
   - [ ] iotlab-m3
   - [ ] limifrog-v1
   - [ ] lobaro-lorabox
   - [ ] lsn50
   - [ ] maple-mini
   - [ ] mbed_lpc1768
   - [ ] mcb2388
   - [ ] mega-xplained
   - [ ] microbit
   - [ ] microduino-corerf
   - [ ] msb-430
   - [ ] msb-430h
   - [ ] msba2
   - [ ] msbiot
   - [ ] mulle
   - [ ] native
   - [ ] nrf51dk
   - [ ] nrf51dongle
   - [ ] nrf52832-mdk
   - [ ] nrf52840-mdk
   - [ ] nrf52840dk
   - [ ] nrf52840dongle
   - [ ] nrf52dk
   - [ ] nrf6310
   - [ ] nucleo-f030r8
   - [ ] nucleo-f031k6
   - [ ] nucleo-f042k6
   - [ ] nucleo-f070rb
   - [ ] nucleo-f072rb
   - [ ] nucleo-f091rc
   - [ ] nucleo-f103rb
   - [ ] nucleo-f207zg
   - [ ] nucleo-f302r8
   - [ ] nucleo-f303k8
   - [ ] nucleo-f303re
   - [ ] nucleo-f303ze
   - [ ] nucleo-f334r8
   - [ ] nucleo-f401re
   - [ ] nucleo-f410rb
   - [ ] nucleo-f411re
   - [ ] nucleo-f412zg
   - [ ] nucleo-f413zh
   - [ ] nucleo-f429zi
   - [ ] nucleo-f446re
   - [ ] nucleo-f446ze
   - [ ] nucleo-f722ze
   - [ ] nucleo-f746zg
   - [ ] nucleo-f767zi
   - [ ] nucleo-l031k6
   - [ ] nucleo-l053r8
   - [ ] nucleo-l073rz
   - [ ] nucleo-l152re
   - [ ] nucleo-l412kb
   - [ ] nucleo-l432kc
   - [ ] nucleo-l433rc
   - [ ] nucleo-l452re
   - [ ] nucleo-l476rg
   - [ ] nucleo-l496zg
   - [ ] nucleo-l4r5zi
   - [ ] nz32-sc151
   - [ ] olimexino-stm32
   - [ ] opencm904
   - [ ] openlabs-kw41z-mini
   - [ ] openlabs-kw41z-mini-256kib
   - [ ] openmote-b
   - [ ] openmote-cc2538
   - [ ] p-l496g-cell02
   - [ ] p-nucleo-wb55
   - [ ] particle-argon
   - [ ] particle-boron
   - [ ] particle-xenon
   - [ ] pba-d-01-kw2x
   - [ ] phynode-kw41z
   - [ ] pic32-wifire
   - [ ] pinetime
   - [ ] pyboard
   - [ ] reel
   - [ ] remote-pa
   - [ ] remote-reva
   - [ ] remote-revb
   - [ ] ruuvitag
   - [ ] samd21-xpro
   - [ ] same54-xpro
   - [ ] saml10-xpro
   - [ ] saml11-xpro
   - [ ] saml21-xpro
   - [ ] samr21-xpro
   - [ ] samr30-xpro
   - [ ] samr34-xpro
   - [ ] seeeduino_arch-pro
   - [ ] sensebox_samd21
   - [ ] serpente
   - [ ] slstk3401a
   - [ ] slstk3402a
   - [ ] sltb001a
   - [ ] slwstk6000b-slwrb4150a
   - [ ] slwstk6000b-slwrb4162a
   - [ ] slwstk6220a
   - [ ] sodaq-autonomo
   - [ ] sodaq-explorer
   - [ ] sodaq-one
   - [ ] sodaq-sara-aff
   - [ ] spark-core
   - [ ] stk3600
   - [ ] stk3700
   - [ ] stm32f030f4-demo
   - [ ] stm32f0discovery
   - [ ] stm32f3discovery
   - [ ] stm32f429i-disc1
   - [ ] stm32f4discovery
   - [ ] stm32f723e-disco
   - [ ] stm32f769i-disco
   - [ ] stm32l0538-disco
   - [ ] stm32l476g-disco
   - [ ] teensy31
   - [ ] telosb
   - [ ] thingy52
   - [ ] ublox-c030-u201
   - [ ] udoo
   - [ ] usb-kw41z
   - [ ] waspmote-pro
   - [ ] wsn430-v1_3b
   - [ ] wsn430-v1_4
   - [ ] yunjia-nrf51822
   - [ ] z1

aabadie, 2020-07-15:
I edited this issue with #14525

closed 2020-07-17 by aabadie

leandrolanzieri, 2020-07-17:
Awesome!

PR #14541 (2020-07-17): tests/kconfig_features: remove boards whitelist

### Contribution description

This PR removes the boards whitelist defined in `tests/kconfig_features`. Since #14148 is almost done (only samd21 based boards remains to be merged), there's no need to maintain this list. This will also enforce new board supports to add Kconfig.

### Testing procedure

A green Murdock

### Issues/PRs references

~Wait for #14189~ and fixes #14148