RIOT-OS/RIOT #14156: RFC 6775 multi-hop NDP test in native board

wspark91, 2020-05-27:
Can I test RFC6775 multi-hop NDP in native board environment on RIOT os?
Are there any documents to reference for environment configuration and testing?

Thank you.

benpicco, 2020-05-28:
You can use [`socket_zep`](https://doc.riot-os.org/group__drivers__socket__zep.html) to simulate 802.15.4 between native nodes.

@miri64 will know more about that though.

wspark91, 2020-05-31:
@miri64 
I am trying to test RFC6775(Neighbor Discovery Optimization for 6LoWPAN) using Native board in RIOT os.
Benpicco said that you can create RFC6775 test environment on Native board using socket_zep.

The following three examples can be used for testing.
gnrc_boarder_router(6LBR) --- gnrc_networking(6LR) --- gcoap(6LN)

If possible, I would appreciate any documentation or explanations to help me build and test my environment.

Thank you.

wspark91, 2020-06-06:
No documentation or examples on how to use socket_zep?

benpicco, 2020-06-07:
There are some bits in the original PR: https://github.com/RIOT-OS/RIOT/pull/6121#issuecomment-272695647

wspark91, 2020-06-07:
@benpicco 
Thank you.
#6121 has already been checked, and I tested socket_zep using gnrc_networking, but it does not work.

miri64, 2020-06-08:
Hi @wspark91, sorry for replying so late, I was quite busy the last few days. Sadly `zep_mesh` seems to be in quite a bit-rotten state, and I don't have the time to fix it ATM. You are of course happy to try to make it work. Alternatively I know, that people were working with [cooja](http://anrg.usc.edu/contiki/index.php/Cooja_Simulator) in the past, but I don't know any details. If you have a newer Linux kernel (4.19+) you can also use the `mac802154_hwsim` driver provided in https://github.com/RIOT-OS/RIOT/pull/9942.

wspark91, 2020-06-09:
@miri64 
Thank you for the reply.
Is it difficult to test RFC6775 with socket_zep using native board in the current state?
Is it possible to test RFC6775 using stm32 mote?
What is the fastest and easiest way to test RFC6775 using RIOT os?

Thank you.

miri64, 2020-06-09:
> Is it difficult to test RFC6775 with socket_zep using native board in the current state?

It should not be, as soon as you are able to create a connection between a handful of nodes.

> Is it possible to test RFC6775 using stm32 mote?

If it supports IEEE 802.15.4, yes.

> What is the fastest and easiest way to test RFC6775 using RIOT os?

It depends a bit of *what* you want to test. If it is just about packet parsing, sending of correct messages or the state machine, I'd say, use `socket_zep` with a tool like [`scapy`](https://scapy.net/).  If you want to test real interaction between nodes, I usually use the [FIT/IoT-LAB](https://www.iot-lab.info). Their website also offers up-to-date tutorials how to get RIOT networking to run on the testbed.

wspark91, 2020-06-09:
@miri64 
Thank you very much. It was very helpful.

closed 2020-06-09 by wspark91