RIOT-OS/RIOT #14158: kconfig is broken on case-insensitive file systems

janschlicht, 2020-05-28:
#### Description
You can either have the file `Kconfig` or the folder `kconfig`. For now either `Kconfig` or `kconfig/Kconfig.consts` and `kconfig/Kconfig.features` are deleted automatically when checking out riot.

#### Steps to reproduce the issue
Clone riot onto a case-insensitive APFS-Volume.

#### Expected results
All files are present.

#### Actual results
Either `Kconfig` or `kconfig/Kconfig.consts` and `kconfig/Kconfig.features` are deleted

#### Versions
MacOs 10.15.5 with a case-insensitive APFS-Volume.

leandrolanzieri, 2020-05-28:
@janschlicht could you check if #14160 fixes the issue for you?

janschlicht, 2020-05-28:
@leandrolanzieri yes, that fixes the problem! Thank you.

leandrolanzieri, 2020-05-28:
Good! Thanks for reporting and testing!

PR #14160 (2020-05-28): Kconfig: rename folder to kconfigs
### Contribution description
As reported in #14158, on case-insensitive file systems the root Kconfig file clashes with the kconfig folder where we keep constants and features, producing the deletion of one of them. This renames the folder to remove the conflict.

### Testing procedure
- Clone the RIOT repo on a case-insensitive file system -> the `kconfigs` folder and `Kconfig` file should be there.

### Issues/PRs references
Fixes #14158
closed 2020-05-28 by jia200x