RIOT-OS/RIOT #14164: Failed assertion when using netdev_ieee802154_mr_fsk

jeandudey, 2020-05-28:
#### Description

There's no device yet known to use `netdev_ieee802154_mr_fsk` module to date, `cc26xx_cc13xx_rf` is on the works (yet :c).

However as it's uses `netdev_ieee802154` and it pulls `ìeee802154` which then on the root `Makefile.dep` pulls `gnrc_sixlowpan` and subsequently with that `gnrc_netif_6lo`:

<details>
<summary>gnrc_sixlowpan dependencies</summary>

```
ifneq (,$(filter gnrc_sixlowpan,$(USEMODULE)))
  DEFAULT_MODULE += auto_init_gnrc_sixlowpan
  USEMODULE += gnrc_nettype_sixlowpan
  USEMODULE += sixlowpan
  ifneq (,$(filter gnrc_netif,$(USEMODULE)))
    USEMODULE += gnrc_netif_6lo
  endif
endif
```
</details>
<details>
<summary>ieee802154 (and other netdev types) dependencies</summary>

```
ifneq (,$(filter ieee802154 nrfmin esp_now cc110x gnrc_sixloenc,$(USEMODULE)))
  ifneq (,$(filter gnrc_ipv6, $(USEMODULE)))
    USEMODULE += gnrc_sixlowpan
  endif
  ifneq (,$(filter gnrc_ipv6_router, $(USEMODULE)))
    USEMODULE += gnrc_sixlowpan_router
  endif
  ifneq (,$(filter gnrc_ipv6_default, $(USEMODULE)))
    USEMODULE += gnrc_sixlowpan_default
  endif
  ifneq (,$(filter gnrc_ipv6_router_default, $(USEMODULE)))
    USEMODULE += gnrc_sixlowpan_router_default
  endif
  ifneq (,$(filter lwip%, $(USEMODULE)))
    USEMODULE += lwip_sixlowpan
  endif
endif
```
</details>

The assertion fails on `gnrc_netif` at `_test_options` when testing the MTU size, 2022 vs 1280 octets. 2022 is the MTU size when using MR-FSK (PDU is of 2047 minus HDR size and FCS).

#### Steps to reproduce the issue

- Use current master.
- Use a device with an IEEE 802.15.4g device using a PHY which has an MTU >1280.
- The PHY needs to be MR-FSK at initialization time, maybe because of this is why `_test_options` doesn't fails with the `at86rf512` driver at initialization.

#### Expected results
- No crashes, failed assertions.

#### Actual results
- Failed assertion:

```
2020-05-28 17:53:18,260 # our MTU 2022, expected 1280
2020-05-28 17:53:18,261 # 0x6e73
2020-05-28 17:53:18,263 # *** RIOT kernel panic:
2020-05-28 17:53:18,264 # FAILED ASSERTION.
2020-05-28 17:53:18,264 # 
2020-05-28 17:53:18,272 # 	pid | name                 | state    Q | pri | stack  ( used) | base addr  | current     
2020-05-28 17:53:18,280 # 	 - | isr_stack            | -        - |   - |    512 (   84) | 0x20000000 | 0x200001c8
2020-05-28 17:53:18,288 # 	 1 | idle                 | pending  Q |  15 |    256 (  124) | 0x20000ac8 | 0x20000b4c 
2020-05-28 17:53:18,296 # 	 2 | main                 | pending  Q |   7 |   1536 (  260) | 0x20000bc8 | 0x200010cc 
2020-05-28 17:53:18,304 # 	 3 | pktdump              | bl rx    _ |   6 |   1536 (  232) | 0x20005bac | 0x200060c4 
2020-05-28 17:53:18,312 # 	 4 | 6lo                  | bl rx    _ |   3 |   1024 (  244) | 0x200061b0 | 0x200064bc 
2020-05-28 17:53:18,320 # 	 5 | ipv6                 | bl rx    _ |   4 |   1024 (  248) | 0x2000290c | 0x20002c14 
2020-05-28 17:53:18,328 # 	 6 | udp                  | bl rx    _ |   5 |   1024 (  244) | 0x200065b0 | 0x200068bc 
2020-05-28 17:53:18,336 # 	 7 | slipdev              | bl rx    _ |   2 |   1024 (  524) | 0x20003764 | 0x20003a34 
2020-05-28 17:53:18,344 # 	 8 | cc26x2_cc13x2_rf     | running  Q |   2 |   1024 (  668) | 0x200030dc | 0x2000345c 
2020-05-28 17:53:18,350 # 	   | SUM                  |            |     |   8960 ( 2628)
```

With an small patch to show the MTU diff:

```patch
From e1056063f2345a800f7944c870464f0085a71b78 Mon Sep 17 00:00:00 2001
From: Jean Pierre Dudey <jeandudey@hotmail.com>
Date: Thu, 28 May 2020 17:56:37 -0500
Subject: [PATCH 28/28] print phy mtu

---
 sys/net/gnrc/netif/gnrc_netif.c | 1 +
 1 file changed, 1 insertion(+)

diff --git a/sys/net/gnrc/netif/gnrc_netif.c b/sys/net/gnrc/netif/gnrc_netif.c
index 14180be44..4c6fa0354 100644
--- a/sys/net/gnrc/netif/gnrc_netif.c
+++ b/sys/net/gnrc/netif/gnrc_netif.c
@@ -1304,6 +1304,7 @@ static void _test_options(gnrc_netif_t *netif)
                    (IEEE802154_LONG_ADDRESS_LEN == netif->l2addr_len));
 #if IS_USED(MODULE_GNRC_NETIF_IPV6)
 #if IS_USED(MODULE_GNRC_NETIF_6LO)
+            printf("our MTU %d, expected %d\n", netif->ipv6.mtu, IPV6_MIN_MTU);
             assert(netif->ipv6.mtu == IPV6_MIN_MTU);
             assert(netif->sixlo.max_frag_size > 0);
             assert(-ENOTSUP != netif->dev->driver->get(netif->dev, NETOPT_PROTO,
-- 
2.17.1

```

#### Versions

Latest `master` branch.

kaspar030, 2020-05-29:
I labeled "bug", maybe "feature request" is more fitting.

benpicco, 2020-05-29:
But it is a bug - that `assert(netif->ipv6.mtu == IPV6_MIN_MTU);` is wrong.

PR #14172 (2020-06-02): gnrc_netif: relax 6lo MTU assertion for 802.15.4g

### Contribution description

802.15.4g devices have a 2047 byte PDU.
So the assertion `netif->ipv6.mtu == IPV6_MIN_MTU` is too strict here.

This is only enforced on init, so changing the modulation at run-time did not catch this bug.

### Testing procedure

To test, use e.g. `at86rf215` with 

    CFLAGS += -DAT86RF215_DEFAULT_PHY_MODE=IEEE802154_PHY_MR_OQPSK

and flash `examples/gnrc_networking`.
On `master` the this leads to a failed `assert()`.

### Issues/PRs references

fixes #14164
commit 840be824 (2020-06-02): gnrc_netif: relax 6lo MTU assertion for 802.15.4g

802.15.4g devices have a 2047 byte PDU.
So the assertion `netif->ipv6.mtu == IPV6_MIN_MTU` is too strict here.

This is only enforced on init, so changing the modulation at run-time
did not catch this bug.

To test, use e.g. `at86rf215` with

    CFLAGS += -DAT86RF215_DEFAULT_PHY_MODE=IEEE802154_PHY_MR_OQPSK

fixes #14164
closed 2020-06-02 by fjmolinas