RIOT-OS/RIOT #14169: Gcoap does not handle separate responses

chrysn, 2020-05-29:
#### Description

Gcoap seems not to handle separate responses at all.

#### Steps to reproduce the issue

* Run a CoAP server that sends separate responses (ie. sends an empty ACK in response to a CON, takes its time, and then sends the response in a CON of its own), for example the [aiocoap demo server](https://aiocoap.readthedocs.io/en/latest/examples.html) at its `/other/separate` resource (which takes several seconds to respond)
* Run the gcoap example
* `coap get -c your-own-host 5683 /other/separate`

#### Expected results

The client should cease retransmission once it receives the ACK (which comes along at the original request's MID), and then wait (until its timeout) to receive a any new message on the requested token.

If that request is a CON, the client ACKs it. (If it has timed out, it may also RST it.)

#### Actual results

The client ignores the empty ACK, keeps retransmitting, ignores the incoming CON when it's finally there and eventually times out.

#### Versions

Current RIOT master (8a2b089cd5db7f063beb7a001baf8ee7ab763fb9), tested on ntive.

#### Next steps

I'll look into how this is best resolved.

One likely first step is to remove the check for "does the message ID match" in the response processing -- on request/response level, only the token matters.

Then, on receipt of a CON in a response, an ACK or RST needs to be sent. (The distinction could be whether a memo was found, but just sending an ACK should be just as fine, gotta check with the spec).

Finally, arriving empty ACKs will be checked for whether they match a memo, and silence its retransmissions. This may involve some nontrivial changes because it seems to me that the retransmission counter doubles as an application timeout -- that means that once the ACK is received, the eventual timeout needs to be calculated for an application timeout to be scheduled then.

@kb2ma or @haukepetersen, I'd appreciate a quick "sounds good" or "that's problematic" on the above steps before I start hacking away.

kb2ma, 2020-05-29:
You are correct, gcoap does not handle separate responses. See #13894 for a step towards that support.

chrysn, 2020-05-30:
From starting to work through this, a first correction: Responses aren't rejected because they come on a different message ID, they just fall into the "response is CON" code path that is documented to be unhandled. That part is already addressed in the WIP PR #14178; the last part (stop sending CON after receiving an ACK) is yet TBD with the abovementioned nontrivialities about timeout calculation, but should otherwise be straightforward. (Also, the remaining issue does not *break* things, it just makes them inefficient by asking again needlessly)

chrysn, 2020-06-02:
As github has not caught the backlink: This is now being worked on in https://github.com/RIOT-OS/RIOT/pull/14178.

chrysn, 2020-10-26:
Reopening as github tried to be smart about free-text "fixes #bug (no matter what else it says on the line)", rather than relying on explicit `Close: #bug` lines in git's metadata format.

chrysn, 2020-10-26:
Oh, this *was* closed correctly but for the wrong reasons (the closing PR #14178 resolves it, but in its introduction claimed it's only a partial fix, which was later in the PR augmented).

Sorry for the noise.

closed 2020-10-26 by chrysn