RIOT-OS/RIOT #14184: Part of puts/printf output is skipped

keestux, 2020-06-01:
#### Description

Output of puts is dropped after a `\n` in the string.

For example `tests/periph_gpio` starts with these statements
```
    puts("GPIO peripheral driver test\n");
    puts("In this test, pins are specified by integer port and pin numbers.\n"
         "So if your platform has a pin PA01, it will be port=0 and pin=1,\n"
         "PC14 would be port=2 and pin=14 etc.\n\n"
         "NOTE: make sure the values you use exist on your platform! The\n"
         "      behavior for not existing ports/pins is not defined!");
```

#### Expected results
(( Note. The first part of the output is not shown because the `term`
is not connected in time to catch all the output. ))

```
2020-06-01 20:00:04,414 # c.
2020-06-01 20:00:04,414 # 
2020-06-01 20:00:04,416 # NOTE: make sure the values you use exist on your platform! The
2020-06-01 20:00:04,417 #       behavior for not existing ports/pins is not defined!
> bench 0 0
2020-06-01 20:00:11,671 #  bench 0 0
2020-06-01 20:00:11,671 # 
2020-06-01 20:00:12,276 #                  nop loop:     14770us  ---   0.014us per call  ---   67704807 calls per sec
2020-06-01 20:00:12,733 #                  gpio_set:     65126us  ---   0.065us per call  ---   15354850 calls per sec
2020-06-01 20:00:13,232 #                gpio_clear:     65126us  ---   0.065us per call  ---   15354850 calls per sec
2020-06-01 20:00:14,042 #               gpio_toggle:     41257us  ---   0.041us per call  ---   24238311 calls per sec
2020-06-01 20:00:14,728 #                 gpio_read:     26077us  ---   0.026us per call  ---   38347969 calls per sec
2020-06-01 20:00:14,731 #                gpio_write:     32149us  ---   0.032us per call  ---   31105166 calls per sec
2020-06-01 20:00:14,731 # 
2020-06-01 20:00:14,731 #  --- DONE ---
> 
```
#### Actual results

```
2020-06-01 19:57:55,930 # Reconnected to serial port /dev/ttyACM0
2020-06-01 19:57:55,931 #  devel-851-gac2eb-add-sodaq-sff)
2020-06-01 19:57:55,932 # GPIO peripheral driver test
2020-06-01 19:57:55,934 # In this test, pins are specified by integer port and pin numbers.
> bench 0 0
2020-06-01 19:58:18,053 #  bench 0 0
2020-06-01 19:58:18,054 # 
2020-06-01 19:58:18,700 #                  nop loop:     14770us  ---   0.014us per call  ---   67704807 calls per sec
2020-06-01 19:58:19,157 #                  gpio_set:     41257us  ---   0.041us per call  ---   24238311 calls per sec
2020-06-01 19:58:19,656 #                gpio_clear:     65126us  ---   0.065us per call  ---   15354850 calls per sec
2020-06-01 19:58:20,424 #               gpio_toggle:     41257us  ---   0.041us per call  ---   24238311 calls per sec
2020-06-01 19:58:21,109 #                 gpio_read:     49946us  ---   0.049us per call  ---   20021623 calls per sec
2020-06-01 19:58:21,110 #                gpio_write:     32149us  ---   0.032us per call  ---   31105166 calls per sec
2020-06-01 19:58:21,110 # 
```

#### Versions
```
$ make print-versions

Operating System Environment
----------------------------
         Operating System: "Ubuntu" "20.04 LTS (Focal Fossa)"
                   Kernel: Linux 5.4.0-28-generic x86_64 x86_64
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 10.0.0-4ubuntu1 

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.0.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: ccache version 3.7.7
                    cmake: cmake version 3.17.1
                 cppcheck: Cppcheck 1.90
                  doxygen: 1.8.17
                      git: git version 2.25.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 2.7.18rc1
                  python2: Python 2.7.18rc1
                  python3: Python 3.8.2
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

keestux, 2020-06-01:
When I change the puts to the following then the output is what it should be.
```
    puts("GPIO peripheral driver test");
    puts("In this test, pins are specified by integer port and pin numbers.");
    puts("So if your platform has a pin PA01, it will be port=0 and pin=1,");
    puts("PC14 would be port=2 and pin=14 etc.");
    puts("");
    puts("NOTE: make sure the values you use exist on your platform! The");
    puts("      behavior for not existing ports/pins is not defined!");
```
It seems that all output after the `\n` is dropped.

keestux, 2020-06-01:
Oh, I forgot to say that this is on a SAMD21 with stdout via USB. BOARD=sodaq-autonomo

keestux, 2020-06-01:
I've add `USEMODULE += stdio_uart` in the test. With that added the problem does not show up. So it looks like to be a USB (CDC ACM) problem.

keestux, 2020-06-01:
With a modified hello-world example, you can see that printf has the same problem.
```
int main(void)
{
    puts("Hello World!\nHello World!\nHello World!");

    printf("You are running RIOT on a(n) %s board.\nAnd this is some text after \\n\n", RIOT_BOARD);
    printf("This board features a(n) %s MCU.\n", RIOT_MCU);

    return 0;
}
```
Give the following output
```
2020-06-01 21:20:52,055 # Hello World!
2020-06-01 21:20:52,056 # You are running RIOT on a(n) sodaq-sara-sff board.
2020-06-01 21:20:52,057 # This board features a(n) samd21 MCU.
```

keestux, 2020-06-01:
Is this something for @bergzand ?

bergzand, 2020-06-01:
> Is this something for @bergzand ?

Likely. Might be the same issues as #14073

keestux, 2020-06-03:
Not that I want to make this problem more urgent than others, but I'd like to know if somebody is going to work on it.

Or maybe I can try to look at it myself. However, I don't have a proper debug environment, other then enabling DEBUG and adding printf/puts. Of course I can't do that when CDC ACM is the stdio, and doing printfs at the same time. Perhaps someone (@bergzand?) could advise me how to debug this.

keestux, 2020-06-07:
Could it be a compiler bug? (( I decided to undust my Atmel-ICE and examine the problem. ))
This is what goes wrong. Let me quickly show `stdio_write` from `cdc_acm_stdio.c`
```
ssize_t stdio_write(const void* buffer, size_t len)
{
    const char *start = buffer;
    do {
        size_t n = usbus_cdc_acm_submit(&cdcacm, buffer, len);
        usbus_cdc_acm_flush(&cdcacm);
        /* Use tsrb and flush */
        buffer = (char *)buffer + n;
        len -= n;
    } while (len);
    return start - (char *)buffer;
}
```
With a breakpoint at the return I see that `start` and `buffer` have the same value. And returning 0 seems to inform printf and friends that something is wrong and the rest of the buffer is skipped. (Oh, and BTW, printf and friend chop up the output in parts ending with `'\n`. That's why we see the first part upto the `\n`, but everything else is skipped.)
I'll build with -O0 and see if that makes a difference.

keestux, 2020-06-07:
No, wait a minute. Why didn't I see this before?
```
    return start - (char *)buffer;
```
Oops. That should have been.
```
    return (char *)buffer - start;
```

PR #14215 (2020-06-07): usbus/cdc/acm: correct return expression of stdio_write
### Contribution description

This is a correction of `stdio_write` in `cdc_acm_stdio.c` where the expression in the return had the operands reversed.

### Testing procedure

Use a board that uses USB for stdio. And run `tests/periph_gpio`. Enter "bench 0 0" and see that the output ends with:
```
2020-06-07 12:00:39,142 #                 gpio_read:     49946us  ---   0.049us per call  ---   20021623 calls per sec
2020-06-07 12:00:39,143 #                gpio_write:     32148us  ---   0.032us per call  ---   31106134 calls per sec
2020-06- [... cut]
closed 2020-06-07 by benpicco