RIOT-OS/RIOT #14217: tests/periph_timer_periodic is failing

keestux, 2020-06-07:
#### Description

Test `tests/periph_timer_periodic` is failing on my SODAQ boards. These are the only boards I have so I can't tell if the test fails on other boards boards as well. My suspicion is that it fails on SAMD21.

#### Steps to reproduce the issue
Build and run `tests/periph_timer_periodic`

#### Expected results
The output should end with
```
TEST SUCCEEDED
```

#### Actual results
```
...
2020-06-07 21:08:53,834 # [0] tick
2020-06-07 21:08:53,838 # [1] tick
2020-06-07 21:08:53,869 # [0] tick
2020-06-07 21:08:53,869 # 
2020-06-07 21:08:53,869 # Cycles:
2020-06-07 21:08:53,870 # channel 0 = 12	[OK]
2020-06-07 21:08:53,870 # channel 1 = 11	[ERROR]
2020-06-07 21:08:53,871 # TEST FAILED
```

keestux, 2020-06-07:
@benpicco sorry, I have "assigned" you, mainly to draw your attention. Maybe you can first answer this question: what SAMD21 board did you run the test on, and did it succeed?

keestux, 2020-06-07:
@benpicco Nevermind. It is a missing define of `XTIMER_DEV` in common sodaq includes. I'll fix that.

benpicco, 2020-06-07:
Yea this relies on the xtimer defines to get a known good timer configuration.
Unfortunately there is no universal timer config (e.g. requesting a 1 MHz hardware timer will not work on all platforms) but the XTIMER configuration is guaranteed to provide a working timer that can measure the ranges of time used by this test.

benpicco, 2020-06-09:
Can this be closed now?

miri64, 2020-07-01:
Ping @keestux?

keestux, 2020-08-12:
The test is passing. Closing

closed 2020-08-12 by keestux