RIOT-OS/RIOT #14234: usbdev for atmega32u4

akshaim, 2020-06-09:
#### Description

I a trying to port RIOT to  [The Things Uno board ](https://www.thethingsnetwork.org/docs/devices/uno/)( based on Arduino Leonardo). The board features a single USB port just like Leonardo but unlike Leonardo, UART pins are connected to an RN2xx3 chip from Microchip. This means that I am forced to use USB for debugging. This, however, does not work since usbdev is not available for the atmega32u4 chip.

I am looking for help to learn how usbdev cab be implemented for 32u4.

### Useful links
 Similar issue: https://github.com/RIOT-OS/RIOT/issues/14139
[The Things Uno Schematic: ](https://github.com/TheThingsProducts/uno/blob/master/hardware/LoRaUNO868%20V2.pdf)

benpicco, 2020-06-09:
If you are lucky the the USB peripheral will be similar to the one found on sam0.
*Maybe* @bergzand and @maribu can give you some pointers.

maribu, 2020-06-09:
I sadly have zero competence in the USB area. (I would like to change that, though.) If I remember correctly, @bergzand mentioned during the RIOT summit '19 that the Atmel USB implementations are pretty similar, regardless if they come in the ARM/SAM or in the AVR/ATmega flavor. So chances are good that large parts of the USB code can be reused.

akshaim, 2020-06-09:
@benpicco  @maribu I am pretty much new to the build system in RIOT. From what I understand the periph/usbdev  in sam0_common is a very low-level implementation. Considering that SAM and AVR has pretty much nothing in common , I doubt whether we can reuse the code. Please correct me if I am wrong.

benpicco, 2020-06-09:
> Considering that SAM and AVR has pretty much nothing in common

On a MCU there is the CPU core and then there are the peripherals. Those are usually mapped to memory.

You can use the same peripheral IP and attach it to different CPU cores (ATmega, Cortex-M0, Cortex-M4, RiscV, …), they will interact with it by writing to the same peripheral registers (addresses in memory).

What differs is clock setup, that is how the peripheral is clocked. That depends on the MCU design, but on ATmega it's usually much simpler (often just on or off, whereas on sam0 you can have multiple clock sources with different frequencies)

The registers are defined in header files provided by the vendor. They might not use the same name, but you may find they are indeed the same registers. You should find those it Atmel ASF.

For a start, compare the sections about USB in the data sheets of [ATmega32u4](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7766-8-bit-AVR-ATmega16U4-32U4_Datasheet.pdf#page=256) with [SAM D21](http://ww1.microchip.com/downloads/en/DeviceDoc/SAM_D21_DA1_Family_DataSheet_DS40001882F.pdf#page=692).

If you notice there is some copy & paste going on you might be in for a good start.

bergzand, 2020-06-09:
Hi @akshaim, you're indeed looking for a very low level implementation of the USB peripheral. The goal of usbdev is to provide a lightweight abstraction for MCU USB peripherals so that the USBUS stack has a uniform view of different peripherals. An usbdev implementations implement a number of functions to control the low level interfaces, such as retrieval of USB endpoint data. The USB protocol layer itself is implemented in USBUS. `cpu/sam0_common/periph/usbdev.c` is an excellent example of an implementation here as that is AFAIK the one with the least amount of workarounds (looking at you Nordic) and hopefully matches the AVR peripheral a bit.

Some generic documentation on usbdev can be found in the [doxygen](https://riot-os.org/api/group__drivers__periph__usbdev.html). Let me know if you're missing info there, we can always extend it. I've also found [this](https://beyondlogic.org/usbnutshell/usb1.shtml) to be excellent documentation on USB in general. Most of the logic explained there is implemented in USBUS, but it does help for grasping the USB protocol bits without having to dive into the spec itself.

For a correct and complete implementation of usbdev, the following is necessary:
 - Initialization, I think you only have to enable the peripheral (USBE) and disable the FRZCLK in the USBCON register. VBUS detection is in general not used by RIOT. You probably also want to unmask a number of interrupts here.
 - Some generic getters and setters for USBUS to retrieve peripheral info:
   - Getters for both the max supported USB version, usually `USB_VERSION_20`, and for the maximum speed, `USB_SPEED_FULL`. You can probably copy this from the sam0 implementation.
   - A setter to signal an attached peripheral to the host by enabling the pull-up, by clearing `DETACH` in `UDCON`
   - A setter to set the USB address, should be set in the `UDADDR` register together with the `ADDEN` bitflag.
 - An event callback. When an interrupt happens on the device the implementation can signal this to USBUS via `usbdev::cb`. USBUS will then call the event callback to retrieve which event happened. This can be events such as an USB reset or suspend/resume events. Usually the interrupt handler masks the interrupt and during the event callback the specific interrupt flag is retrieved and cleared. This allows for short interrupt handlers so that the usual threading is not interrupted for too long.

A separate set of functions are required for the event handling:
 - `ep_init` provides a way to allocate and retrieve a new endpoint. It should allocated the necessary buffer space for the endpoint data and return a free endpoint number with the correct direction and type, this can already configure the `EPTYPE` and EPDIR` in the UECFG0X.
 - `ep_get` must implement at least a getter for the number of bytes available in the endpoint, when there is new data in the endpoint. Strictly speaking this only has to be implemented in OUT endpoints.
 - `ep_set` has a few more requirements, it must:
   - a way to enable the endpoint, a disabled endpoint must not respond to any traffic, an enabled endpoint should respond with `NAK` when there is no data available (usually handled in hardware).
   - A way to set a stall condition on the endpoint.
 - `ready` signals to the USB peripheral that there is new data in the endpoint buffer of the provided endpoint and that it is ready to be transferred to the host. For OUT endpoints it signals that the stack has read all the new data from the host and that the data can be discarded and overwritten with the next data from the host.
 - An endpoint specific event handler. Similar to the generic one, but this one allows the USBUS stack to retrieve endpoint specific event information, such as a transfer completion event.

For debugging Wireshark on Linux is able to sniff pretty much everything USB and has some decent packet parsing capabilities. This allows you to see where the connection might get stuck during enumeration. Debugging can also be enabled in USBUS to get information on where the stack is in it's state machine and if it got stuck somewhere. Note that this does require a working serial connection and that the DEBUG statements slow down the code, possibly causing timing issues in the USB protocol.

Let me know if you have any further questions here, I'm happy to discuss and help you here and there with this.

akshaim, 2020-06-09:
@benpicco and @bergzand. This helps. I shall investigate and contact you soon.

closed 2021-02-13 by stale[bot]