RIOT-OS/RIOT #14281: Output of tests/posix_semaphore depends on timing

keestux, 2020-06-13:
#### Description

Test `posix_semaphore` may depend on timing, or perhaps on other active threads. This test has four sub tests, TEST1 .. TEST4. The text output of TEST1 is different when running on stdio via UART or CDC ACM (USB).

#### Steps to reproduce the issue
Run tests/posix_semaphore on a board which uses CDC ACM as stdio, and just look at TEST1.

#### Expected results
A well defined sequence of output lines.

#### Actual results
On a board with CDC ACM as stdio, the first few lines of TEST1 are
```
######################### TEST1:
first: sem_init
first: thread create
first: thread created
second: sem_trywait
second: sem_trywait failed
second: sem_trywait done with == 0
second: wait for post
first: sem_getvalue
first: sem_getvalue != 0
first: do yield
first: done yield
```

On a board with "normal" UART, the first few lines of TEST1 are:
```
######################### TEST1:
first: sem_init
first: thread create
first: thread created
first: sem_getvalue
first: sem_getvalue != 0
first: do yield
second: sem_trywait
second: sem_trywait failed
second: sem_trywait done with == 0
second: wait for post
first: done yield
```
The remainder of the test output (including TEST2..TEST4) is identical in the above two scenarios.

keestux, 2020-06-13:
@kaspar030 I need your help again.

kaspar030, 2020-06-13:
I think the issue is again starting a higher-priority thread (lower value) with THREAD_CREATE_WOUT_YIELD. With the usb thread interrupting, the logic breaks.
To fix, I suggest removing "THREAD_CREATE_WOUT_YIELD". The output should then always look like with CDC ACM as stdio.

PR #14284 (2020-06-15): tests/posix_semaphore: better guarantee sequence of output lines
### Contribution description

As suggested by @kaspar030 in #14281 the THREAD_CREATE_WOUT_YIELD flag was removed.
Now the sequence of output lines is identical in two cases: stdio via UART,
and stdio via CDC ACM (which needs a thread for that).

The expects have been changed to match the new situation, which is what
was already given in the CDC ACM case.

### Testing procedure

Run `tests/posix_semaphore` with `compile_and_test_for_board`. Choose a board which uses CDC ACM as stdio. That means there are three threads active when  [... cut]
closed 2020-06-15 by aabadie