RIOT-OS/RIOT #14288: pkg/ccn-lite: unable to `make -j -C examples/ccn-lite-relay`

miri64, 2020-06-15:
#### Description
I am unable to build `examples/ccn-lite-relay` with more than one job

#### Steps to reproduce the issue
```sh
RIOT_CI_BUILD=1 make -C examples/ccn-lite-relay/ --no-print-directory -j
```

#### Expected results
Same as without `-j` parameter:

```
$ RIOT_CI_BUILD=1 make -C examples/ccn-lite-relay/ --no-print-directory
Building application "ccn-lite-relay" for "native" with MCU "native".

[INFO] updating ccn-lite /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/.pkg-state.git-downloaded
echo e37b02c5cb20e9acccea2394be40f7e570a66a4b > /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/.pkg-state.git-downloaded
[INFO] patch ccn-lite
[INFO] updating ccn-lite /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/.pkg-state.git-downloaded
echo e37b02c5cb20e9acccea2394be40f7e570a66a4b > /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/.pkg-state.git-downloaded
[INFO] patch ccn-lite
/home/mlenders/Repositories/RIOT-OS/RIOT/dist/tools/cmake/generate-xcompile-toolchain.sh > /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/xcompile-toolchain.cmake
cmake -B/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/bin -H/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/src \
      -DCMAKE_TOOLCHAIN_FILE=/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/xcompile-toolchain.cmake \
      -DCCNL_RIOT=1 -DRIOT_CFLAGS="-I/home/mlenders/Repositories/RIOT-OS/RIOT/core/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/drivers/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/sys/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/boards/native/include -DNATIVE_INCLUDES -I/home/mlenders/Repositories/RIOT-OS/RIOT/boards/native/include/ -I/home/mlenders/Repositories/RIOT-OS/RIOT/core/include/ -I/home/mlenders/Repositories/RIOT-OS/RIOT/drivers/include/ -I/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/sys/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/src/ccnl-riot/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/src/ccnl-core/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/src/ccnl-pkt/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/src/ccnl-fwd/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/sys/posix/include" -DBUILD_TESTING=OFF
-- The C compiler identification is GNU 10.1.0
-- The CXX compiler identification is GNU 10.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.18") found components: doxygen missing components: dot
Doxygen build started
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/bin
make -C /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/bin
Scanning dependencies of target ccnl-core
[  1%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-array.c.obj
[  2%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-buf.c.obj
[  4%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-callbacks.c.obj
[  5%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-content.c.obj
[  7%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-crypto.c.obj
[  8%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-dump.c.obj
[ 10%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-face.c.obj
[ 11%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-forward.c.obj
[ 13%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-frag.c.obj
[ 14%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-http-status.c.obj
[ 16%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-if.c.obj
[ 17%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-interest.c.obj
[ 19%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-logging.c.obj
[ 20%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-malloc.c.obj
[ 22%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-mgmt.c.obj
[ 23%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-os-time.c.obj
[ 25%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-pkt-util.c.obj
[ 26%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-pkt.c.obj
[ 27%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-prefix.c.obj
[ 29%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-producer.c.obj
[ 30%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-relay.c.obj
[ 32%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-sched.c.obj
[ 33%] Building C object ccnl-core/CMakeFiles/ccnl-core.dir/src/ccnl-sockunion.c.obj
[ 35%] Linking C static library ../lib/libccnl-core.a
[ 35%] Built target ccnl-core
Scanning dependencies of target ccnl-pkt
[... 59 lines cut ...]

[INFO] updating ccn-lite /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/.pkg-state.git-downloaded
echo e37b02c5cb20e9acccea2394be40f7e570a66a4b > /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/.pkg-state.git-downloaded
[INFO] patch ccn-lite
[INFO] updating ccn-lite /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/.pkg-state.git-downloaded
/home/mlenders/Repositories/RIOT-OS/RIOT/dist/tools/cmake/generate-xcompile-toolchain.sh > /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/xcompile-toolchain.cmake
echo e37b02c5cb20e9acccea2394be40f7e570a66a4b > /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/.pkg-state.git-downloaded
[INFO] patch ccn-lite
cmake -B/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/bin -H/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/src \
      -DCMAKE_TOOLCHAIN_FILE=/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/xcompile-toolchain.cmake \
      -DCCNL_RIOT=1 -DRIOT_CFLAGS="-I/home/mlenders/Repositories/RIOT-OS/RIOT/core/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/drivers/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/sys/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/boards/native/include -DNATIVE_INCLUDES -I/home/mlenders/Repositories/RIOT-OS/RIOT/boards/native/include/ -I/home/mlenders/Repositories/RIOT-OS/RIOT/core/include/ -I/home/mlenders/Repositories/RIOT-OS/RIOT/drivers/include/ -I/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/sys/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/src/ccnl-riot/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/src/ccnl-core/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/src/ccnl-pkt/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/src/ccnl-fwd/include -I/home/mlenders/Repositories/RIOT-OS/RIOT/sys/posix/include" -DBUILD_TESTING=OFF
CMake Error at /usr/share/cmake-3.17/Modules/CMakeDetermineSystem.cmake:99 (message):
  Could not find toolchain file:
  /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/xcompile-toolchain.cmake
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

[... cut ...] ool.
-- Configuring incomplete, errors occurred!
make[1]: *** [Makefile:29: /home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/bin/pkg/native/ccn-lite/bin/Makefile] Error 1
make: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/examples/ccn-lite-relay/../../Makefile.include:639: pkg-build-ccn-lite] Error 2
```

#### Versions
According to my bisect, this seems to be an issue ever since https://github.com/RIOT-OS/RIOT/pull/13036 was merged (builds with `-j` in `c8461a4bef^1` just fine).

aabadie, 2020-06-15:
I can't reproduce locally (local toolchain or docker). Can you fill the `Versions` section ? Versions of make and cmake could be helpful.

aabadie, 2020-06-15:
for your log, cmake version is 3.17.

miri64, 2020-06-15:
Done.

fjmolinas, 2020-07-01:
Does this happen if you build in docker?

miri64, 2020-07-01:
It does not happen on my Ubuntu 19.10 machine (`cmake v3.13.4`) and it does not happen in docker (Ubuntu 19.04, cmake v???). I think @aabadie is right in his assessment, that this is only affected by newer versions of `cmake` (confirmed for `cmake v3.17.3`).

miri64, 2020-07-01:
[Ubuntu 20.20, the latest LTS has `cmake v3.16.3`](https://packages.ubuntu.com/focal/cmake) btw. No idea if the bug appears there as well. Would be worth a try by someone who has it installed or wants to test https://github.com/RIOT-OS/riotdocker/pull/104

aabadie, 2020-07-01:
I don't think this is related to CMake but rather to make. You make version is 4.3, ubuntu 20.04 provides 4.2

fjmolinas, 2020-07-01:
Our docker version uses make 4.1, AFAIK there are some backward incompatible changes between 4.1 and <4.2. But I don't know the details.

miri64, 2020-07-01:
> I don't think this is related to CMake but rather to make. You make version is 4.3, ubuntu 20.04 provides 4.2

But the error is a missing file in CMake and building in parallel without `cmake` packages works just fine.

miri64, 2020-07-01:
Regardless: Make on Ubuntu 19.04 is 4.2.1.

fjmolinas, 2020-07-01:
> Regardless: Make on Ubuntu 19.04 is 4.2.1.

Yes I'm just saying that since the docker version is 4.1, then we haven't tested make <4.2.

miri64, 2020-07-01:
4.1 is <4.2...

fjmolinas, 2020-07-01:
> 4.1 is <4.2...

Sorry I meant make >4.2 (so 4.2, 4.3)

miri64, 2020-07-01:
>=4.2 ;-). Anyway: as I use Ubuntu 19.04 and Arch for my personal testing and experiments, I would not claim we did not test make >=4.2.

miri64, 2021-05-20:
Nope, seems to be fixed.

closed 2021-05-20 by miri64