RIOT-OS/RIOT #14328: sam0_common: make debug broken for boards using EDBG

bergzand, 2020-06-22:
#### Description

As of 8b9a98233e, `make debug` is broken for at least the same54-xpro and the samr21-xpro. Using `PROGRAMMER=openocd BOARD=samr21-xpro make debug` does work.

#### Steps to reproduce the issue

`BOARD=samr21-xpro make -C examples/default debug`

#### Expected results

A GDB session to the board is started

#### Actual results

Make returns immediately without starting OpenOCD and GDB

#### Versions

<details>
<summary> summary </summary>

```
Operating System Environment
----------------------------
         Operating System: Gentoo 
                   Kernel: Linux 5.6.14-gentoo x86_64 Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz
             System shell: GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Gentoo 9.3.0 p2) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 9.0.1 

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: ccache version 3.7.9
                    cmake: cmake version 3.16.5
                 cppcheck: Cppcheck 1.88
                  doxygen: 1.8.17
                      git: git version 2.26.2
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01288-g5a79481d-dirty (2020-06-19-14:39)
                   python: Python 3.7.7
                  python2: Python 2.7.18
                  python3: Python 3.7.7
                   flake8: 3.7.9 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.7.7 on Linux
               coccinelle: missing
```
</details>

benpicco, 2020-06-22:
Thank you, I noticed that too!
I thought I botched by openocd install…

I doubt 8b9a98233e166301cabcc08ca93b53f3423fc4e0 is the culprit - that only touches `boards/wemos-zero`.
Did you do a bisect?

bergzand, 2020-06-22:
> I doubt [8b9a982](https://github.com/RIOT-OS/RIOT/commit/8b9a98233e166301cabcc08ca93b53f3423fc4e0) is the culprit - that only touches `boards/wemos-zero`.

That's a copy-paste mistake, should have been 98824df68ba665e2c54b715a12991ae8e51e7115, at least that's what a bisect is telling me

benpicco, 2020-06-22:
A solution would be to *always* 

```
include $(RIOTMAKE)/tools/openocd.inc.mk
```

PR #14346 (2020-06-24): makefiles/boards/sam0.inc.mk: dap: use OpenOCD for debugging

### Contribution description

`ebdg` can only be used for flashing, we still have to use OpenOCD for debugging.

### Testing procedure

make debug` should work again on e.g. `samr21-xpro`.

### Issues/PRs references

fixes #14328
commit a6536671 (2020-06-24): makefiles/boards/sam0.inc.mk: dap: use OpenOCD for debugging

ebdg can only be used for flashing, we still have to use OpenOCD for
debugging.

fixes #14328
closed 2020-06-24 by fjmolinas