RIOT-OS/RIOT #14330: Improve driver implementation documentation

HendrikVE, 2020-06-22:
I recently wrote my first RIOT device driver. In the process of porting my driver I discovered the following (incomplete) documentation here: http://doc.riot-os.org/driver-guide.html.
There is a `General Device Driver Checklist`. It tells us for example, that there needs to be a `include/devab_params.h` file for default parameters of the driver `devab`. But in order to get access to these params from outside of the driver module, one needs to add
```
ifneq (,$(filter devab,$(USEMODULE)))
  USEMODULE_INCLUDES += $(RIOTBASE)/drivers/devab/include
endif
```
to `Makefile.include` within the drivers directory. This is not explained in the documentation as far as I can see.

I also tested this with the command `make generate-driver`. It created two header files `include/bla_constants.h` and `include/bla_params.h` for me. The problem is that they are included with `#include "bla_constants.h"` and `#include "bla_params.h"`. Trying to use this new driver gives `fatal error: bla_constants.h: No such file or directory` on compilation, because they are "hidden" in the include directoy. The script should also adapt `Makefile.include` as stated above.

HendrikVE, 2020-06-22:
@aabadie I believe you wrote the `riotgen` tool?

aabadie, 2020-06-22:
> I believe you wrote the riotgen tool?

Yes! And PRs are welcome :) https://github.com/aabadie/riot-generator

The update of `drivers/Makefile.include` and `drivers/Makefile.dep` is not part of riotgen because it doesn't fit in the current design of the tool. Indeed, the source files are generated from template files. I didn't want it to modify RIOT itself.
I guess the best here is just to update the documentation in RIOT.

HendrikVE, 2020-06-24:
@aabadie Cool, I just raised a small PR. With it these header files will at least be found locally then. Question about the documentation: Is the [Wiki](https://github.com/RIOT-OS/RIOT/wiki/Guide:-Writing-a-device-driver-in-RIOT) linked to the [website](https://github.com/RIOT-OS/RIOT/wiki/Guide:-Writing-a-device-driver-in-RIOT) in some way? I would like to add the missing section mentioned above.

aabadie, 2020-06-24:
> I would like to add the missing section mentioned above.

You can open a PR in doxygen. The driver implementation guide is [here](https://github.com/RIOT-OS/RIOT/blob/master/doc/doxygen/src/driver-guide.md).

Regarding the PR on riot-generator, I'm sorry but I don't like the approach: if the generated driver is PRed upstream, this will have to be changed in order to stay inline with the convention used with the already available ones.
I would rather add a way to extend the list of includes from a Makefile.include located in the driver code. Then from the riot-generator side, it will just be a matter of adding an extra template file.

PR #14369 (2020-07-15): drivers: remove drivers/Makefile.include and add driver specific Makefile.include

### Contribution description

This PR add Makefile.include in each driver modules when necessary and removes the main drivers/Makefile.include one. Each Makefile.include file is included automatically based on the content of the USEMODULE variable, similar to what is done with USEPKG for packages.

### Testing procedure

- A green Murdock should be enough

### Issues/PRs references

Some documentation is added to the driver guide, so this fixes #14330.
There was an initial effort in this direction provided by #6 [... cut]
closed 2020-07-15 by benpicco