RIOT-OS/RIOT #14356: [tracking] Transition to `inline`-able IRQ API

maribu, 2020-06-25:
### Description

In https://github.com/RIOT-OS/RIOT/pull/13999, @fjmolinas added a more efficient `inline`-able IRQ API. The new API requires updating the implementations of every platform. This issue should help tracking what is left

#### State

- [x] [ARM Cortex M](https://github.com/RIOT-OS/RIOT/pull/13999)
- [x] [ARM v7](https://github.com/RIOT-OS/RIOT/pull/14088)
- [x] [AVR](https://github.com/RIOT-OS/RIOT/pull/14085)
- [ ] ESP32 / ESP8266
- [ ] MSP430
- [ ] RISCV
- [ ] MIPS

maribu, 2020-06-25:
@gschorcht: Care to tackle the ESP platform? I could also do it, if you want.

@francois-berder: Care to tackle the MIPS platform? I sadly don't have MIPS hardware.

@kenrabold: Care to tackle the RISC-V platform. (I really should get some RISC-V board myself...)

I can tackle MSP430 and use the FIT/IoT-Lab for testing.

aabadie, 2020-06-25:
@maribu, I can have a look at the risc-v CPU, I have one hifive1b :)

maribu, 2020-06-25:
> @maribu, I can have a look at the risc-v CPU, I have one hifive1b :)

Very cool! It would be super if we could have this done before the feature freeze, so that we have a consistent IRQ API in the next release.

gschorcht, 2020-06-25:
Hm, I'm not sure that makes much sense to ESPs:

1. The ROM size increases slightly (half a kbyte).
2. And I get a lot of compilation errors because the implementation of these inline functions requires to insert some SDK headers that create naming conflicts wherever `irq.h` is included.

So the performance gain does not seem to be worth enough to make such a fundamental change.

maribu, 2020-06-25:
>     1. The ROM size increases (slightly, half a kbyte).

Hmm, that should not happen. Except for strange beasts like AVR, `irq_disable()` and `irq_restore()` should be one machine instruction each. So just the function call should be more expensive (in terms of both ROM size and CPU cycles) than the instruction. Let me take a look.

gschorcht, 2020-06-25:
> > ```
> > 1. The ROM size increases (slightly, half a kbyte).
> > ```
> 
> Hmm, that should not happen. Except for strange beasts like AVR, `irq_disable()` and `irq_restore()` should be one machine instruction each.

Unfortunately not. Since you have to mask some bits in a 32 bit register to restore the interrupts, the assembler code for `irq_restore` is implemented as following:
```
    __asm__ volatile ("extui  %1, %1, 0, 4  \n" /* mask intlevel bits in param */
                      "rsr.ps %0            \n" /* read current PS value */
                      "movi.n a4, -16       \n"
                      "and    a4, a4, %0    \n" /* mask out intlevel bits in PS */
                      "or     a4, a4, %1    \n" /* or intlevel with PS */
                      "wsr.ps a4            \n" /* write back PS */
                      "rsync                \n"
```
The assembler code produces 20 bytes of code. Calling a function, just loads the parameter and the function address into registers and calls the function. These are 6 bytes of instructions plus a function overhead of 4 bytes on entry and return. That is, the whole function call requires only 10 bytes while the inline code requires 20 bytes.