RIOT-OS/RIOT #14361: Bug : bad assertion in cpu/stm32/periph/pwm.c

hugueslarrive, 2020-06-25:
Line 48: `assert((pwm < PWM_NUMOF) && ((freq * res) < timer_clk));`
This prevent to set optimal resolution for a given frequency which is (freq * res) = timer_clk.

```diff --git a/cpu/stm32/periph/pwm.c b/cpu/stm32/periph/pwm.c
index 4bd4342f2..3fbf5c0e7 100644
--- a/cpu/stm32/periph/pwm.c
+++ b/cpu/stm32/periph/pwm.c
@@ -45,7 +45,7 @@ uint32_t pwm_init(pwm_t pwm, pwm_mode_t mode, uint32_t freq, uint16_t res)
     uint32_t timer_clk = periph_timer_clk(pwm_config[pwm].bus);
 
     /* verify parameters */
-    assert((pwm < PWM_NUMOF) && ((freq * res) < timer_clk));
+    assert((pwm < PWM_NUMOF) && ((freq * res) <= timer_clk));
 
     /* power on the used timer */
     periph_clk_en(pwm_config[pwm].bus, pwm_config[pwm].rcc_mask);
diff --git a/tests/periph_pwm/Makefile b/tests/periph_pwm/Makefile
index ab079f2f7..b7c245c3c 100644
--- a/tests/periph_pwm/Makefile
+++ b/tests/periph_pwm/Makefile```

miri64, 2020-07-06:
@hugueslarrive would you be willing to [provide a pull request](https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md#contribute-code)?

hugueslarrive, 2020-07-08:
@miri64 Ok I'll try

closed 2020-07-08 by benpicco

miri64, 2020-07-08:
Thanks for the contribution @hugueslarrive!