RIOT-OS/RIOT #14399: cord_ep example timed out when registering to a resource directory

pokgak, 2020-06-30:
#### Description

The `cord_ep` example application timed out when registering to `aiocoap-rd` resource directory (RD). Latest known working release is 2020.01.

#### Steps to reproduce the issue

With the latest master (e8389a58470edf53800c3a9ad202c25d7c663112) or release 2020.04, start a `cord_ep` example application and try to register to the RD.

I'm using aiocoap version `0.4b3`. Install with `pip3 install --upgrade "git+https://github.com/chrysn/aiocoap@0.4b3#egg=aiocoap[all,docs]"`, the latest from aiocoap master seems to have some issue.

1. Start the RD: `aiocoap-rd`
2. From `cord_ep` example terminal: `cord_ep register [<RD ipv6 ip>]:5683`

#### Expected results

Using RIOT release version 2020.01:

```
main(): This is RIOT! (Version: 2020.01)
CoRE RD client example!

Client information:
  ep: RIOT-9F2723239F272323
  lt: 60s
> cord_ep register [fe80::a4b8:d3ff:feb5:d5aa]:5683
cord_ep register [fe80::a4b8:d3ff:feb5:d5aa]:5683
Registering with RD now, this may take a short while...
RD endpoint event: now registered with a RD
registration successful

CoAP RD connection status:
RD address: coap://[fe80::a4b8:d3ff:feb5:d5aa]:5683
   ep name: RIOT-9F2723239F272323
  lifetime: 60s
    reg if: /resourcedirectory
  location: /reg/1/
```

#### Actual results

Using latest master (e8389a58470edf53800c3a9ad202c25d7c663112) or release 2020.04:

```
main(): This is RIOT! (Version: 2020.07-devel-1658-ge8389)
CoRE RD client example!

Client information:
  ep: RIOT-272A2323272A2323
  lt: 60s
> cord_ep register [fe80::a4b8:d3ff:feb5:d5aa]:5683
cord_ep register [fe80::a4b8:d3ff:feb5:d5aa]:5683
Registering with RD now, this may take a short while...
error: registration failed

```

#### Versions

```
user@riot-vm:~/RIOT$ make print-versions

Operating System Environment
----------------------------
         Operating System: "Ubuntu" "18.04.4 LTS (Bionic Beaver)"
                   Kernel: Linux 4.15.0-76-generic x86_64 x86_64
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2018.09-03 for MIPS MTI Bare Metal) 6.3.0
               msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
     riscv-none-embed-gcc: riscv-none-embed-gcc (GNU MCU Eclipse RISC-V Embedded GCC, 64-bit) 8.2.0
     xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
   xtensa-esp8266-elf-gcc: xtensa-esp8266-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
                    clang: clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: "2.5.0"
  riscv-none-embed-newlib: "3.0.0"
  xtensa-esp32-elf-newlib: "2.2.0"
xtensa-esp8266-elf-newlib: "2.2.0"
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.4.1
                    cmake: cmake version 3.10.2
                 cppcheck: Cppcheck 1.82
                  doxygen: 1.8.13
                      git: git version 2.17.1
                     make: GNU Make 4.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01047-g09ac9ab1 (2020-02-19-16:15)
                   python: Python 2.7.17
                  python2: Python 2.7.17
                  python3: Python 3.6.9
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

leandrolanzieri, 2020-06-30:
When using a link global address I get to connect to the RD with current master:
```
main(): This is RIOT! (Version: 2020.07-devel-1667-g325b7)
CoRE RD client example!

Client information:
  ep: RIOT-9B5623239B562323
  lt: 60s
> ifconfig 7 add unicast fd00:bbbb::2
ifconfig 7 add unicast fd00:bbbb::2
success: added fd00:bbbb::2/64 to interface 7
> cord_ep register [fd00:bbbb::1]:5683
cord_ep register [fd00:bbbb::1]:5683
Registering with RD now, this may take a short while...
RD endpoint event: now registered with a RD
registration successful

CoAP RD connection status:
RD address: coap://[fd00:bbbb::1]:5683
   ep name: RIOT-9B5623239B562323
  lifetime: 60s
    reg if: /resourcedirectory
  location: /reg/2/
> 
```

pokgak, 2020-06-30:
Yup confirmed working with global address. Is this suppose to be the actual behavior and the result in release 2020.01 due to bug thats now fixed in master?

miri64, 2020-06-30:
> Yup confirmed working with global address. Is this suppose to be the actual behavior and the result in release 2020.01 due to bug thats now fixed in master?

Sounds to me like https://github.com/RIOT-OS/RIOT/pull/12994 strikes again.

pokgak, 2020-06-30:
With  `CFLAGS += -DGNRC_NETIF_SINGLE=1` added, registering using the link-local address is working again using master branch.

miri64, 2020-06-30:
This still means a bug in at least the `cord_ep` shell command ;-)

pokgak, 2020-06-30:
I got this working by changing `make_sock_ep()` with the following diff:
```diff
diff --git a/sys/shell/commands/sc_cord_ep.c b/sys/shell/commands/sc_cord_ep.c
index e2abdfdb0..593a77924 100644
--- a/sys/shell/commands/sc_cord_ep.c
+++ b/sys/shell/commands/sc_cord_ep.c
@@ -32,8 +32,17 @@ static int make_sock_ep(sock_udp_ep_t *ep, const char *addr)
     if (sock_udp_str2ep(ep, addr) < 0) {
         return -1;
     }
+    /* if netif not specified in addr */
+    if (!ep->netif) {
+        if (gnrc_netif_numof() == 1) {
+            /* assign the single interface found in gnrc_netif_numof() */
+            ep->netif = (uint16_t)gnrc_netif_iter(NULL)->pid;
+        }
+        else {
+            ep->netif = SOCK_ADDR_ANY_NETIF;
+        }
+    }
     ep->family  = AF_INET6;
-    ep->netif   = SOCK_ADDR_ANY_NETIF;
     if (ep->port == 0) {
         ep->port = COAP_PORT;
     }
```

But I'm not sure if that is truly fixing the root cause or just a workaround. AFAIK the issue here is caused by `remote` having `SOCK_ADDR_ANY_NETIF` as the netif. So using link-local address and specifying the interface will also work here (`fe80::a4b8:d3ff:feb5:d5aa%7`). gcoap cli also uses the check as above to assign netif. That means, in the case where the user does not specify which interface to use and there is more than one interface available, the send will actually fail. The `SOCK_ADDR_ANY_NETIF` assignment in the `else` statement does not actually do anything.

TLDR: 
1. is the above solution enough?
2. When is the right time to use `SOCK_ADDR_ANY_NETIF`?

@miri64 thoughts?

miri64, 2020-06-30:
> TLDR:
> 
>     1. is the above solution enough?

Kinda. This would prevent a redundant assignment ([`SOCK_ADDR_ANY_NETIF` is 0](https://github.com/RIOT-OS/RIOT/blob/e1d27cb7d59af77c1bde54f9090fe39dd6d8425f/sys/include/net/sock.h#L152))

```C
    /* if netif not specified in addr */
    if ((ep->netif == SOCK_ADDR_ANY_NETIF) && (gnrc_netif_numof() == 1)) {
        /* assign the single interface found in gnrc_netif_numof() */
        ep->netif = (uint16_t)gnrc_netif_iter(NULL)->pid;
    }
    ep->family  = AF_INET6;
    if (ep->port == 0) {
        ep->port = COAP_PORT;
    }
```

However, now that I see the code, it turns out the command wasn't broken after all, just its usage changed. But it is good that it now also works again with a single interface without providing it.

>     2. When is the right time to use `SOCK_ADDR_ANY_NETIF`?

With a global address e.g.

pokgak, 2020-06-30:
Thanks I'll make a PR for this

miri64, 2020-06-30:
> ```c
>         ep->netif = (uint16_t)gnrc_netif_iter(NULL)->pid;
> ```

More correct would be (but virtually unused until now and maybe because of the mixed code with `gnrc_netif_numof()` not such a good idea after all)

```C
        ep->netif = netif_get_id(netif_iter(NULL))
```

PR #14403 (2020-06-30): cord_ep: allow not specify netif if only one interface exist
### Contribution description

This PR modifies the `make_sock_ep()` function in `cord_ep` shell function to allow not specifiying the interface when using link-local address.

### Testing procedure

1. Use `aiocoap-rd` version 0.4b3 provided by aiocoap as RD: `pip3 install --upgrade "git+https://github.com/chrysn/aiocoap@0.4b3#egg=aiocoap[all,docs]"`
2. Start the RD: aiocoap-rd
3. From cord_ep example terminal: `cord_ep register/discover [<RD ipv6 ip>]:5683`

Test running `cord_ep register`, `cord_ep discover`:
- using link-local add [... cut]
closed 2020-06-30 by miri64