RIOT-OS/RIOT #14407: gnrc/RPL: instance ID not updated on reconnect

haukepetersen, 2020-07-01:
#### Expected results
When  debugging something unrelated I just stumbled over the following behavior:
- start two RIOT nodes in the same network, node A as RPL root: let `instance ID := 11`, node B as RPL node
- node B selects node A as parent and updates its internal state:

```
node A:
2020-07-01 10:06:31,591 # rpl                           
2020-07-01 10:06:31,593 # instance table:       [X]                                                                    
2020-07-01 10:06:31,595 # parent table: [ ]     [ ]     [ ]
2020-07-01 10:06:31,595 #                                                                                              
2020-07-01 10:06:31,601 # instance [11 | Iface: 8 | mop: 2 | ocp: 0 | mhri: 256 | mri 0]                               
2020-07-01 10:06:31,608 #       dodag [affe::1 | R: 256 | OP: Router | PIO: on | TR(I=[8,20], k=10, c=1, TC=6s)]  
```

```
node B:
2020-07-01 10:06:33,161 # rpl      
2020-07-01 10:06:33,163 # instance table:       [X]
2020-07-01 10:06:33,166 # parent table: [X]     [ ]     [ ]
2020-07-01 10:06:33,166 #                                  
2020-07-01 10:06:33,171 # instance [11 | Iface: 8 | mop: 2 | ocp: 0 | mhri: 256 | mri 0]                               
2020-07-01 10:06:33,179 #       dodag [affe::1 | R: 512 | OP: Router | PIO: on | TR(I=[8,20], k=10, c=0, TC=3s)]       
2020-07-01 10:06:33,183 #               parent [addr: fe80::d18c:56ff:fe24:3ef | rank: 256] 
```

- now we reboot the RPL root (node A) and re-initialize it with a new instance ID (`22`):
```
node A:
2020-07-01 10:07:03,783 # instance table:       [X]     
2020-07-01 10:07:03,786 # parent table: [ ]     [ ]     [ ]
2020-07-01 10:07:03,786 #                    
2020-07-01 10:07:03,791 # instance [22 | Iface: 8 | mop: 2 | ocp: 0 | mhri: 256 | mri 0]                               
2020-07-01 10:07:03,799 #       dodag [affe::1 | R: 256 | OP: Router | PIO: on | TR(I=[8,20], k=10, c=0, TC=0s)]
```
**Problem:** node B will does re-selects node A as parent after the reboot, but it will not update its instance ID:
```
node B:
2020-07-01 10:07:14,090 # rpl send dis  
2020-07-01 10:07:14,094 # success: send a DIS                                                                          
2020-07-01 10:07:14,094 #                     
2020-07-01 10:07:20,075 #  rpl                                                                                         
2020-07-01 10:07:20,077 # instance table:       [X]                                                                    
2020-07-01 10:07:20,080 # parent table: [X]     [ ]     [ ]                                                            
2020-07-01 10:07:20,080 #                                                                                              
2020-07-01 10:07:20,085 # instance [11 | Iface: 8 | mop: 2 | ocp: 0 | mhri: 256 | mri 0]                               
2020-07-01 10:07:20,093 #       dodag [affe::1 | R: 512 | OP: Router | PIO: on | TR(I=[8,20], k=10, c=0, TC=8s)]       
2020-07-01 10:07:20,097 #               parent [addr: fe80::d18c:56ff:fe24:3ef | rank: 256]    
```
I would expect node B updating its parent table to the new instance ID, right?!

#### Actual results
see above

#### Versions
Observed in current master, using two `nrf52dk` nodes running `IP-over-BLE`.

cgundogan, 2020-07-06:
>Problem: node B will does re-selects node A as parent after the reboot, but it will not update its instance ID:

I don't think that B reselects node A .. what you see is just the old routing state from node A.

What happens is that node B ignores the DIOs from node A (instance 22), because your instance table is configured to hold only one instance at a time. Multiple dodags with different instance IDs can exist in parallel. For Node B, the new dodag with instance 22 is just another dodag .. your dodag with instance ID 11 should time out and then node B will choose the dodag with instance 22. From the point of view of Node B, it is impossible to figure out that Node A crashed and instantiated a new dodag with another instance ID.

haukepetersen, 2020-07-09:
Alright, I did a new test and I can confirm that behavior now -> it turned out the timeout took quite a bit longer than I expected, so I never let the setup run long enough so far... So letting it run for some time (>10min, didn't really track the time precisely), the old DODAG is removed from node B and the new DODAG is selected instead.

But I have one more thing I'd like to clarify: I would have expected node B to updated its instance table quite a bit faster when (manually) sending out DIS messages, something in the lines of `send DIS -> check which instances get a reply -> remove the stale ones from the instance table`. But I guess the removal part is entirely coupled to the parent timeout timer and its not influenced by any other operations?!

cgundogan, 2020-07-09:
>when (manually) sending out DIS messages

how did you trigger a manual (multicast) DIS? With the shell?

>something in the lines of send DIS -> check which instances get a reply -> remove the stale ones from the instance table

how often would you repeat such a probing? The link might also be lossy .. in the worst case a negative probing would delete active instances.

>But I guess the removal part is entirely coupled to the parent timeout timer and its not influenced by any other operations?!

yes, the parent timeout is not changed when you *probe manually*. How often do you want to probe with a multicast DIS to identify stale DODAGs? If it's periodical, then you can also just reduce the DODAG lifetime (that's more efficient, because then the probing is done with unicast DIOs). If it's based on application-specific events, then we would need to expose some of the RPL functions to alter parent timeouts.

haukepetersen, 2020-07-09:
> how did you trigger a manual (multicast) DIS? With the shell?

si :-)

> how often would you repeat such a probing? The link might also be lossy .. in the worst case a negative probing would delete active instances.

Yes, I have been thinking about that, and I don't have a clear idea what would be a sensible behavior here.

> If it's based on application-specific events, then we would need to expose some of the RPL functions to alter parent timeouts.

True, that would be more sensible in the default setup, so never mind the issue described above :-)

But thanks to your explanation I found a missing piece to my particular setup: in the case of a connection oriented link layer (aka BLE), we now when there is a connection to a parent node, or when this connection is dropped. So we simply need a mean to relay this information to the RPL implementation so we can update the instance table accordingly. I will look into this to see what the most sensible behavior could be:
- marking the DODAG floating for any potential children of the effected node?
- removing the DODAG right away when the corresponding parent is lost?
- coupling the parent timeout to BLE connection timeout?
--> will think about it.

Anyway, this issue is solved, or better is not an issue in the first place, so closing it. Thanks @cgundogan for you elaboration!

closed 2020-07-09 by haukepetersen