RIOT-OS/RIOT #14410: MIPS: toolchain objcopy doesn't work and no .bin can be generated

aabadie, 2020-07-01:
#### Description

In the current RIOT, the link step for MIPS CPUs is done using the system `objcopy` and not the one that comes from the toolchain.

https://github.com/RIOT-OS/RIOT/blob/799ee40ea73fbd613efc7a1b9879687965e91990/cpu/mips_pic32mz/Makefile.include#L14

The problem with this approach is that it doesn't work when generating `BINFILE` see #14385 and especially #14385 (comment) for details.

#### Steps to reproduce the issue

- Apply the following patch:
```diff
diff --git a/cpu/mips_pic32mz/Makefile.include b/cpu/mips_pic32mz/Makefile.include
index 051b8171f5..e126e742ba 100644
--- a/cpu/mips_pic32mz/Makefile.include
+++ b/cpu/mips_pic32mz/Makefile.include
@@ -34,4 +34,4 @@ OFLAGS += --change-section-lma .gcc_except_table-0x80000000
 endif
 
 # system objcopy is not able to generate a binfile for MIPS
-BINFILE =
+# BINFILE =
```

- Build a firmware for pic32-wifire:

<details>

```
make BOARD=pic32-wifire -C examples/hello-world --no-print-directory 
Building application "hello-world" for "pic32-wifire" with MCU "mips_pic32mz".

"make" -C /work/riot/RIOT/boards/pic32-wifire
"make" -C /work/riot/RIOT/core
"make" -C /work/riot/RIOT/cpu/mips_pic32mz
"make" -C /work/riot/RIOT/cpu/mips32r2_common
"make" -C /work/riot/RIOT/cpu/mips32r2_common/newlib_syscalls_mips_uhi
"make" -C /work/riot/RIOT/cpu/mips32r2_common/periph
"make" -C /work/riot/RIOT/cpu/mips_pic32_common
"make" -C /work/riot/RIOT/cpu/mips_pic32_common/periph
"make" -C /work/riot/RIOT/cpu/mips_pic32mz/p32mz2048efg100
"make" -C /work/riot/RIOT/drivers
"make" -C /work/riot/RIOT/drivers/periph_common
"make" -C /work/riot/RIOT/sys
"make" -C /work/riot/RIOT/sys/auto_init
"make" -C /work/riot/RIOT/sys/stdio_uart
objcopy: Unable to recognise the format of the input file `/work/riot/RIOT/examples/hello-world/bin/pic32-wifire/hello-world.elf'
objcopy: --change-section-lma .startdata+0xffffffff80000000 never used
objcopy: --change-section-lma .bss+0xffffffff80000000 never used
objcopy: --change-section-lma .data+0xffffffff80000000 never used
objcopy: --change-section-lma .rodata+0xffffffff80000000 never used
objcopy: --change-section-lma .dtors+0xffffffff80000000 never used
objcopy: --change-section-lma .ctors+0xffffffff80000000 never used
objcopy: --change-section-lma .jcr+0xffffffff80000000 never used
objcopy: --change-section-lma .eh_frame+0xffffffff80000000 never used
objcopy: --change-section-lma .fini+0xffffffff80000000 never used
objcopy: --change-section-lma .init+0xffffffff80000000 never used
objcopy: --change-section-lma .text+0xffffffff80000000 never used
objcopy: --change-section-lma .exception_vector+0xffffffff80000000 never used
objcopy: --change-section-lma .bootflash2+0xffffffff60000000 never used
objcopy: --change-section-lma .bootflash1+0xffffffff60000000 never used
objcopy: --change-section-lma .lowerbootflashalias+0xffffffff60000000 never used
make: *** [/work/riot/RIOT/examples/hello-world/../../Makefile.include:591: /work/riot/RIOT/examples/hello-world/bin/pic32-wifire/hello-world.bin] Error 1
```
</details>

#### Expected results

=> the build doesn't fail when creating the BINFILE.

#### Actual results

=> the build fails when trying to create the BINFILE.

#### Versions

<details><summary>make print-versions</summary>

```
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "20.04 LTS (Focal Fossa)"
                   Kernel: Linux 5.4.0-33-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1759) 5.4.0
         mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
               msp430-gcc: missing
     riscv-none-embed-gcc: riscv-none-embed-gcc (GNU MCU Eclipse RISC-V Embedded GCC, 64-bit) 8.2.0
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 10.0.0-4ubuntu1 

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: "2.1.0"
  riscv-none-embed-newlib: "3.0.0"
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.7.7
                    cmake: cmake version 3.16.3
                 cppcheck: Cppcheck 1.90
                  doxygen: 1.8.17
                      git: git version 2.25.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01243-ge41c0f49-dirty (2020-05-21-12:54)
                   python: Python 3.8.2
                  python2: Python 2.7.18rc1
                  python3: Python 3.8.2
                   flake8: 3.7.9 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.8.2 on Linux
               coccinelle: missing
```

</details>

miri64, 2020-07-01:
Had the same issue when trying to reproduce https://github.com/RIOT-OS/RIOT/issues/7753

aabadie, 2022-11-01:
MIPS is not supported anymore, so closing this one.

closed 2022-11-01 by aabadie