RIOT-OS/RIOT #14415: Flashing nrf52dk with bmp breaks board

Citrullin, 2020-07-01:
#### Description
This quite worrying. Just tested to flash a nrf52dk with the bmp. I was able to flash it one time. But when I reconnect the USB, it didn't work anymore. I also wasn't able to flash it with a J-Link anymore. Haven't tested it with other boards yet. 
Hopefully it is just a soft break, so I am able to recover the devices. 

#### Steps to reproduce the issue
1. Create a BMP according to [this guide](https://buger.dread.cz/)
2. Connect the BMP to a nrf52dk (can be a cheap nrf52832 test module)
3. Flash hello world.
4. Reconnect the USB
5. You shouldn't be able to flash it again.
6. Connect it to a J-Link. The J-Link also shouldn't be able to flash it.

#### Expected results
Be able to flash it without breaking it.

#### Actual results
After reconnecting the USB, I am not able to reflash the microcontroller.

#### Versions
```
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "19.10 (Eoan Ermine)"
                   Kernel: Linux 5.3.0-22-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: "3.0.0"
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: ccache version 3.7.3
                    cmake: cmake version 3.13.4
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.26.2
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01000-gdb23c13d (2020-01-10-01:06)
                   python: Python 2.7.17rc1
                  python2: Python 2.7.17rc1
                  python3: Python 3.7.5rc1
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

benemorius, 2020-07-01:
I recall some talk about using caution when flashing `nrf52` because the onboard regulator defaults to 1.2V instead of 3.3V when the flash is cleared. Could that be the cause here?

Link: https://www.rototron.info/circuitpython-nrf52840-dongle-openocd-pi-tutorial/ (search for `RegOut0`)

Citrullin, 2020-07-02:
> I recall some talk about using caution when flashing `nrf52` because the onboard regulator defaults to 1.2V instead of 3.3V when the flash is cleared. Could that be the cause here?
> 
> Link: https://www.rototron.info/circuitpython-nrf52840-dongle-openocd-pi-tutorial/ (search for `RegOut0`)

"Warning:  wiping the chip will reset the voltage regulator. " I already some nrf52 before which broke. I reseted them, so that can be the reason. I try the recovery guide. Thanks for the link. Looks really useful. Maybe that unbricks it. If so, I think we need to add this to the docs.

maribu, 2020-07-03:
Citing: https://github.com/blacksphere/blackmagic/issues/532#issuecomment-542604161

> Please test again, only flashing the code without full erase and without touching the fuses. That way the IO Voltage is kept and things should work.

Unless I overlooked something, `make PROGRAMMER=bmp flash` should not perform a full erase.

However, I'm not even able to flash anything on my nrf52840dongle using BMP v1.6.1, so I cannot verify.

Citrullin, 2020-07-05:
I was so brave and tried another one and it just works, without any issues. So, I can't reproduce it... I don't get it...

closed 2020-07-05 by Citrullin

maribu, 2020-07-05:
Thanks for testing this.

Btw: While trying to reproduce the issue I read some entries in forums that people got the 1.8V issue *without* explicitly performing a full erase. Maybe that was just human error and the did accidentally trigger a full erase, or some bug messed with the `RegOut0` register. In any case: They were able to unbrick their board using either an external logic level shifter or a debugger/programmer with an integrated logic level shifter to restore the `RegOut0` settings. So this would be my first bet, too.

If you have a "real" black magic probe (the official hardware) or an Atmel ICE at hand, you already have programmer with an integrated logic level shifter that works for 1.8V.

Citrullin, 2020-07-05:
> Thanks for testing this.
>
> Btw: While trying to reproduce the issue I read some entries in forums that people got the 1.8V issue _without_ explicitly performing a full erase. Maybe that was just [...]

No, I don't have a real one. Only the Bluepill. Once I figured it out, I will add it to the documentation. So that people with the same issue can fix it as well. Tried the logic level shift register already, but it didn't work with the jlink. But I will give it another try. I just want to continue with my main topic ^^ :D