RIOT-OS/RIOT #14424: adc is not a ADC-Driver but a analog pin abstraction

kfessel, 2020-07-02:
#### Description

adc is not a ADC-Driver but a analog pin abstraction.
At least for STM the adc also manages the pins making it a analog_in abstraction, this makes it quiet difficult to uses for its other applications like measuring AVCC, or the chips internal temperature.

maribu, 2020-07-03:
You might want to have a look at https://github.com/RIOT-OS/RIOT/pull/13247 - I think the proposed API there might fit your needs better.

kfessel, 2020-07-13:
The adc-ng is better than the old API but it does not adress this issue. which is that an analog_pin and the ADC are not the same.
An Analog Pin may use the internal ADC (or a bus to speak to an external Analog Pin) the ADC has more features  (which are partialy adressed by the new implementation (faling to devide ADC from Analog Pin))

Analog Pin may also be Analog Input (or even Output) its an other Abstraction Layer from ADC (but the ADC has to mange this single resource (and do it well))

maribu, 2020-07-13:
> which is that an analog_pin and the ADC are not the same.

Neither the old API nor the new API have pins as entity in the API. In the old API there are "lines", in the new APIs there are "channels". Both *can* correspond to a GPIO routed to the ADC, but also to an internal input. Both APIs have implementations of a line/channel mapping to an internal input (thus, no analog pin).

kfessel, 2020-07-14:
> Neither the old API nor the new API have pins as entity in the API. In the old API there are "lines", in the new APIs there are "channels". Both _can_ correspond to a GPIO routed to the ADC, but also to an internal input. 

And there in is the problem: they hide the fact that they are some wierd kind of analog_in (analogRead for Arduino users) (the old one is even worse talking about lines which relate to nothing (not a pin on the board, not a pin on the mcu, not a channel on the adc (you have to lookup the board configuration), 

And they lie in their name that they are abtractions of the ADC.

> Both APIs have implementations of a line/channel mapping to an internal input (thus, no analog pin).

in the old case there it seems like a workaround to an API (the is a line which has a channel but we do not provide the pin even though it is part of the API)

for the new one it is:  you said you want to use the channel let me reconfigure the GPIO for you.

How about the esp32 with a analog mux in front at the board there is still one ADC, one channel but at board level it would have multiple analogIn (of cause your API can some how handle that it you have to work around) .

maribu, 2020-07-14:
> How about the esp32 with a analog mux in front at the board there is still one ADC, one channel but at board level it would have multiple analogIn (of cause your API can some how handle that it you have to work around) .

Can be easily implemented by implementing an `adc_ng_driver_t` that sits on top of the `adc_ng_driver_t` and uses the external mux to select the channel :-)

kfessel, 2020-07-14:
> (of cause your API can some how handle that it you have to work around)

maribu, 2020-07-14:
But that is way better than "some how handle that" ;-).

The `adc_ng_driver_t` would just initialize the function pointers to point to the implementation of the underlying `adc_ng_driver_t` except for the function initializing the channel. Thus, this could be implemented with few lines of code and no overhead for all but the initialization function. If the `adc_ng_driver_t` if the underlying ADC is never referred to, it would be optimized out at link time and only the multiplexed `adc_ng_driver_t` would be linked in.

I doubt that there is much efficiency ore elegance to gain by splitting out the channel setup of the ADC API compared to this approach.

maribu, 2023-05-19:
We now have adc "lines" mapping to internal analog signals for a number of boards, including STM. (Even without ADC NG.) I think the issue can be considered as solved

closed 2023-05-19 by maribu