RIOT-OS/RIOT #14439: (almost solved) SPI SD-Card driver: SPI initialisation freeze until timeout

dpenard, 2020-07-05:
#### Description
Implementation of a SD Card with STM32L073RZ. (It works under mbed os and now under RIOT).
-- BUT --
using pkg_fatfs with R1_POLLING_RETRY_CNT equal to 1000000 as it is defined in sdcard_spi_internal.h cause a timeout.

#### Steps to reproduce the issue
Change the R1_POLLING_RETRY_CNT value to 1000000 in sdcard_spi_internal.h
It doesn't work
Change it to 10 or less: it works

#### Expected results
Get a working filesystem with SD Card over SPI

#### Actual results
Putty terminal freeze in a long long wait before getting giving hand to shell.

#### Versions
Operating System Environment
-----------------------------
Operating System: "Ubuntu" "16.04.6 LTS (Xenial Xerus)"
Kernel: Linux 4.4.0-184-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609
arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Installed compiler libs
-----------------------
arm-none-eabi-newlib: "3.1.0"

Installed development tools
---------------------------
git: git version 2.7.4
make: GNU Make 4.1
openocd: Open On-Chip Debugger 0.10.0+dev-01289-g8833c88-dirty (2020-06-25-18:10)
python: Python 2.7.12
python2: Python 2.7.12
python3: Python 3.5.2
flake8: error: /usr/bin/python3: No module named flake8

MichelRottleuthner, 2020-07-06:
Thanks for the detailed report. Same as #11388 this should be fixed with #14420, right?
@dpenard can you confirm this issue is solved in master?

benpicco, 2020-10-01:
So what is missing here?

maribu, 2022-09-08:
Judging by the lack of activity in the issue and that generally SD card do work, the problem has been solved.

closed 2022-09-08 by maribu