RIOT-OS/RIOT #14466: Linking C++

sqrtroot, 2020-07-08:
#### Description
When linking the usage of c++ is not always detected correctly, and thus the wrong linker is used resulting in linker errors.

#### Steps to reproduce the issue
Folder structure like this:
```
├── main.c
├── Makefile
└── src
    ├── blink.cpp
    └── Makefile
```

Here's the contents of the files:
[example.tar.gz](https://github.com/RIOT-OS/RIOT/files/4891522/example.tar.gz)
(If you want to try this you'll have to set the correct path for RIOT)

#### Expected results
Either g++ is used as a linker (which is down when there is an cpp file in the main application) or libstdc++ is also included in the link process when ld is used.

#### Actual results
gcc is used as a linker which doesn't link against libstdc++. Just adding a single empty file ending in .cpp makes RIOT use g++ as a linker

#### Versions
```

Operating System Environment
----------------------------
         Operating System: "Arch Linux" 
                   Kernel: Linux 5.7.7-arch1-1 x86_64 unknown
             System shell: GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 10.1.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 10.1.0
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 10.0.0 

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: ccache version 3.7.9
                    cmake: cmake version 3.17.3
                 cppcheck: Cppcheck 2.1
                  doxygen: missing
                      git: git version 2.27.0
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 3.8.3
                  python2: Python 2.7.18
                  python3: Python 3.8.3
                   flake8: error: /sbin/python3: No module named flake8
               coccinelle: missing
```

sqrtroot, 2020-07-08:
A possible solution would be; always using g++ as the linker, or using ld itself correctly (linking against libstdc++ and newlib etc)
AFAIK linking only c files with g++ doesn't have any impact on performance etc. So could be an easy workaround

sqrtroot, 2020-07-09:
Okay so after looking trough some makefiles I found `FEATURES_REQUIRED += cpp` this solves this issue.
When settings this in either the main make file or the makefile in src, it chooses the correct linker.

I think this might be important to talk about somewhere in the documentation. Together with settings things like which standard to use etc. Which also was a big search for me.

Besides the documentation I still don't think it would be a problem always using g++ as the linker.

maribu, 2020-07-09:
> Okay so after looking trough some makefiles I found `FEATURES_REQUIRED += cpp` this solves this issue.

I thought so at first as well. However, with current master and the example you have given, linking still fails for me. Can you confirm that this indeed solves the issue for you?

> Besides the documentation I still don't think it would be a problem always using g++ as the linker.

For platforms supporting C++ this might be true. But e.g. if someone wants to port RIOT to a platform for which only a C compiler is available, this change would be a major issue. This is why `cpp` is a feature in the first place, that is supported by most boards, but not necessarily by all. (E.g. in addition to the availability of the toolchain, all statically allocated class instance want their constructor to be called during startup. This might not be implemented for some platform.)

closed 2020-07-28 by benpicco