RIOT-OS/RIOT #14472: Proposal to unify MQTT-SN configurations in Kconfig

akshaim, 2020-07-09:
#### Description

[Asymcute](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/asymcute.h) and [emcute](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/emcute.h) are MQTT-SN clients available in RIOT with similar configurations but for different use cases.  The proposal looks at modelling the client's configurations to make it module agnostic similar to CoAP. This could be done by moving the configurations to [mqttsn.h](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/mqttsn.h) and leaving module-specific configurations if any in respective headers. 

The difference between asumcute and emcute is that one has ASYMCUTE_KEEPALIVE_PING. Also XX_TOPIC_MAXLEN configuration is slightly different . Default port configuration  (EMCUTE_DEFAULT_PORT) in emcute is left unused hence that may be skipped.

I also assume that both emcute and asymcute won't be used at the same time. If they need to be used simultaneously we may keep things as it is, but how likely is it? 

### Useful links

Asymcute was already exposed
- [x] sys/include/net/asymcute.h : [#14444](https://github.com/RIOT-OS/RIOT/pull/14444)
    - [x] [~~ASYMCUTE_BUFSIZE~~](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/asymcute.h#L136-L141)
    - [x] [~~ASYMCUTE_TOPIC_MAXLEN~~](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/asymcute.h#L76-L83)
    - [x] [~~ASYMCUTE_KEEPALIVE~~](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/asymcute.h#L85-L94)
    - [x] [~~ASYMCUTE_KEEPALIVE_PING~~](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/asymcute.h#L96-L106)
    - [x] [~~ASYMCUTE_T_RETRY~~](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/asymcute.h#L108-L120)
    - [x] [~~ASYMCUTE_N_RETRY~~](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/asymcute.h#L122-L133)

Emcute is pending exposure.

- [ ]  sys/include/net/emcute.h
    - [ ] [EMCUTE_DEFAULT_PORT ](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/emcute.h#L98-L105) - May be skipped
    - [ ] [EMCUTE_BUFSIZE](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/emcute.h#L105-L115)
    - [ ] [EMCUTE_TOPIC_MAXLEN](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/emcute.h#L117-L125)
    - [ ] [EMCUTE_KEEPALIVE](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/emcute.h#L139-L146)
    - [ ] [EMCUTE_T_RETRY](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/emcute.h#L139-L146)
    - [ ] [EMCUTE_N_RETRY](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/emcute.h#L148-L155)

@miri64 @haukepetersen

miri64, 2020-09-29:
Mhhh... One could imagine that one might want to use both clients (with different ports) for some reason, so it might make more sense to keep the static configurations distinct.

akshaim, 2020-09-29:
Oh okay. But that could be rarest of the rare case right ? or am I missing something?

miri64, 2020-09-29:
Yes, and it might not be too sensible. I was arguing more on the basis of why it would make sense to keep the config separate. IMHO those two modules should be seen as what they are: different implementations of the same protocol. You wouldn't use the same config for two different browsers either.

akshaim, 2020-09-30:
Oh okay. Then I shall skip it.

miri64, 2020-09-30:
Then can we close this issue?

akshaim, 2020-09-30:
Done.

closed 2020-09-30 by akshaim