RIOT-OS/RIOT #14505: Add the "riot-os" topic to the repository description

jhaand, 2020-07-13:
#### Description
At the moment a Github topics in the about field for RIOT-OS is not present. I can find projects with the topics "riotos" and "riot-os". Could you add the topic "riot-os" to the description of the pinned repositories (RIOT, Tutorials, applications). 
This makes searching for RIOT-OS applications much easier. This allows people to start faster with their application. Adding this topic also creates a uniform topic name for people to adapt. 

### Useful links
https://github.com/RIOT-OS/RIOT
https://github.com/RIOT-OS/Tutorials
https://github.com/RIOT-OS/applications
[image]

wosym, 2020-07-17:
I suggest also adding related tags and de-facto names for RIOT-os.
e.g. "rtos", "riotos", "riot"

jhaand, 2020-07-26:
You could also mention the other tags. That makes it easier to click through. But mention somewhere which tags remains the preferred one.

miri64, 2020-07-27:
Good point. I'm going to update the respective repositories.

closed 2020-07-27 by miri64

jhaand, 2020-07-27:
Thnx. 
I'm glad this is now closed.