RIOT-OS/RIOT #14506: mdt_erase success, but vfs_format resets board (esp32-heltec-lora32-v2)

remiphilippe, 2020-07-13:
#### Description
Cannot format using vfs_format on Heltec Lora v2 (esp32-heltec-lora32-v2)

#### Steps to reproduce the issue
I'm following the SPIFFs example (https://github.com/RIOT-OS/RIOT/tree/master/examples/filesystem) and test (https://github.com/RIOT-OS/RIOT/blob/master/tests/pkg_spiffs/main.c)

Makefile (content related to fs):
```
BOARD ?= esp32-heltec-lora32-v2
# filesystem
USEMODULE += mtd
USEMODULE += vfs
USEMODULE += esp_spiffs
```

Code (related to fs):
```
#define _dev (MTD_0)

/* Flash mount point */
#define FLASH_MOUNT_POINT   "/sda"

/* include file system header */
#include "vfs.h"
#include "mtd.h"
#include "fs/spiffs_fs.h"

/* file system specific descriptor
 * as for littlefs, some fields can be changed if needed,
 * this example focus on basic usage, i.e. entire memory used */
static spiffs_desc_t spiffs_desc = {
    .lock = MUTEX_INIT,
};

static vfs_mount_t spiffs_mount = {
    .fs = &spiffs_file_system,
    .mount_point = FLASH_MOUNT_POINT,
    .private_data = &spiffs_desc,
};

/* Command handlers */
static int _mount(int argc, char **argv)
{
    (void)argc;
    (void)argv;
    int res = vfs_mount(&spiffs_mount);
    if (res < 0) {
        printf("Error while mounting %s...try format\n", FLASH_MOUNT_POINT);
        return 1;
    }

    printf("%s successfully mounted\n", FLASH_MOUNT_POINT);
    return 0;
}

static int _format(int argc, char **argv)
{
    (void)argc;
    (void)argv;
    // struct statvfs vstat;

    // if(vfs_statvfs(FLASH_MOUNT_POINT, &vstat) < 0) {
    //     printf("Error stat for %s\n", FLASH_MOUNT_POINT);
    //     //return 1;
    // } else {
    //     printf("FSID: %lu\n", vstat.f_fsid);
    // }

    printf("erasing mtd...\t");
    if(mtd_erase(_dev, 0, _dev->page_size * _dev->pages_per_sector * _dev->sector_count) != 0) {
[... 165 lines cut ...]
                "sha256:ddc500d8499442f954ee7184ef015b5041629e55bf8b336304c9614154008bcd",
                "sha256:481fa7102edea497243b81816119865829429105602b529357d5ec3efc081790",
                "sha256:f478f691ced81e48f8ee65de9edc5bf9bb4da390c3c14d9b7678ab8911005672",
                "sha256:d46b5cd0d3d2959211188b28599003c3a013a3a40b6fad32c8e5d63464a583b6",
                "sha256:47f4a94db7ec4855d150f0bb0a790c50e43beca4ebd45f9b9642b700fc8f351c",
                "sha256:b65ff9a2b045067e80ccd77de462c4e1be18925fa8220f8fd4d6da05c89c2950",
                "sha256:13162280875643f21dc040a1669af391383bdcd877a8199f66d5201e621a7633",
                "sha256:6368aa4a566f5a46d5f531ba50b62573f0b2d8887716ff53d60d39cccf29daaf",
                "sha256:22eee7ca8a2a3db73830a1d505c98ea1e16932eb290122d5913800fcb2c5a9d7",
                "sha256:95a70fcd9693785f0b3149eb53c05b2e54134092cc6c16ca110f202e3725179b",
                "sha256:88be8a29411e7fd0fc322957a96530b2814fc4ea410d2ae14c712be2b5ea4347",
                "sha256:f0c9a582de5080bb0fe5961940b25f5fb655bcfda59b212656862163d14c8e51",
                "sha256:8c6fcdad33f3be449f0b6f0f675ef165cc6eaa30f3e4aaff8d504f8362146dc6",
                "sha256:b32fe7329011273c6053f9fa8be0fe60de88ecc6065e3e1d7a9e6393b2508726",
                "sha256:cf55e64673e3c168ee1198142922aca8920d15069fee6f90b4ece8a0763732a8",
                "sha256:93e37531002a8902eb86b2bbeea573f216ada0fd683f71d050097b5b70493a90",
                "sha256:aa83b7f73175af0ae45435653fbdc5b8c82d4d81d29c55c4c8f3fc8dda0847db"
            ]
        },
        "Metadata": {
            "LastTagTime": "0001-01-01T00:00:00Z"
        }
    }
]
```

remiphilippe, 2020-07-13:
Just to add, I just tried running `make debug` then `make flash-only`, and it seems to work:
```
> format
2020-07-13 12:07:55,316 #  format
2020-07-13 12:08:15,849 # erasing mtd...	[OK]
2020-07-13 12:10:57,513 # formating /sda....	[OK]
```

remiphilippe, 2020-07-14:
It seems to exhibit the same behavior as here: https://github.com/RIOT-OS/RIOT/pull/11449#issuecomment-522617986

gschorcht, 2020-07-15:
Hm, the problem described in https://github.com/RIOT-OS/RIOT/pull/11449#issuecomment-522617986 was fixed with #12035.

> Just to add, I just tried running `make debug` then `make flash-only`, and it seems to work:

This does not really make sense, since `make debug` does nothing for ESP32. Maybe you just observed that it crashed on the first try, but worked on the second try because the flash already had a different content.

Have you tried to erase the flash completely
```
dist/tools/esptool/esptool.py --port /dev/ttyUSB0 erase_flash
```
before you flash your program?

gschorcht, 2020-07-15:
> I'm following the SPIFFs example (https://github.com/RIOT-OS/RIOT/tree/master/examples/filesystem) and test (https://github.com/RIOT-OS/RIOT/blob/master/tests/pkg_spiffs/main.c)

Do these programs work for you with this board?

gschorcht, 2020-07-15:
I tried your code which works for me for the board.
```
> format
erasing mtd...	[OK]
formating /sda....	[OK]
```
So I can't reproduce the problem.

remiphilippe, 2020-07-20:
Thanks for looking into it.
It failed about 50 times before I posted here, then `make debug` and it worked like magic, but as you mentioned could be pure luck if `make debug` doesn't really impact ESP32.

I can't remember if I used the `erase_flash` on this board, but I can't seem to be able to reproduce since it worked.

I'll try when I order the same board again (in a week or so) and report here, for now, I think we can close this as unreproducible.

closed 2026-10-20 by crasbe