RIOT-OS/RIOT #14545: SAMD21 can lockup and not wake from sleep

keestux, 2020-07-17:
#### Description
On several platforms[1] there were bug reports about SAMD21 locking up in sleep and not waking up anymore. If the watchdog is enabled it would reset the MCU (luckily). It goes back to 2016, and perhaps earlier when people started to see this.

#### Steps to reproduce the issue
Well, this is tricky. It is pretty hard to make a test that just fails. You basically need an application that goes to deep sleep, and wake up by interrupts (timer, peripheral, pins, etc). Do some work and go back to sleep.

#### Expected results
The MCU should wake up from sleep by regular interrupt sources (hardware, timer, etc).

#### Actual results
The MCU is not woken up by an interrupt.

I don't have actual RIOT application where this shows up. But we, at SODAQ, have suffered from this bug with Arduino applications. The sad story about it is, that we reported this to the chip manufacturer, however they never bothered to inform us with the work around. The work around, see [2] and function `sleep` in [3], turns out quite effectively. Our application died roughly once a day, and now it is already running for days.

[1] https://www.avrfreaks.net/forum/samd21-samd21e16b-sporadically-locks-and-does-not-wake-standby-sleep-mode
[2] https://www.avrfreaks.net/comment/2616121#comment-2616121
[3] https://github.com/arduino-libraries/ArduinoLowPower/blob/master/src/samd/ArduinoLowPower.cpp

keestux, 2020-07-17:
Here is a short summary of the code with the workaround
```
// Disable systick interrupt
SysTick->CTRL &= ~SysTick_CTRL_TICKINT_Msk;	
SCB->SCR |= SCB_SCR_SLEEPDEEP_Msk;
__DSB();
_WFI();
// Enable systick interrupt
SysTick->CTRL |= SysTick_CTRL_TICKINT_Msk;	
```

keestux, 2020-07-17:
@benpicco @aabadie Maybe this is of interest to you too, since you have been (and maybe still are) using SAMD21. I have no idea if this problem only exists in SAMD21. I haven't yet seen any information from Microchip.

The problem needs to be solved in `cpu/cortexm_common/include/cpu.h`, however that is common for all Cortex-M.

benpicco, 2020-07-17:
Well RIOT doesn't use SysTick (yet) so we might not be affected by this.

keestux, 2020-07-19:
But when we do use SysTick (PR #14553) ...

benpicco, 2020-07-19:
Yea :smiley: 
If this only affects sam0, we can plug the work-around (manually stopping SysTick before entering sleep) into `sam0_cortexm_sleep()`.
That's where the other work-around wrt sleep live anyway.

keestux, 2020-07-20:
It would be really helpful if Microchip would publish errata for this. Right now it is unclear if this just affects SAMD21.

closed 2021-03-19 by stale[bot]