RIOT-OS/RIOT #14548: edbg: long lines flooded over serial become garbled

bergzand, 2020-07-17:
#### Description

The virtual COM port provided by the EDBG debuggers doesn't correctly handled rapid bulk transfers from the host computer. This becomes apparent when using a local setup. A remote setup or a slow host such as a raspberry pi doesn't flood the USB serial fast enough to cause issues. That's also why it doesn't fail on the nightlies.

Bisecting turns up this commit: https://github.com/RIOT-OS/RIOT/commit/68a35f53de2c881f82edb09ae5cd6f0d5a5f22ee, but that only uncovered the issue.

#### Steps to reproduce the issue

*Locally* run `tests/shell` on a board using EDBG such as the samr21-xpro or the same54-xpro. This fails on the long line test, where it starts to produce garbage chars.

#### Expected results

The test succeeds both locally and on Murdock/IoT-LAB.

#### Actual results

The test fails after the EDBG serial produces garbage instead of the long line:

<details><summary>local node <b>FAIL</b></summary>

```
$ BOARD=samr21-xpro make -C tests/shell/ flash test
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/shell'
Building application "tests_shell" for "samr21-xpro" with MCU "samd21".

"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/boards/samr21-xpro
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/core
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/cpu/samd21
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/cpu/cortexm_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/cpu/cortexm_common/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/cpu/sam0_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/cpu/sam0_common/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/cpu/samd21/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/drivers/periph_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys/app_metadata
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys/auto_init
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys/isrpipe
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys/newlib_syscalls_default
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys/pm_layered
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys/ps
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys/shell
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys/shell/commands
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys/stdio_uart
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT-release/sys/tsrb
   text	   data	    bss	    dec	    hex	filename
  12848	    132	   2624	  15604	   3cf4	/home/mlenders/Repositories/RIOT-OS/RIOT/tests/shell/bin/samr21-xpro/tests_shell.elf
/home/mlenders/Repositories/RIOT-OS/RIOT-release/dist/tools/edbg/edbg  --target samr21 --verbose --file /home/mlenders/Repositories/RIOT-OS/RIOT/tests/shell/bin/samr21-xpro/tests_shell.bin --verify || /home/mlenders/Repositories/RIOT-OS/RIOT-release/dist/tools/edbg/edbg  --target samr21 --verbose --file /home/mlenders/Repositories/RIOT-OS/RIOT/tests/shell/bin/samr21-xpro/tests_shell.bin --verify --program
Debugger: ATMEL EDBG CMSIS-DAP ATML2127031800005161 01.1A.00FB (S)
Clock frequency: 16.0 MHz
Target: SAM R21G18 (Rev C)
Verification...................................................... done.

socat - open:/dev/ttyACM0,b115200,echo=0,raw
�	"R�
shell: incorrect quoting
> bufsize
x�
> 
> bufsize
128
> ________________________________________________________________________________________________________________________________verylong
________________________________________________________________�8����A 8
                                                                         	$�`�@��^�吧���!:۳
shell: command not found: ________________________________________________________________�8����A
> -j1/��$n�H�"<P)�I�{mː�v�?DnTimeout in expect script at "child.expect('shell: maximum line length exceeded')" (../RIOT/tests/shell/tests/01-run.py:150)

make: *** [/home/mlenders/Repositories/RIOT-OS/RIOT-release/Makefile.include:770: test] Error 1
make: Leaving directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/shell'
[... 152 lines cut ...]
> help
help
Command              Description
---------------------------------------
bufsize              Get the shell's buffer size
start_test           starts a test
end_test             ends a test
echo                 prints the input command
reboot               Reboot the node
version              Prints current RIOT_VERSION
pm                   interact with layered PM subsystem
ps                   Prints information about running threads.
app_metadata         Returns application metadata
> reboot
reboot
main(): This is RIOT! (Version: 2020.10-devel-18-gab072-2020.07-branch)
test_shell.
> end_test
end_test
[TEST_END]
> 
make: Leaving directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/shell'
```

</details>

miri64, 2020-07-17:
See also discussion in https://github.com/RIOT-OS/Release-Specs/issues/172#issuecomment-660120288 and the following comments.

miri64, 2020-07-17:
It was also noted that this might be related to https://github.com/RIOT-OS/RIOT/issues/10639

maribu, 2023-05-19:
Nope, on my samr21-xpro a freshly updated edbg firmware still won't work with long shell command lines.