RIOT-OS/RIOT #14555: periph flashpage bug

MyCodeMyMoney, 2020-07-19:
#### Description

When calling the function flashpage_write_raw(void *target_addr, const void *data, size_t len), if the address of my input parameter data is not divisible by 4, I will get an assertion error. The assertion is 
```
assert(!(((unsigned)target_addr % FLASHPAGE_RAW_ALIGNMENT) || ((unsigned)data % FLASHPAGE_RAW_ALIGNMENT)));
```
In addition, I would like to ask a question. When an assertion error occurs, I cannot see a detailed assertion description, so this increases my time for troubleshooting. What should I do to get a detailed assertion description?

#### Steps to reproduce the issue

Define a global array: static char data[10] = {0},Pass data+1 as the parameter data, and then you will get an assertion error, because the address you passed in is not divisible by 4。

#### Expected results

The test was successfully tested on the stm32f103 platform. When I write data to the specified flash page through the flashpage_write function, if the start address of the data I want to write is not divisible by 4, I will get an assertion error.

#### Actual results

When calling the function flashpage_write_raw(void *target_addr, const void *data, size_t len), if the address of my input parameter data is not divisible by 4, I will get an assertion error. 

#### Version
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "20.04 LTS (Focal Fossa)"
                   Kernel: Linux 5.4.0-40-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: missing
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.25.1
                     make: GNU Make 4.2.1
                  openocd: missing
                   python: Python 3.8.2
                  python2: Python 2.7.18rc1
                  python3: Python 3.8.2
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing

bergzand, 2020-07-19:
> When calling the function flashpage_write_raw(void *target_addr, const void *data, size_t len), if the address of my input parameter data is not divisible by 4, I will get an assertion error. The [...]

This is usually a limitation of the underlying hardware. Most MCU's only support word-aligned writes to the internal flash and anything else is either undefined behavior or will throw an runtime exception. The assertion here prevents that during development.

> In addition, I would like to ask a question. When an assertion error occurs, I cannot see a detailed assertion description, so this increases my time for troubleshooting. What should I do to get a detailed assertion description?

Adding [`CFLAGS += -DDEBUG_ASSERT_VERBOSE`](https://riot-os.org/api/assert_8h.html#a2200149ba880bf26fed140bdcf318113) to your Makefile will turn on printing the filename and line number for assertions.

MyCodeMyMoney, 2020-07-20:
> > When calling the function flashpage_write_raw(void *target_addr, const void *data, size_t len), if the address of my input parameter data is not divisible by 4, I will get an assertion error. The assertion is
> > ```
> > assert(!(((unsigned)target_addr % FLASHPAGE_RAW_ALIGNMENT) || ((unsigned)data % FLASHPAGE_RAW_ALIGNMENT)));
> > ```
> 
> This is usually a limitation of the underlying hardware. Most MCU's only support word-aligned writes to the internal flash and anything else is either undefined behavior or will throw an runtime exception. The assertion here prevents that during development.
> 
> > In addition, I would like to ask a question. When an assertion error occurs, I cannot see a detailed assertion description, so this increases my time for troubleshooting. What should I do to get a detailed assertion description?
> 
> Adding [`CFLAGS += -DDEBUG_ASSERT_VERBOSE`](https://riot-os.org/api/assert_8h.html#a2200149ba880bf26fed140bdcf318113) to your Makefile will turn on printing the filename and line number for assertions.
>>The assertion requires the word-aligned of the flash address that I write to I think it is correct, but is it reasonable to require that my data parameter address must be word-aligned ? If I define a structure, I just want to write some specific data into the flash, the starting address of this data may not meet the requirements of the assertion.The parameter data requires word-aligned, is it to prevent errors from occurring?

closed 2021-03-19 by stale[bot]