RIOT-OS/RIOT #14568: tests/pkg_libfixmath_unittests: failing on multiple platforms

fjmolinas, 2020-07-21:
#### Description

`tests/pkg_libfixmath_unittests` fails, specifically testcases for division. Found while running https://github.com/RIOT-OS/Release-Specs/issues/172 on multiple boards, see https://github.com/miri64/RIOT-2020.07-results/tree/master/2020.07-RC2.

#### Steps to reproduce the issue

<details><summary>BOARD=samr21-xpro BUILD_IN_DOCKER=1 make -C tests/pkg_libfixmath_unittests/ flash test
</summary>

```
...
...

----Running testcases for division----

1.000000 / -32768.000000 = -0.000031
1.000000 / -32768.000000 = -0.000031

-1.000000 / -32768.000000 = 0.000031
-1.000000 / -32768.000000 = 0.000031

2.000000 / -32768.000000 = -0.000061
2.000000 / -32768.000000 = -0.000061

-2.000000 / -32768.000000 = 0.000061
-2.000000 / -32768.000000 = 0.000061

3.000000 / -32768.000000 = -0.000092
3.000000 / -32768.000000 = -0.000092

-3.000000 / -32768.000000 = 0.000092
-3.000000 / -32768.000000 = 0.000092

4.000000 / -32768.000000 = -0.000122
4.000000 / -32768.000000 = -0.000122

-4.000000 / -32768.000000 = 0.000122
-4.000000 / -32768.000000 = 0.000122

5.000000 / -32768.000000 = -0.000153
5.000000 / -32768.000000 = -0.000153

-5.000000 / -32768.000000 = 0.000153
-5.000000 / -32768.000000 = 0.000153

6.000000 / -32768.000000 = -0.000183
6.000000 / -32768.000000 = -0.000183

-6.000000 / -32768.000000 = 0.000183
-6.000000 / -32768.000000 = 0.000183

0.500000 / -32768.000000 = -0.000015
0.500000 / -32768.000000 = -0.000015

-0.500000 / -32768.000000 = 0.000015
-0.500000 / -32768.000000 = 0.000015

0.999985 / -32768.000000 = -0.000031
0.999985 / -32768.000000 = -0.000031

-0.999985 / -32768.000000 = 0.000031
-0.999985 / -32768.000000 = 0.000031

1.999985 / -32768.000000 = -0.000061
1.999985 / -32768.000000 = -0.000061

-1.999985 / -32768.000000 = 0.000061
-1.999985 / -32768.000000 = 0.000061

3.999985 / -32768.000000 = -0.000122
3.999985 / -32768.000000 = -0.000122

-3.999985 / -32768.000000 = 0.000122
-3.999985 / -32768.000000 = 0.000122

32767.999985 / -32768.000000 = -1.000000
32767.999985 / -32768.000000 = -1.000000

-32768.000000 / -32768.000000 = 1.000000
-32768.000000 / -32768.000000 = 1.000000

12693.159363 / -32768.000000 = -0.387360
12693.159363 / -32768.000000 = -0.387364

8770.674332 / -32768.000000 = -0.267654
8770.674332 / -32768.000000 = -0.267660

32764.774979 / -32768.000000 = -0.999908
32764.774979 / -32768.000000 = -0.999902

-7158.829559 / -32768.000000 = 0.218475
-7158.829559 / -32768.000000 = 0.218470

-14673.028793 / -32768.000000 = 0.447784
-14673.028793 / -32768.000000 = 0.447785

28106.082672 / -32768.000000 = -0.857727
28106.082672 / -32768.000000 = -0.857730

30406.164383 / -32768.000000 = -0.927917
30406.164383 / -32768.000000 = -0.927922

7941.976990 / -32768.000000 = -0.242371
7941.976990 / -32768.000000 = -0.242370

8552.903992 / -32768.000000 = -0.261017
8552.903992 / -32768.000000 = -0.261014

-31967.057037 / -32768.000000 = 0.975555
-31967.057037 / -32768.000000 = 0.975557

-13380.394638 / -32768.000000 = 0.408340
-13380.394638 / -32768.000000 = 0.408337

-1034.753860 / -32768.000000 = 0.031586
-1034.753860 / -32768.000000 = 0.031578

32748.826462 / -32768.000000 = -0.999420
32748.826462 / -32768.000000 = -0.999415

7779.875778 / -32768.000000 = -0.237427
7779.875778 / -32768.000000 = -0.237423

-16381.433975 / -32768.000000 = 0.499924
-16381.433975 / -32768.000000 = 0.499922

0.465698 / -32768.000000 = -0.000015
0.465698 / -32768.000000 = -0.000014

-1.275436 / -32768.000000 = 0.000046
-1.275436 / -32768.000000 = 0.000039

0.627579 / -32768.000000 = -0.000015
0.627579 / -32768.000000 = -0.000019

0.642960 / -32768.000000 = -0.000015
0.642960 / -32768.000000 = -0.000020

0.893204 / -32768.000000 = -0.000031
0.893204 / -32768.000000 = -0.000027

1.283905 / -32768.000000 = -0.000046
1.283905 / -32768.000000 = -0.000039

-0.431564 / -32768.000000 = 0.000015
-0.431564 / -32768.000000 = 0.000013

1.387238 / -32768.000000 = -0.000046
1.387238 / -32768.000000 = -0.000042

0.303116 / -32768.000000 = -0.000015
0.303116 / -32768.000000 = -0.000009

0.506454 / -32768.000000 = -0.000015
0.506454 / -32768.000000 = -0.000015

1.248840 / -32768.000000 = -0.000031
1.248840 / -32768.000000 = -0.000038

-1.011246 / -32768.000000 = 0.000031
-1.011246 / -32768.000000 = 0.000031

-0.964584 / -32768.000000 = 0.000031
-0.964584 / -32768.000000 = 0.000029

-0.678391 / -32768.000000 = 0.000015
-0.678391 / -32768.000000 = 0.000021

1.286240 / -32768.000000 = -0.000046
1.286240 / -32768.000000 = -0.000039

0.725021 / -32768.000000 = -0.000015
0.725021 / -32768.000000 = -0.000022

-0.600037 / -32768.000000 = 0.000015
-0.600037 / -32768.000000 = 0.000018
FAILED: fix16_unittests.c:194 failures == 0

...
...

Some tests FAILED!

```
</details>

#### Expected results

Tests succeeds.

#### Actual results

Test fails.

#### Versions

- Tested with BUILD_IN_DOCKER

- Local toolchains

<details><summary>make print-versions</summary>

```
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "18.04.4 LTS (Bionic Beaver)"
                   Kernel: Linux 5.3.0-62-generic x86_64 x86_64
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
               msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
     riscv-none-embed-gcc: riscv-none-embed-gcc (GNU MCU Eclipse RISC-V Embedded GCC, 64-bit) 8.2.0
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: xtensa-esp8266-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.0.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: "3.0.0"
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: "2.2.0"
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.4.1
                    cmake: cmake version 3.14.0-rc3
                 cppcheck: Cppcheck 1.82
                  doxygen: 1.8.16
                      git: git version 2.27.0
                     make: GNU Make 4.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01100-g51dd4ce6-dirty (2020-03-03-15:33)
                   python: Python 3.6.9
                  python2: Python 2.7.17
                  python3: Python 3.6.9
                   flake8: 3.7.7 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.6.9 on Linux
               coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

</details>

miri64, 2020-07-22:
I noticed that this test has quite a long time-out waiting for the `SUCCESS` at the very end of that test. Might it make sense to rather wait and check for intermediate outputs with shorter timeouts? I suspect the fail is because the boards are shorter in general.

miri64, 2020-07-28:
Edited my last post, to prevent misunderstandings: the test is indeed failing and we should blacklist the boards in question / whitelist the succeeding boards not just for CI but also for the build system.

fjmolinas, 2020-10-07:
Investigating this again, the tests passes with `llvm`, `gcc 8.2.1`, but fails in docker `gcc 9.2.1`.

fjmolinas, 2020-10-07:
> Investigating this again, the tests passes with llvm, gcc 8.2.1, but fails in docker gcc 9.2.1.

Some division factors are clearly off, maybe we can blacklist gcc for that test?

fjmolinas, 2020-10-07:
> Some division factors are clearly off, maybe we can blacklist arm-gcc for that test?

Blacklist for `arm` cpu's?

aabadie, 2021-01-21:
> I suggest blacklisting gcc for arm boards

I agree!

aabadie, 2021-01-21:
Can we consider this as fixed by #15828 ?

closed 2021-01-27 by aabadie