RIOT-OS/RIOT #14578: tests/cpp1: failing on i-nucleo-lrwan1

fjmolinas, 2020-07-22:
#### Description

All cpp11 tests are failing on ` i-nucleo-lrwan1` they all hang right away/

#### Steps to reproduce the issue

` BOARD=i-nucleo-lrwan1 make -C tests/cpp11_condition_variable/ flash test`

` BOARD=i-nucleo-lrwan1 make -C tests/cpp11_mutex/ flash test`

 `BOARD=i-nucleo-lrwan1 make -C tests/cpp11_tthread/ flash term -j3`

#### Expected results

Test passes.

#### Actual results

- `tests/cpp1_condition_variable`

```
2020-07-22 14:01:21,614 # START
2020-07-22 14:01:21,618 # main(): This is RIOT! (Version: 2020.04)
2020-07-22 14:01:21,618 #
2020-07-22 14:01:21,623 # ************ C++ condition_variable test ***********
2020-07-22 14:01:21,626 # Wait with predicate and notify one ...
```

- `tests/cpp1_condition_mutex`

```
2020-07-22 13:59:58,041 # START
2020-07-22 13:59:58,044 # main(): This is RIOT! (Version: 2020.04)
2020-07-22 13:59:58,045 #
2020-07-22 13:59:58,048 # ************ C++ mutex test ***********
2020-07-22 13:59:58,050 # Lock and unlock ...
```

- `tests/cpp1_condition_thread`

```
2020-07-22 14:05:08,979 # START
2020-07-22 14:05:08,983 # main(): This is RIOT! (Version: 2020.04)
2020-07-22 14:05:08,983 #
2020-07-22 14:05:08,986 # ************ C++ thread test ***********
2020-07-22 14:05:08,991 # Creating one thread and passing an argument ...
```

#### Versions

Built in docker.

keestux, 2020-08-12:
@fjmolinas why do you use `%` in the title? Don't you mean `*`, which is common for "glob"?

fjmolinas, 2020-08-13:
> @fjmolinas why do you use % in the title? Don't you mean *, which is common for "glob"?

Yes, well `%` its the equivalent for makefiles, change if you find it more understandable.

keestux, 2020-08-13:
I see. However, `%` in makefiles is not equivalent to `*` wildcard (or glob).

In the title of this issue it makes more sense to use `*`, so I'll change it.

maribu, 2023-05-19:
I cannot reproduce the issues on any of my hardware and didn't see any issues in the last couple of release tests for any of the boards I tested on. I suspend an toolchain issue.

I'm closing this for now. If the issue can be reproduced, this issue can be reopened easily.

closed 2023-05-19 by maribu