RIOT-OS/RIOT #14618: tools: cleanup host-side network setup

miri64, 2020-07-27:
#### The Issue
With https://github.com/RIOT-OS/RIOT/pull/14454 now merged, we have a total of 3 `start_network.sh` scripts:

- The original one for [ethos](https://github.com/RIOT-OS/RIOT/blob/42eb044ec67758e1664dd354cf07652c9632aace/dist/tools/ethos/start_network.sh),
- A slightly improved, but [SLIP-specific version](https://github.com/RIOT-OS/RIOT/blob/42eb044ec67758e1664dd354cf07652c9632aace/dist/tools/sliptty/start_network.sh) (using `getopt` to parse the command-line parameters), and finally
- The new one for [USB CDC-ECM](https://github.com/RIOT-OS/RIOT/blob/42eb044ec67758e1664dd354cf07652c9632aace/dist/tools/usb-cdc-ecm/start_network.sh).

For new communication protocols to allow for network communication between host and node such as PPP, the potential for further such scripts popping up, exists.

Moreover, for `ethos` there is also the [`setup_network.sh` script](https://github.com/RIOT-OS/RIOT/blob/42eb044ec67758e1664dd354cf07652c9632aace/dist/tools/ethos/setup_network.sh) used to setup testing environments only once which not only copies a lot from the `ethos` version of the `start_network.sh` script, but also has some overlap with [`tapsetup`](https://github.com/RIOT-OS/RIOT/blob/42eb044ec67758e1664dd354cf07652c9632aace/dist/tools/tapsetup/tapsetup). `tapsetup` set's up TAP interfaces for `native` in a bridge and has the distinct advantage of supporting multiple platforms (Linux, Mac OSX, and FreeBSD) and being well tested on them.

Another issue is that both `tapsetup` and the `start_network.sh`/`setup_network.sh` scripts of `ethos` destroy each others environments if not used carefully, which can lead to a lot of frustration when running host-interaction based tests, like the `scapy` once on multiple boards including `native`.

Finally, it is quite annoying for testing, that the host-interaction between host and node can be quite different for `native` and `ethos`-based communication: `tapsetup` creates the TAP interfaces in a bridge, so multiple `native` instances can communicate among each other and with the host, `ethos`'s script just create a bare TAP interface. For the `native` case the host needs to communicate over the bridge's interface, for `ethos`-based tests, over the TAP interface. The first is the case since interfaces in a bridge are only reachable from the bridge on the host machine, and not directly accessible for reception of packages.

#### Proposed solution

1. Unify `start_network.sh` to use `setup_network.sh` which in turn uses `tapsetup` to create TAP interfaces including a bridge (or for SLIP and USB CDC-ECM: add respective interface to bridge).
   - the [uplink type](https://github.com/RIOT-OS/RIOT/pull/13802) could be provided as a parameter
2. Fold `native` as a new uplink type into that script.
3. Use `start_network.sh` as terminal program for all applications that require it.
4. (optional) Add state file (to `/tmp`) that keeps track of created interfaces and their associated bridges.

#### Issues solved as a side-effect

- `ethos`-based applications may not work on Arch Linux when used with raw `ethos` as the interface needs to brought up manually before starting the application (see https://github.com/RIOT-OS/RIOT/pull/13802): The outright usage of `start_network.sh` sets the TAP interface to `up`.
- multiple `ethos`/SLIP/CDC-ECM could communicate with each other over the same host.
- There may be potential for making `reset` work better with `native`.

maribu, 2023-05-19:
Ping?

miri64, 2023-05-23:
Started working on that, got other responsibilities, project got staled, no idea which state it is now ^^" Still, would be nice to have!