RIOT-OS/RIOT #14620: compile_and_test_for_board.py: provide xunit file?

miri64, 2020-07-27:
Does it make sense to add optional xunit XML support for `compile_and_test_for_board.py`? This way the results of #14587 could be folded into the results of https://github.com/RIOT-OS/Release-Specs/pull/174 and the nightly HiL tests.

MrKevinWeiss, 2020-08-09:
I think it would be nice to have a simplified xunit output as this is a standard way of interpreting the data. We would have to somehow agree on a naming scheme (eg. toolchain_testname_boardname.xml) or should that just be part of the directory structure?

miri64, 2020-08-09:
I was rather thinking putting the whole result for a in one xunit file: every application could be a "test suite", every step in the (compile, flash, test) could be a "test case", if a test is skipped (if that is implemented e.g. for tests that require sudo without being provided sudo).

kaspar030, 2020-08-10:
> I think it would be nice to have a simplified xunit output as this is a standard way of interpreting the data.

Is it possible to nicely add text output to xunit, if that includes xml tags and/or binary data? Last time I tried, that was causing issues.

miri64, 2020-08-11:
It works for the [release tests](https://github.com/RIOT-OS/RIOT/actions/runs/199999175) (see ZIP under "Upload test report").

miri64, 2020-08-11:
@kaspar030 here is how they did it in pytest (which is used to generate the release test reports): https://github.com/pytest-dev/pytest/blob/2bd0d97fcc25911ede3f883091d591cbf030eb73/src/_pytest/junitxml.py#L176-L193

MrKevinWeiss, 2020-08-18:
> I was rather thinking putting the whole result for a in one xunit file: every application could be a "test suite", every step in the (compile, flash, test) could be a "test case", if a test is skipped (if that is implemented e.g. for tests that require sudo without being provided sudo).

So it would be similar to this [script](https://github.com/RIOT-OS/RobotFW-scripts/blob/master/hil-xml.sh). I think it would be nice to just enable the xunit output from pytest and combine and organize them after.

miri64, 2020-08-18:
Oh I also got some stuff lying around using the `python-xunit` package: https://github.com/miri64/RIOT/commit/ab484a05df89302ce74952a8411bcb01c8389245

miri64, 2020-08-18:
I can PR that if we want it.

MrKevinWeiss, 2020-08-18:
That is another way to do it. Are you sure you don't want to use the pytest xunit stuff, maybe even for the record property? If not then you can PR and I will test.

miri64, 2020-08-18:
If I judge this correctly, I would say using pytest for this would require a complete rewrite of the script, as every test case must then be written as such as pytest expects it. With my approach, the jUnit stuff remains optional.

miri64, 2020-08-18:
According to its doc `junit-xml` [supports record properties as well](https://github.com/kyrus/python-junit-xml/blob/master/junit_xml/__init__.py#L26-L28). And then again, adding some extra XML tags isn't black magic.

miri64, 2020-08-18:
Also consider that `pytest` is a very big dependency that with a restructuring would become mandatory, just to generate another file. With the approach in my branch, the `junit-xml` is still optional.

MrKevinWeiss, 2020-08-18:
> According to its doc junit-xml supports record properties as well. And then again, adding some extra XML tags isn't black magic.

Good enough for me

miri64, 2020-08-25:
See https://github.com/RIOT-OS/RIOT/pull/14851

closed 2020-08-27 by MrKevinWeiss